SmingHub/Sming #998: Making basic blink example throws the error

aruldd, 2017-03-04:
Using the following config:
Platform : Ubuntu 14.04 x64
Precompiled esp-open-sdk x64 1.5.4 (compling from scratch throws [#181](https://github.com/pfalcon/esp-open-sdk/issues/181))
Everything else same as linux quickstart

```
:Basic_Blink$ make
make[1]: Entering directory `/opt/sming/Sming'
make[2]: Entering directory `/opt/sming/Sming/third-party/esp-open-lwip'
make[2]: Nothing to be done for `all'.
make[2]: Leaving directory `/opt/sming/Sming/third-party/esp-open-lwip'
make[1]: Leaving directory `/opt/sming/Sming'
LD out/build/app.out
/opt/esp-open-sdk/xtensa-lx106-elf/bin/../xtensa-lx106-elf/sysroot/usr/lib/liblwip_open.a(tcp.o): In function `tcp_connect':
tcp.c:(.irom0.text+0xacc): undefined reference to `DEBUG_printf'
tcp.c:(.irom0.text+0xae2): undefined reference to `DEBUG_printf'
tcp.c:(.irom0.text+0xb26): undefined reference to `DEBUG_printf'
tcp.c:(.irom0.text+0xb32): undefined reference to `DEBUG_printf'
tcp.c:(.irom0.text+0xb3e): undefined reference to `DEBUG_printf'
/opt/esp-open-sdk/xtensa-lx106-elf/bin/../xtensa-lx106-elf/sysroot/usr/lib/liblwip_open.a(tcp.o):tcp.c:(.irom0.text+0xb4a): more undefined references to `DEBUG_printf' follow
collect2: error: ld returned 1 exit status
make: *** [out/build/app.out] Error 1

```

ADiea, 2017-03-04:
Will investigate

ADiea, 2017-03-04:
Please show me how you compile sming lib. The make line

ADiea, 2017-03-04:
Try building blink with ```make ENABLE_CUSTOM_LWIP=0``` and let me know if this still fails. By default ENABLE_CUSTOM_LWIP is 1 and you have to use the lib compiled from sources

mgottschlag, 2017-03-05:
I am seeing the same problem. Fedora 25 x64, everything as in the linux quickstart page.

~/opt/esp-open-sdk/esp-open-lwip/include/arch/cc.h contains a reference to DEBUG_printf, but that function is never defined anywhere. Neither in esp-open-sdk, nor in Sming's custom lwip. Therefore, ENABLE_CUSTOM_LWIP=0 does not seem to help (unless I am doing it wrong somewhere). Not defining LWIP_DEBUG might help, though I don't know how to do that.

mgottschlag, 2017-03-05:
Oh, actually, ENABLE_CUSTOM_LWIP seems to work, and I indeed did it wrong :)

aruldd, 2017-03-06:
I succeeded building the example using the following line.
`make ENABLE_CUSTOM_LWIP=0 DISABLE_SPIFFS=0`

slaff, 2017-03-06:
@aruldd Please, test with the following commands:

```
cd  $SMING_HOME
make dist-clean
make
```

If there is a step that fails, please, re-run that step with V=1. For example if `make` fails re-run it like this `make V=1`.  Check also this Wiki document for [hints](https://github.com/SmingHub/Sming/wiki/Sample-Compilation-Issues).

slaff, 2017-03-16:
@aruldd Feel free to re-open the issue once you have tested the commands are written in the previous comment.

closed 2017-03-16 by slaff