SmingHub/Sming #999: Errors while compiling LWIP under Windows

vg78, 2017-03-04:
Compiling Sming develop using UDK 2.2 (SDK 2.0.0) 
gives plenty of warnings on LWIP stack.
Implicit declaration of functions vPortFree, pvPortMalloc, pvPortZalloc.

Here is the make output (partially shortened ...):

```makefile
10:30:12 **** Incremental Build of configuration Sming for project SmingFramework ****
make all 
Fetching third-party/rboot/ ...
Submodule 'Sming/rboot' (https://github.com/raburton/rboot.git) registered for path 'third-party/rboot'
Cloning into 'D:/deva/esp/proj/Sming/Sming/third-party/rboot'...
Submodule path 'third-party/rboot': checked out '81caa867dbfbb2447b0af66c8110bb8641d8c29d'
Checking patch Makefile...
Checking patch rboot.c...
Applied patch Makefile cleanly.
Applied patch rboot.c cleanly.
Fetching third-party/spiffs/ ...
Submodule 'Sming/third-party/spiffs' (https://github.com/pellepl/spiffs.git) registered for path 'third-party/spiffs'
Cloning into 'D:/deva/esp/proj/Sming/Sming/third-party/spiffs'...
Submodule path 'third-party/spiffs': checked out '21fe570974103f71cf7ff02f9888c77746fbc66d'
error: unrecognized input
make: [third-party/spiffs/makefile] Error 128 (ignored)
Fetching third-party/esp-open-lwip/ ...
Submodule 'Sming/third-party/esp-open-lwip' (https://github.com/pfalcon/esp-open-lwip.git) registered for path 'third-party/esp-open-lwip'
Cloning into 'D:/deva/esp/proj/Sming/Sming/third-party/esp-open-lwip'...
Submodule path 'third-party/esp-open-lwip': checked out 'b9a23a6e9ef88d995ae74014e71a6a4227f61d62'
Checking patch include/user_config.h...
Checking patch Makefile.open...
Checking patch include/arch/cc.h...
Checking patch include/lwipopts.h...
Checking patch include/lwip/ip_addr.h...
Checking patch include/lwip/mem.h...
Checking patch lwip/app/dhcpserver.c...
Applied patch include/user_config.h cleanly.
Applied patch Makefile.open cleanly.
Applied patch include/arch/cc.h cleanly.
Applied patch include/lwipopts.h cleanly.
Applied patch include/lwip/ip_addr.h cleanly.
Applied patch include/lwip/mem.h cleanly.
Applied patch lwip/app/dhcpserver.c cleanly.
make[1]: Entering directory `/D/deva/esp/proj/Sming/Sming'
make[2]: Entering directory `/D/deva/esp/proj/Sming/Sming/third-party/esp-open-lwip'
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -I/c/Espressif/ESP8266_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/core/def.o lwip/core/def.c
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -I/c/Espressif/ESP8266_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/core/dhcp.o lwip/core/dhcp.c
In file included from include/lwip/stats.h:37:0,
                 from lwip/core/dhcp.c:73:
lwip/core/dhcp.c: In function 'dhcp_cleanup':
include/lwip/mem.h:55:21: warning: implicit declaration of function 'vPortFree' [-Wimplicit-function-declaration]
 #define mem_free(s) vPortFree(s, "", 0)
                     ^
lwip/core/dhcp.c:629:5: note: in expansion of macro 'mem_free'
     mem_free(netif->dhcp);
     ^
lwip/core/dhcp.c: In function 'dhcp_start':
include/lwip/mem.h:58:23: warning: implicit declaration of function 'pvPortMalloc' [-Wimplicit-function-declaration]
 #define mem_malloc(s) pvPortMalloc(s, "", 0)
                       ^
lwip/core/dhcp.c:673:27: note: in expansion of macro 'mem_malloc'
     dhcp = (struct dhcp *)mem_malloc(sizeof(struct dhcp));
                           ^
lwip/core/dhcp.c: In function 'dhcp_bind':
lwip/core/dhcp.c:1059:3: warning: implicit declaration of function 'system_station_got_ip_set' [-Wimplicit-function-declaration]
   system_station_got_ip_set(&ip, &mask, &gw);
   ^

...

/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -I/c/Espressif/ESP8266_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/core/tcp_in.o lwip/core/tcp_in.c
In file included from include/lwip/tcp.h:40:0,
[... 89 lines cut ...]
 #define mem_free(s) vPortFree(s, "", 0)
                     ^
include/lwip/mem.h:102:20: note: in expansion of macro 'mem_free'
 #define os_free(p) mem_free((p))
                    ^
lwip/app/dhcpserver.c:675:15: note: in expansion of macro 'os_free'
               os_free(pnode);
               ^
lwip/app/dhcpserver.c:703:11: warning: implicit declaration of function 'wifi_softap_set_station_info' [-Wimplicit-function-declaration]
       if (wifi_softap_set_station_info(m->chaddr, &client_address) == false) {
           ^
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -I/c/Espressif/ESP8266_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o espconn_dummy.o espconn_dummy.c
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar rcs /D/deva/esp/proj/Sming/Sming/compiler/lib/liblwip_open.a lwip/core/def.o lwip/core/dhcp.o lwip/core/dns.o lwip/core/init.o lwip/core/mem.o lwip/core/memp.o lwip/core/netif.o lwip/core/pbuf.o lwip/core/raw.o lwip/core/sntp.o lwip/core/stats.o lwip/core/sys_arch.o lwip/core/sys.o lwip/core/tcp.o lwip/core/tcp_in.o lwip/core/tcp_out.o lwip/core/timers.o lwip/core/udp.o lwip/core/ipv4/autoip.o lwip/core/ipv4/icmp.o lwip/core/ipv4/igmp.o lwip/core/ipv4/inet.o lwip/core/ipv4/inet_chksum.o lwip/core/ipv4/ip_addr.o lwip/core/ipv4/ip.o lwip/core/ipv4/ip_frag.o lwip/netif/etharp.o lwip/app/dhcpserver.o espconn_dummy.o
make[2]: Leaving directory `/D/deva/esp/proj/Sming/Sming/third-party/esp-open-lwip'
AS system/irq_check.s
AS gdb/gdbstub-entry.s
CC system/crash_handler.c
CC system/flashmem.c
...
C+ Libraries/si4432/si4432.cpp
AR out/build/app_app.a
Installing libsming
(!) SSL support is not enabled. To enable it type: 'make clean; make ENABLE_SSL=1'
Done
```

ADiea, 2017-03-04:
I have a PR submitted in upstream for this, but we can also fix in the patch.

vg78, 2017-03-04:
I added the functions declaration in the esp-open-lwip/include/user_config.h file

```c
extern void *pvPortMalloc(size_t xWantedSize, const char *file, uint32 line);
extern void *pvPortZalloc(size_t xWantedSize, const char *file, uint32 line);
extern void pvPortFree(void *ptr);
extern void vPortFree(void *ptr, const char *file, uint32 line);
extern void *vPortMalloc(size_t xWantedSize);
```

but there are new warnings:

```make
lwip/core/dhcp.c: In function 'dhcp_bind':
lwip/core/dhcp.c:1059:3: warning: implicit declaration of function 'system_station_got_ip_set' [-Wimplicit-function-declaration]
   system_station_got_ip_set(&ip, &mask, &gw);
   ^
lwip/core/tcp.c: In function 'tcp_slowtmr':
lwip/core/tcp.c:813:33: warning: implicit declaration of function 'system_get_data_of_array_8' [-Wimplicit-function-declaration]
         if (pcb->persist_cnt >= system_get_data_of_array_8(tcp_persist_backoff, pcb->persist_backoff-1)) {
                                 ^
lwip/core/ipv4/ip_addr.c: In function 'ipaddr_aton':
lwip/core/ipv4/ip_addr.c:170:10: warning: implicit declaration of function 'isdigit' [-Wimplicit-function-declaration]
     if (!isdigit(c))
          ^
lwip/app/dhcpserver.c: In function 'parse_msg':
lwip/app/dhcpserver.c:703:11: warning: implicit declaration of function 'wifi_softap_set_station_info' [-Wimplicit-function-declaration]
       if (wifi_softap_set_station_info(m->chaddr, &client_address) == false) {
           ^
```

missing some function declarations that i can't find in Sming and SDK source

anakod, 2017-03-09:
I found workaround for this issue:
http://stackoverflow.com/questions/11816133/disable-function-declaration-error-in-avr-g

So probably we can add declaration for known funcions and than add flags to ignore others.
Try to add in "Makefile", line 180:
EXTRA_CFLAGS_LWIP = -I../../system/include -I../../Wiring **-Wno-implicit-function-declaration**

@vg78, can you check and make pull request with that fix and added the functions declaration?

ADiea, 2017-03-09:
I think it's better to add funciton signature to os_port.h rather than disable the warning

anakod, 2017-03-10:
You are right, but them isn't declared in Sming\LWIP as I understand.

UPD: I found them: https://github.com/IOCare/esp8266GUI/blob/master/include/espmissingincludes-ip.h

closed 2019-08-07 by slaff