SmingHub/Sming #1009: rBoot + SPIFFS not working

riban-bw, 2017-03-05:
I use SPIFFS in a project but when I enable rBOOT I get `read errno -10011` debug output and SPIFFS is not working. I am using the configuration from Basic_rBoot Makefile-user.mk on a NodeMCU (ESP-12) module.

riban-bw, 2017-03-05:
If I use `spiffs_mount_manual(0x100000, 65536);` then it works so it seems that, when building with rBoot, the automatic detection of SPIFFS location or size fails.

slaff, 2017-03-06:
rBoot saves the SPIFF file system at a [pre-defined location](https://github.com/SmingHub/Sming/blob/develop/Sming/Makefile-rboot.mk#L30). Which by default is `0x100000`. When you use `spiffs_mount_manual(0x100000, 65536)` you are instructing the SpifFS to be mounted starting from address `0x100000`, which is the same one that is hard-coded. 

If you use spiff_mount() is does something else - it calculates the first available sector on flash after your ROMs and looks for SpifFS starting from that sector. Which is the way non-rBoot ROMs flashing works. Since rBoot does not follow that logic you cannot expect `spiffs_mount()` to work for you. 

That can be unified and if desired you should create a new issue and we will tackle the issue in the 3.2.0 release.

riban-bw, 2017-03-06:
I have changed the title of this issue to better reflect the problem.
Please make spiffs_mount() compatible with rBoot. There is a plan to make rBoot the default boot loader so this could / should be done to support that change. Maybe spiffs_mount() could check for rBoot and request information about spiffs from rBoot integration or a more robust and exhaustive search for the spiffs file system. It would also be advantageous for mount_spiffs() to return success indication to the application. Currently it prints to debug output but this does not allow the runtime application to deduce success.

jochenjagers, 2017-10-22:
I opened a pull request to solve this issue
https://github.com/SmingHub/Sming/pull/1269

PR #1292 (2017-12-01): Made spiffs_mount() compatible with rBoot.
Fixes #1009.
Closes #1269
closed 2017-12-01 by slaff