SmingHub/Sming #1011: another Windows compile problem

kwis2, 2017-03-06:
While compile commit c250014  I got errors (translated)

> cc1.exe System error
> The program can't start because libwinpthread-1.dll is missing from your computer. Please reinstall the application to fix this problem

while making Spiffy.c

> Making spiffy utility
CC spiffy.c
make[2]: *** [spiffy.o] Error 1
make[1]: *** [spiffy/spiffy] Error 2
make[1]: Leaving directory `/C/tools/sming/Sming'
make: *** [reload] Error 2

problems were with mingw64 and mingw32

kwis2, 2017-03-06:
problem starts with commit 4e20be5 #964

kwis2, 2017-03-06:
It was possible to compile sming c250014 with procedure:
`make clean` ->OK
`make all` -> ERROR
`make spiffy` -> OK
`make all`

kwis2, 2017-03-06:
after compile sming there a lot of problems with python code with memanalyzer.py
first `python` from 

> Makefile-windows.mk 

do not link to `python.exe`.  

> #Memory / Section info:
> ------------------------------------------------------------------------------
> /bin/sh: python: command not found
> make.exe": *** [out/build/app.out] Error 127
> 
> BUILD FAILED (exit value 2, total time: 9s)

I have changed to `c:/python25/python.exe`
and then I got 

>  #Memory / Section info:
> ------------------------------------------------------------------------------
> Traceback (most recent call last):
>   File "c:/tools/sming/Sming/../tools/memanalyzer.py", line 9, in <module>
>     from collections import OrderedDict
> ImportError: cannot import name OrderedDict
> make.exe": *** [out/build/app.out] Error 1
> 
> BUILD FAILED (exit value 2, total time: 9s)

Unfortunatelly I don't know python, and I don't know how to add collection to windows ....

slaff, 2017-03-06:
python 2.5 is quite old. Is it possible to update your python to 2.7?

kwis2, 2017-03-06:
python 2.7 solved problem.
With choco config 

> esptool.exe

 is placed in 

> c:/Espressif/utils

 so I had to change in 

> Makefile-windows.mk

 from

`ESPTOOL		 ?= $(SDK_TOOLS)/ESP8266/esptool.exe `
to 
`ESPTOOL		 ?= $(SDK_TOOLS)/esptool.exe`

kwis2, 2017-03-06:
Another problem with rboot example:

>  cd 'C:\tools\sming\samples\Basic_rBoot'
> C:\tools\mingw64\msys\1.0\bin\make.exe -f Makefile
> /c/tools/sming/samples/Basic_rBoot/"C:/tools/mingw64/msys/1.0/bin/make.exe" -C /C/tools/sming/Sming/third-party/rboot RBOOT_GPIO_ENABLED=0
> /bin/sh: /c/tools/sming/samples/Basic_rBoot/C:/tools/mingw64/msys/1.0/bin/make.exe: No such file or directory
> make.exe": *** [out/firmware/rboot.bin] Error 127
> 
> BUILD FAILED (exit value 2, total time: 460ms)

in my oppinion it is syntax error in 

> Makefile-rboot.mk

, but I am not able to find ....

slaff, 2017-03-06:
Test these commands

```
cd 'C:\tools\sming\samples\Basic_rBoot'
make V=1
```

Be sure that `make` is in your PATH.

kwis2, 2017-03-06:
> C:\tools\sming\samples\Basic_rBoot>make clean
> 
> C:\tools\sming\samples\Basic_rBoot>make V=1
> mkdir -p out/build/app
> mkdir -p out/build//C/tools/sming/Sming/third-party/rboot/appcode
> mkdir -p out/firmware
> OC out/build/libmain2.a
> make -C /C/tools/sming/Sming/third-party/rboot RBOOT_GPIO_ENABLED=0
> make[1]: Entering directory `/C/tools/sming/Sming/third-party/rboot'
> CC rboot-stage2a.c
> xtensa-lx106-elf-gcc -Os -O3 -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline
> -functions -nostdlib -mlongcalls -mtext-section-literals  -D__ets__ -DICACHE_FLA
> SH -DBOOT_BIG_FLASH -I. -c rboot-stage2a.c -o /c/tools/sming/samples/Basic_rBoot
> /out/build/rboot-stage2a.o
> LD /c/tools/sming/samples/Basic_rBoot/out/build/rboot-stage2a.elf
> xtensa-lx106-elf-gcc -Trboot-stage2a.ld -nostdlib -Wl,--no-check-sections -u cal
> l_user_start -Wl,-static -Wl,--start-group /c/tools/sming/samples/Basic_rBoot/ou
> t/build/rboot-stage2a.o -Wl,--end-group -o /c/tools/sming/samples/Basic_rBoot/ou
> t/build/rboot-stage2a.elf
> E2 /c/tools/sming/samples/Basic_rBoot/out/build/rboot-hex2a.h
> c:/Espressif/utils/esptool2 -quiet -header /c/tools/sming/samples/Basic_rBoot/ou
> t/build/rboot-stage2a.elf /c/tools/sming/samples/Basic_rBoot/out/build/rboot-hex
> 2a.h .text
> CC rboot.c
> xtensa-lx106-elf-gcc -Os -O3 -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline
> -functions -nostdlib -mlongcalls -mtext-section-literals  -D__ets__ -DICACHE_FLA
> SH -DBOOT_BIG_FLASH -I. -I/c/tools/sming/samples/Basic_rBoot/out/build -c rboot.
> c -o /c/tools/sming/samples/Basic_rBoot/out/build/rboot.o
> LD /c/tools/sming/samples/Basic_rBoot/out/build/rboot.elf
> xtensa-lx106-elf-gcc -Teagle.app.v6.ld -nostdlib -Wl,--no-check-sections -u call
> _user_start -Wl,-static -Wl,--start-group /c/tools/sming/samples/Basic_rBoot/out
> /build/rboot.o -Wl,--end-group -o /c/tools/sming/samples/Basic_rBoot/out/build/r
> boot.elf
> E2 /c/tools/sming/samples/Basic_rBoot/out/firmware/rboot.bin
> c:/Espressif/utils/esptool2 -quiet -bin -boot0 -4096 -qio -40 /c/tools/sming/sam
> ples/Basic_rBoot/out/build/rboot.elf /c/tools/sming/samples/Basic_rBoot/out/firm
> ware/rboot.bin .text .rodata
> make[1]: Leaving directory `/C/tools/sming/Sming/third-party/rboot'
> /c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -Iapp -I/C/tools/sming/Sm
> ing/third-party/rboot/appcode -Iapp/include -I/C/tools/sming/Sming/third-party/r
> boot/appcode/include -Iinclude -I/C/tools/sming/Sming/include -I/C/tools/sming/S
> ming/ -I/C/tools/sming/Sming/third-party/esp-open-lwip/include -I/C/tools/sming/
> Sming/system/include -I/C/tools/sming/Sming/Wiring -I/C/tools/sming/Sming/Librar
> ies -I/C/tools/sming/Sming/SmingCore -I/C/tools/sming/Sming/Services/SpifFS -I/c
> /Espressif/ESP8266_SDK/../include -I/C/tools/sming/Sming/third-party/rboot -I/C/
> tools/sming/Sming/third-party/rboot/appcode -I/C/tools/sming/Sming/third-party/s
> piffs/src -Ithird-party/esp-open-lwip/include -I/c/Espressif/ESP8266_SDK/include
>  -Wpointer-arith -Wundef -Werror -Wl,-EL -nostdlib -mlongcalls -mtext-section-li
> terals -finline-functions -fdata-sections -ffunction-sections -D__ets__ -DICACHE
> _FLASH -DARDUINO=106 -DCOM_SPEED_SERIAL=115200  -Os -g -DCUST_FILE_BASE=applicat
> ion -DDEBUG_VERBOSE_LEVEL=2 -DDEBUG_PRINT_FILENAME_AND_LINE=0 -DBOOT_BIG_FLASH -
> DSPIFF_SIZE=65536 -DRBOOT_SPIFFS_0=0x100000 -DRBOOT_SPIFFS_1=0x300000 -DRBOOT_IN
> TEGRATION -fno-rtti -fno-exceptions -std=c++11 -felide-constructors -c app/appli
> cation.cpp -o out/build/app/application.o
> /c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -Iapp -I/C/tools/sming/Sm
> ing/third-party/rboot/appcode -Iapp/include -I/C/tools/sming/Sming/third-party/r
> boot/appcode/include -Iinclude -I/C/tools/sming/Sming/include -I/C/tools/sming/S
> ming/ -I/C/tools/sming/Sming/third-party/esp-open-lwip/include -I/C/tools/sming/
> Sming/system/include -I/C/tools/sming/Sming/Wiring -I/C/tools/sming/Sming/Librar
> ies -I/C/tools/sming/Sming/SmingCore -I/C/tools/sming/Sming/Services/SpifFS -I/c
> /Espressif/ESP8266_SDK/../include -I/C/tools/sming/Sming/third-party/rboot -I/C/
> tools/sming/Sming/third-party/rboot/appcode -I/C/tools/sming/Sming/third-party/s
> piffs/src -Ithird-party/esp-open-lwip/include -I/c/Espressif/ESP8266_SDK/include
>  -Wpointer-arith -Wundef -Werror -Wl,-EL -nostdlib -mlongcalls -mtext-section-li
> terals -finline-functions -fdata-sections -ffunction-sections -D__ets__ -DICACHE
> _FLASH -DARDUINO=106 -DCOM_SPEED_SERIAL=115200  -Os -g -DCUST_FILE_BASE=rboot-ap
> i -DDEBUG_VERBOSE_LEVEL=2 -DDEBUG_PRINT_FILENAME_AND_LINE=0 -DBOOT_BIG_FLASH -DS
> PIFF_SIZE=65536 -DRBOOT_SPIFFS_0=0x100000 -DRBOOT_SPIFFS_1=0x300000 -DRBOOT_INTE
> GRATION -c /C/tools/sming/Sming/third-party/rboot/appcode/rboot-api.c -o out/bui
> ld//C/tools/sming/Sming/third-party/rboot/appcode/rboot-api.o
[... 9 lines cut ...]
>  -Wpointer-arith -Wundef -Werror -Wl,-EL -nostdlib -mlongcalls -mtext-section-li
> terals -finline-functions -fdata-sections -ffunction-sections -D__ets__ -DICACHE
> _FLASH -DARDUINO=106 -DCOM_SPEED_SERIAL=115200  -Os -g -DCUST_FILE_BASE=rboot-bi
> gflash -DDEBUG_VERBOSE_LEVEL=2 -DDEBUG_PRINT_FILENAME_AND_LINE=0 -DBOOT_BIG_FLAS
> H -DSPIFF_SIZE=65536 -DRBOOT_SPIFFS_0=0x100000 -DRBOOT_SPIFFS_1=0x300000 -DRBOOT
> _INTEGRATION -c /C/tools/sming/Sming/third-party/rboot/appcode/rboot-bigflash.c
> -o out/build//C/tools/sming/Sming/third-party/rboot/appcode/rboot-bigflash.o
[... cut ...]
> art_local  -Wl,--start-group out/build/app_app.a -lmicroc -lmicrogcc -lhal -lphy
>  -lpp -lnet80211 -llwip_open -lwpa -lmain2 -lsming -lcrypto -lpwm -lsmartconfig
> -Wl,--end-group -o out/build/app_0.out
> python /C/tools/sming/Sming/../tools/memanalyzer.py /c/Espressif/xtensa-lx106-el
> f/bin/xtensa-lx106-elf-objdump.exe out/build/app_0.out > out/firmware/fwMeminfo.
> new
> /bin/sh: python: command not found
> make: *** [out/build/app_0.out] Error 127
> 
> C:\tools\sming\samples\Basic_rBoot>

slaff, 2017-03-06:
> python /C/tools/sming/Sming/../tools/memanalyzer.py /c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-objdump.exe out/build/app_0.out > out/firmware/fwMeminfo.new

Now, put `python` also in your PATH.

kwis2, 2017-03-06:
When I add PATH `make V=1` start to work, even `make` works but in netbeans I use there is 

```
cd 'C:\tools\sming\samples\Basic_rBoot'
C:\tools\mingw64\msys\1.0\bin\make.exe -f Makefile
OC out/build/libmain2.a
/c/tools/sming/samples/Basic_rBoot/"C:/tools/mingw64/msys/1.0/bin/make.exe" -C /C/tools/sming/Sming/third-party/rboot RBOOT_GPIO_ENABLED=0
/bin/sh: /c/tools/sming/samples/Basic_rBoot/C:/tools/mingw64/msys/1.0/bin/make.exe: No such file or directory
make.exe": *** [out/firmware/rboot.bin] Error 127

BUILD FAILED (exit value 2, total time: 437ms)
```

kwis2, 2017-03-06:
I found. Netbeans converts `$(MAKE)` into whole path, and there is no separating mark in 

> makefie-rboot.mk

change `$(MAKE)` into `make` in line 433 solves problem.

slaff, 2017-03-06:
The right solution is to change your environment to match the requirements. And not vice-versa. This means that you should change globally your PATH environment so that `make` can be found from your OS.

kwis2, 2017-03-06:
@slaff you are right, netbeans makes its own enviroment with its own paths. 
When I have changed `C:\tools\mingw64\msys\1.0\bin\make.exe` to `make.exe` `Basic_rBoot` was compiled.
Problem is that I would preffer to know which `make` was used, and the most interesting question is why programs with 

> Makefile-project.mk

 compile but 

> Makefile-rboot.mk

 not?

kwis2, 2017-03-06:
With new commit 89d529c problem with `make flash` there is problem with python and esptool

> /c/Espressif/utils/ESP8266/esptool.exe -p COM5 -b 115200 write_flash -ff 40m -fm
>  dio -fs 4m 0x00000 out/firmware/0x00000.bin 0x0a000 out/firmware/0x0a000.bin  0
> x4D000 out/firmware/spiff_rom.bin
> LoadLibrary(pythondll) failedNie mo¬na odnalečŠ okreťlonego modu-u.
> c:\Espressif\utils\ESP8266\PYTHON27.DLLmake: *** [flash] Error 255

in translation 

> pythondll failed, module not found
and there is no pythondll file in python27 dir

kwis2, 2017-03-07:
Source of problem it was place of esptool.exe file. According to appveyor.yml I have copied esptool.exe from c:\Espressif\utils\ to c:\Espressif\utils\ESP8266\. I have tried several times to flash board with no success.
When I have changed makefile-windows.mk `ESPTOOL ?= $(SDK_TOOLS)/esptool.exe` and removed ESP8266 folder from c:/Espressif/utils flash work

anakod, 2017-03-07:
> I have copied esptool.exe from c:\Espressif\utils\ to c:\Espressif\utils\ESP8266

Please try to install new UDK version - it should have right path.

kwis2, 2017-03-07:
With UDK 2.2.1 there was problem - all paths ok, programs - examples compiled but not works. There were serial communication with unknown speed, I have returned to UDK 2.0.9

closed 2017-03-09 by kwis2