SmingHub/Sming #1016: Can only attach one interrupt to a pin

riban-bw, 2017-03-08:
If you try to assign multiple interrupts, using different interrupt modes, to the same pin, only the last attached will work and all others are ignored. Example:
```
attachInterrupt(PIN_4, posEdgeInt, GPIO_PIN_INTR_POSEDGE);
attachInterrupt(PIN_4, negEdgeInt, GPIO_PIN_INTR_NEGEDGE);
```
This will result in calls to negEdgeInt() on a falling edge on PIN_4 but nothing on a rising edge on PIN_4.
I appreciate there are coding methods to work around this but they all add processing time which in some applications is undesirable. As a minimum this should be documented but ideally, it should be possible to attach multiple interrupt events to a pin.

teicors, 2017-03-08:
Hi,
I'm using this trick:

void IRAM_ATTR interruptHandler00()
{
     detachInterrupt(PIN_BUTTON);
     attachInterrupt(PIN_BUTTON, interruptHandler01, RISING);
}

void IRAM_ATTR interruptHandler01()
{
     detachInterrupt(PIN_BUTTON);
     attachInterrupt(PIN_BUTTON, interruptHandler00, FALLING);
}

ad it works.

Cheers,
Angelo

neut2012, 2017-03-13:
You need to do it like this:
`attachInterrupt(PIN_4, func, GPIO_PIN_INTR_ANYEDGE);`

riban-bw, 2017-03-13:
@neut2012 that does not allow different routines to be called for each edge. 
@teicors that is an option but it adds time in the call which is what I am trying to avoid.

anakod, 2017-03-13:
Does ESP SDK internals add multiple callbacks?

neut2012, 2017-03-13:
Oh, I'm sorry I skimmed through the question and didn't get what you're trying to do.

I think it's an SDK limitation and your only way is to have one call back and check the state of the pin.