SmingHub/Sming #1017: WDT can't wake up ESP

kwis2, 2017-03-09:
Sometimes module hangs and can not wake up. This time I have logs with `Serial.systemDebugOutput(true)`.
My own program, www interaction hangs module. I wasn't able to reproduce problem with any samples. 
8:08:33 - module hangs, reset, 
8:08:41 once again WDT reset 
8:10:19 I have pressed reset button
Problem is not that module hangs, but that WDT do not wake up module to normal work. 
Module ESP8266-01S, Sming v 3.1.2, but in previous version module sometimes hangs too.

with XXXXX I have masked SSID and passwd

> Terminal log file
> Date: 2017-03-09 - 08:07:38
> -----------------------------------------------
> 8:07:47.263, rver onClient: 192.168.0.8
> 8:08:29.201, onAccept state: 0 K=0Free heap size=30688, K=0+TCP csthost === 192.168.0.26parsedRequest: GonAccept state: 0 K=0Free heap size=30464, K=0+TCP connectiontimeout updating: 70 -> sendBodyREAD Template (0)found var: ESP_frm_reload, at 274 (273) - 288, send size 273Written: 273, Available: 273, isFinished: 0, PushCount: 1READ Template (2)StartVar ESP_frm_reload(6543 bytes)SSID: XXXXXXXXXSSID: XXXXXXXXXSSID: XXXXXXXXXSSID: UPCTCP sent: 336onReadyToSendData: 2response sendBodyREAD Template (3)continue to plain Available: 663, isFinished: 0, PushCount: 2READ Template (2)StartVar ESP_IDWritten: 6, Available: 6,TCP sent: 1390onReadyToSendData: 2response108, send size 98Written: 98, Available: 98, isFinished: 0, PushCount: 10READ Template (2)StartVar encrypt_5Written: 4, Available: 4, isFinished: 0, PushCount: 11READ Template (3)continue to plaint textplain template text pos: 6255, len: 288Written: 288, Available: 288, isFinished: 1, PushCount: 12Stream completedTCP connection closing TCP connectiontinue to plaint textfound var: select_net_6, at 111 (5952) - 123, send size 110Written: 110, Available: 110, isFinished: 0, PushCount: 4READ Template (2)var select_net_6 not foundWritten: 14, Available: 14, isFinished: 0, PushCount: 5READ Template (0)found var: wifi_power_6, at 42 (6007) - 54, send size 41Written: 41, Available: 41, isFinished: 0, PushCount: 6READ Template (2)StartVar wifi_power_6Written: 11, Available: 11, isFinished: 0, Pu 39parsedRequest: POST, 39 bytesItem: Name = SSID_name, Size = 10, Value = XXXXXXXXXItem: rm 0del if0bcn 0del if1usl118Written: 118, Available: 118, isFinished: 0, PushCount: 23READ Template (2)StartVar net_3Written: 4, Available: 4, isFinished: 0, PushCount: 24READ Template (3)continue to plaint textfound var: encrypt_3, at 99 (5005) - 108, send size 98Written: 98, Available: 98, isFinished: 0, PushCount: 25
> 8:08:29.201, >>          -----  networks try to connect
> 8:08:29.309, onAccept state: 0 K=0Free heap size=30688, K=0+TCP csthost === 192.168.0.26parsedRequest: GonAccept state: 0 K=0Free heap size=30464, K=0+TCP connectiontimeout updating: 70 -> sendBodyREAD Template (0)found var: ESP_frm_reload, at 274 (273) - 288, send size 273Written: 273, Available: 273, isFinished: 0, PushCount: 1READ Template (2)StartVar ESP_frm_reload(6543 bytes)SSID: XXXXXXXXXSSID: XXXXXXXXXSSID: XXXXXXXXXSSID: UPCTCP sent: 336onReadyToSendData: 2response sendBodyREAD Template (3)continue to plain Available: 663, isFinished: 0, PushCount: 2READ Template (2)StartVar ESP_IDWritten: 6, Available: 6,TCP sent: 1390onReadyToSendData: 2response108, send size 98Written: 98, Available: 98, isFinished: 0, PushCount: 10READ Template (2)StartVar encrypt_5Written: 4, Available: 4, isFinished: 0, PushCount: 11READ Template (3)continue to plaint textplain template text pos: 6255, len: 288Written: 288, Available: 288, isFinished: 1, PushCount: 12Stream completedTCP connection closing TCP connectiontinue to plaint textfound var: select_net_6, at 111 (5952) - 123, send size 110Written: 110, Available: 110, isFinished: 0, PushCount: 4READ Template (2)var select_net_6 not foundWritten: 14, Available: 14, isFinished: 0, PushCount: 5READ Template (0)found var: wifi_power_6, at 42 (6007) - 54, send size 41Written: 41, Available: 41, isFinished: 0, PushCount: 6READ Template (2)StartVar wifi_power_6Written: 11, Available: 11, isFinished: 0, Pu2, Available: 12, isFinished: 0, PushCount: 7READ Template (3)continue to plaint textfound var: ESP_l if0bcn 0del if1usl118Written: 118, Available: 118, isFinished: 0, PushCount: 23READ Template (2)StartVar net_3Written: 4, Available: 4, isFinished: 0, PushCount: 24READ Template (3)continue to plaint textfound var: encrypt_3, at 99 (5005) - 108, send size 98Written: 98, Available: 98, isFinished: 0, PushCount: 25
> 8:08:29.309, >>          -----  networks try to connect
> 8:08:29.419, onAccept state: 0 K=0Free heap size=30688, K=0+TCP csthost === 192.168.0.26parsedRequest: GonAccept state: 0 K=0Free heap size=30464, K=0+TCP connectiontimeout updating: 70 -> sendBodyREAD Template (0)found var: ESP_frm_reload, at 274 (273) - 288, send size 273Written: 273, Available: 273, isFinished: 0, PushCount: 1READ Template (2)StartVar ESP_frm_reload(6543 bytes)SSID: XXXXXXXXXSSID: XXXXXXXXXSSID: XXXXXXXXXSSID: UPCTCP sent: 336onReadyToSendData: 2response sendBodyREAD Template (3)continue to plain Available: 663, isFinished: 0, PushCount: 2READ Template (2)StartVar ESP_IDWritten: 6, Available: 6,TCP sent: 1390onReadyToSendData: 2response108, send size 98Written: 98, Available: 98, isFinished: 0, PushCount: 10READ Template (2)StartVar encrypt_5Written: 4, Available: 4, isFinished: 0, PushCount: 11READ Template (3)continue to plaint textplain template text pos: 6255, len: 288Written: 288, Available: 288, isFinished: 1, PushCount: 12Stream completedTCP connection closing TCP connectiontinue to plaint textfound var: select_net_6, at 111 (5952) - 123, send size 110Written: 110, Available: 110, isFinished: 0, PushCount: 4READ Template (2)var select_net_6 not foundWritten: 14, Available: 14, isFinished: 0, PushCount: 5READ Template (0)found var: wifi_power_6, at 42 (6007) - 54, send size 41Written: 41, Available: 41, isFinished: 0, PushCount: 6READ Template (2)StartVar wifi_power_6Written: 11, Available: 11, isFinished: 0, Pu_SSIDWritten: 10, Available: 10, isFinished: 0, PushCount: 2READ Template (3)continue to plaint 119 (4900) - 124, send size 118Written: 118, Available: 118, isFinished: 0, PushCount: 23READ Template (2)StartVar net_3Written: 4, Available: 4, isFinished: 0, PushCount: 24READ Template (3)continue to plaint textfound var: encrypt_3, at 99 (5005) - 108, send size 98Written: 98, Available: 98, isFinished: 0, PushCount: 25
> 8:08:29.419, >>          -----  networks try to connect
> 8:08:29.525, onAccept state: 0 K=0Free heap size=30688, K=0+TCP csthost === 192.168.0.26parsedRequest: GonAccept state: 0 K=0Free heap size=30464, K=0+TCP connectiontimeout updating: 70 -> sendBodyREAD Template (0)found var: ESP_frm_reload, at 274 (273) - 288, send size 273Written: 273, Available: 273, isFinished: 0, PushCount: 1READ Template (2)StartVar ESP_frm_reload(6543 bytes)SSID: XXXXXXXXXSSID: XXXXXXXXXSSID: XXXXXXXXXSSID: UPCTCP sent: 336onReadyToSendData: 2response sendBodyREAD Template (3)continue to plain Available: 663, isFinished: 0, PushCount: 2READ Template (2)StartVar ESP_IDWritten: 6, Available: 6,TCP sent: 1390onReadyToSendData: 2response108, send size 98Written: 98, Available: 98, isFinished: 0, PushCount: 10READ Template (2)StartVar encrypt_5Written: 4, Available: 4, isFinished: 0, PushCount: 11READ Template (3)continue to plaint textplain template text pos: 6255, len: 288Written: 288, Available: 288, isFinished: 1, PushCount: 12Stream completedTCP connection closing TCP connectiontinue to plaint textfound var: select_net_6, at 111 (5952) - 123, send size 110Written: 110, Available: 110, isFinished: 0, PushCount: 4READ Template (2)var select_net_6 not foundWritten: 14, Available: 14, isFinished: 0, PushCount: 5READ Template (0)found var: wifi_power_6, at 42 (6007) - 54, send size 41Written: 41, Available: 41, isFinished: 0, PushCount: 6READ Template (2)StartVar wifi_power_6Written: 11, Available: 11, isFinished: 0, Puilable: 14, isFinished: 0, PushCount: 12READ Template (0)found var: wifi_power_2, at 43 (4355) - 519 (4900) - 124, send size 118Written: 118, Available: 118, isFinished: 0, PushCount: 23READ Template (2)StartVar net_3Written: 4, Available: 4, isFinished: 0, PushCount: 24READ Template (3)continue to plaint textfound var: encrypt_3, at 99 (5005) - 108, send size 98Written: 98, Available: 98, isFinished: 0, PushCount: 25
> 8:08:29.525, >>          -----  networks try to connect
[... cut ...] le: 78, isFinished: 1, PushCount: 1TcpClient request completedTCP sent: 78onReadyToSendData: 2Header pos: 143Date === Thu, 09 Mar 2017 07:10:27 GMTServer === ApacheX-Powered-By === PHP/5.4.45-ogc0Connection === closeContent-Type === text/htmlTCP received: 422 bytesonReadyToSendData: 1TcpConnection::staticOnReceive: pbuf is NULLTCP received: (null)time updated? 1>>Time      -----  09.03.2017 08:10:27
> 
> -----------------------------------------------
> Date: 2017-03-09 - 08:10:35
> End log file

slaff, 2017-03-09:
@kwis2 Most probably you have flashed a new application and forgot or left the ESP8266 attached to the USB cable which holds the GPIO pins. My recommendation is once the application is uploaded to the device to power it off, detach the USB cable and then power it on.

kwis2, 2017-03-09:
If it was reason then module wouldn't be wake up after hard reset - ESP8266-01 has no USB driver, just serial

kwis2, 2017-03-09:
Afternoon I'll solder led to a pin, what should claryfy

slaff, 2017-03-09:
> 8:10:19 I have pressed reset button
Do you use NodeMCU ?

kwis2, 2017-03-09:
I use simple ESP8266-01s with USB/serial dongle. I have mdified dongle with 1000uF cap to be sure that power is OK

kwis2, 2017-03-09:
LED connected to GPIO2, with blink timer stopped. So board hangs. 
When I program ESP and then run module system hangs with some circumstances. The same procedure with system hard reset after program download don't kill system, but after some time it is able to wake up with very strange way. Below is log with such situation.
I add a lot of comments with `>>       -----notes`. and `>>tick     ---` it is led blink procedure
Proper reset procedure should look like one in 18:48:44. In program there is possibility to go to deep sleep, but in this module there is no phisycal connection allow ESP to wake up from deep sleep. 
Additionally in init there is `Serial.println` with reset and reset2 strings, so it is not possible to perform reset without such information.
I have no idea what happened... 

```
17:48:40.301, onAccept state: 0 K=0Free heap size=32048, K=0+TCP connectiontimeout updating: 70 -> 90TcpServer onClient: 192.168.0.8
17:48:42.812, ***** Software Watchdolready freedúřct result:, 0OnConnectedTCP connectedonReadyToSendData: 0Written: 78, Available: 78, isFinished: 1, PushCount: 1TcpClient request completedTCP sent: 78onReadyToSendData: 2Header pos: 143Date === Thu, 09 Mar 2017 16:48:47 GMTServer === ApacheX-Powered-By === PHP/5.4.45-ogc0Connection === closeContent-Type === text/htmlTCP received: 422 bytesonReadyToSendData: 1TcpConnection::staticOnReceive: pbuf is NULLTCP received: (null)time updated? 1>>Time      -----  09.03.2017 17:48:47
17:48:42.812, >>          -----  Deep Sleep
17:48:42.812, >>          -----  Station Close
17:48:42.812, >>SLEEP     -----  09.03.2017 16:48:47 tn 1489078127 t% 227 td 72000
17:48:42.812, 3fffff70:  3ffed744 3fffdcc0 3ffeafd0 3ffeafd0
17:48:42.812, 3fffff80:  4021a9aa 300000007 00000001
17:48:42.812, path=/config.htmlName: hosthost === 192.168.0.18parsedRequest: GET, nodataTCP received: 375 bytesonReadyToSendData: 1attached file: config.html (6543 bytes)SSID: XXXXXXXXSSID: XXXXXXXXSSID: XXXXXXXXSSID: XXXXXXXXSSID: XXXXXXXX***ABORT***
17:48:42.812, state: 5 -> 0 (0)rm 0TCP connection error: -8del if0usl 4021aa37 3ffee9ec
17:48:42.921, >>          -----  Deep Sleep
17:48:42.921, >>          -----  Station Close
17:48:42.921, >>SLEEP     -----  09.03.2017 16:48:47 tn 1489078127 t% 227 td 72000
17:48:42.921, 3ffff860:  00000000 00000009 0000ff00 402411fc
17:48:42.921, 3ffff870:  3fff42actedonReadyToSendData: 0Written: 78, Available: 78, isFinished: 1, PushCount: 1TcpClient request completedTCP sent: 78onReadyToSendData: 2Header pos: 143Date === Thu, 09 Mar 2017 16:48:47 GMTServer === ApacheX-Powered-By === PHP/5.4.45-ogc0Connection === closeContent-Type === text/htmlTCP received: 422 bytesonReadyToSendData: 1TcpConnection::staticOnReceive: pbuf is NULLTCP received: (null)time updated? 1>>Time      -----  09.03.2017 17:48:47
17:48:42.921, be4
17:48:42.921, state: 5 -> 0 (0)rm 0TCP connection error: -8del if0usl 4021aa37 3ffee9ec
17:48:43.031, >>          -----  Deep Sleep
17:48:43.031, >>          -----  Station Close
17:48:43.031, >>SLEEP     -----  09.03.2017 16:48:47 tn 1489078127 t% 227 td 72000
17:48:43.031, 3ffff960:  00353733 00000000 00000000 00000000
17:48:43.031, 3ffff970:  00000000 edonReadyToSendData: 0Written: 78, Available: 78, isFinished: 1, PushCount: 1TcpClient request completedTCP sent: 78onReadyToSendData: 2Header pos: 143Date === Thu, 09 Mar 2017 16:48:47 GMTServer === ApacheX-Powered-By === PHP/5.4.45-ogc0Connection === closeContent-Type === text/htmlTCP received: 422 bytesonReadyToSendData: 1TcpConnection::staticOnReceive: pbuf is NULLTCP received: (null)time updated? 1>>Time      -----  09.03.2017 17:48:47
17:48:43.031, 3fffff70:  3ffed744 3fffdcc0 3ffeafd0 3ffeafd0
17:48:43.031, 784
17:48:43.031, state: 5 -> 0 (0)rm 0TCP connection error: -8del if0usl 4021aa37 3ffee9ec
17:48:43.140, >>          -----  Deep Sleep
17:48:43.140, >>          -----  Station Close
17:48:43.140, >>SLEEP     -----  09.03.2017 16:48:47 tn 1489078127 t% 227 td 72000
17:48:43.140, 002 0000ff00 401055c0
17:48:43.140, 3ffffa70:  00000003 4020ec1c 3ffffb00 00000002
17:48:43.140, 3fffonReadyToSendData: 0Written: 78, Available: 78, isFinished: 1, PushCount: 1TcpClient request completedTCP sent: 78onReadyToSendData: 2Header pos: 143Date === Thu, 09 Mar 2017 16:48:47 GMTServer === ApacheX-Powered-By === PHP/5.4.45-ogc0Connection === closeContent-Type === text/htmlTCP received: 422 bytesonReadyToSendData: 1TcpConnection::staticOnReceive: pbuf is NULLTCP received: (null)time updated? 1>>Time      -----  09.03.2017 17:48:47
17:48:43.140, state: 5 -> 0 (0)rm 0TCP connection error: -8del if0usl 4021aa37 3ffee9ec
17:48:43.249, >>          -----  Deep Sleep
17:48:43.249, >>          -----  Station Close
17:48:43.249, >>SLEEP     -----  09.03.2017 16:48:47 tn 1489078127 t% 227 td 72000
17:48:43.249, 365 3ffffc30 40100be4
17:48:43.249, 3ffffb70:  0000000f 4020ec1c 3ffffbf0 0000000e
17:48:43.249, 3fffff70:  3ffed744 3fffdcc0 3ffeafd0 3ffeafd0
17:48:43.249, 3ffffonReadyToSendData: 0Written: 78, Available: 78, isFinished: 1, PushCount: 1TcpClient request completedTCP sent: 78onReadyToSendData: 2Header pos: 143Date === Thu, 09 Mar 2017 16:48:47 GMTServer === ApacheX-Powered-By === PHP/5.4.45-ogc0Connection === closeContent-Type === text/htmlTCP received: 422 bytesonReadyToSendData: 1TcpConnection::staticOnReceive: pbuf is NULLTCP received: (null)time updated? 1>>Time      -----  09.03.2017 17:48:47
[... cut ...]
17:48:44.123, >>          -----  wake up soft WDT
17:48:44.232, >>          -----  brak DS
17:48:45.105, >>          -----  AP DoConnect >> doNotConnect
17:48:45.105, >>          -----  FTP DoConnect >> doNotConnect
17:48:45.105, >>          -----  Start Tick AP >> 2 Station 5
17:48:45.105, >>          -----  Stationt connected with XXXXXXXX, channel 4dhcp client start...add if0Station configuration is: XXXXXXXXf r-40,
17:48:46.088, >>tick          -----  Tick AP 0 Station 5
17:48:47.071, >>tick          -----  Tick AP 99 Station 99

```

kwis2, 2017-03-09:
I have checked another board - WeMOS  D1 mini with the same results, so In my oppiniom it is no problem with power, connection or FLASH size. For now, In my oppinion http POST crash system.
Log below - ESP hangs and 19:20:10 I have pressed reset button.
I think I should simplify program to find the reason.

> 19:19:17.166, >>tick          -----  Tick AP 99 Station 99
> 19:19:17.385, >>          -----  networks try to connect
> 19:19:17.385, D: XXXXXXSSID: XXXXXXSSID: XXXXXXSSID: XXXXXXodebrano POST
> 19:19:17.385, XXXXXX  XXXXXXS PushCount: 10READ Template (2)StartVar ESP_HostWritten: 10, Available: 10, isFinished: 0, PushCount: 11READ Template (3)continue to plaint textfound var: curr_data, at 91 (2229) - 100, send size 90Written: 90, Available: 90, isFinished: 0, PushCount: 12READ Template (2)StartVar curr_dataWritten: 19, Available: 19, isFinished: 0, PushCount: 13READ Template (3)continue to plaint textplain template text pos: 2240, len: 445Written: 445, Available: 445, isFinished: 0, PushCount: 14
> 19:19:17.496, >>          -----  networks try to connect
> 19:19:17.496, extfound var: ESP_Host, at 44 (2129) - 52, send size 43Written: 43, Available: 43, isFinished: 0, PushCount: 10READ Template (2)StartVar ESP_HostWritten: 10, Available: 10, isFinished: 0, PushCount: 11READ Template (3)continue to plaint textfound var: curr_data, at 91 (2229) - 100, send size 90Written: 90, Available: 90, isFinished: 0, PushCount: 12READ Template (2)StartVar curr_dataWritten: 19, Available: 19, isFinished: 0, PushCount: 13READ Template (3)continue to plaint textplain template text pos: 2240, len: 445Written: 445, Available: 445, isFinished: 0, PushCount: 14
> 19:19:17.712, >>          -----  XXXXXX
> 19:19:17.712, >>          -----  XXXXXX
> 19:19:18.151, >>tick          -----  Tick AP 99 Station 99
> 19:19:19.134, >>tick          -----  Tick AP 99 Station 99
> 19:19:20.114, scandonescan failed 1state: 5 -> 0 (0)rm 0TCP connection error: -8~TCP connectionStation configuration is: XXXXXX Template (3)continue to plaint textfound var: ESP_MAC, at 43 (2026) - 50, send size 42Written: 42, Available: 42, isFinished: 0, PushCount: 6READ Template (2)StartVar ESP_MACWritten: 12, Available: 12, isFinished: 0, PushCount: 7READ Template (3)continue to plaint textfound var: ESP_IP, at 44 (2078) - 50, send size 43Written: 43, Available: 43, isFinished: 0, PushCount: 8READ Template (2)StartVar ESP_IPWritten: 12, Available: 12, isFinished: 0, PushCount: 9READ Template (3)continue to plaint t>>tick          -----  Tick AP 99 Station 99
> 19:19:20.882, onAccept state: 0 K=0Free heap size=30688, K=0+TCP connectiontimeout updating: 70 -> 90ished: 0, PushCount: 10READ Template (2)StartVar ESP_HostWritten: 10, Available: 10, isFinished: 0, PushCount: 11READ Template (3)continue to plaint textfound var: curr_data, at 91 (2229) - 100, send size 90Written: 90, Available: 90, isFinished: 0, PushCount: 12READ Template (2)StartVar curr_dataWritten: 19, Available: 19, isFinished: 0, PushCount: 13READ Template (3)continue to plaint textplain template text pos: 2240, len: 445Written: 445, Available: 445, isFinisheip:192.168.0.30,maskst cause:2, boot mode:(1,6)
> 19:19:20.882, onnect success XXXXXX
> 19:19:20.985,  ets Jan  8 2013,rst cause:2, boot mode:(1,6)
> 19:19:20.985, onAccept state: 0 K=0Free heap size=30688, K=0+TCP connectiontimeout updating: 70 -> 90ished: 0, PushCount: 10READ Template (2)StartVar ESP_HostWritten: 10, Available: 10, isFinished: 0, PushCount: 11READ Template (3)continue to plaint textfound var: curr_data, at 91 (2229) - 100, send size 90Written: 90, Available: 90, isFinished: 0, PushCount: 12READ Template (2)StartVar curr_dataWritten: 19, Available: 19, isFinished: 0, PushCount: 13READ Template (3)continue to plaint textplain template text pos: 2240, len: 445Written: 445, Available: 445, isFinishe
> 19:19:29.286,  ets Jan  8 2013,rst cause:4, boot mode:(1,6)
> 19:19:29.286, state: 5 -> 0 (0)rm 0del if0bcn 0del if1usl
> 19:19:29.286, wdt reset
> 19:20:10.039, $boâ|ŚělcŚňggď<0>d„Źl`g'd„d`gs‡““nűl‡d`ónźŕc'd„Śűggď<0>Źlpűoŕs„śśâŕboâ|lŹb„űooď<0>lŚd`nodŚl`'sŽ››oěŽl`űožŕbo$ŚŚógnç<0>Źlxňgŕ{Śśśăŕěcgă|„äŹpŚňggç<0>d„l`ogÇ$o{Ź›’gd`r’“gl`don't use rtc mem data
> 19:20:10.366, >>          -----  reset
> 19:20:10.366, t:0x4d000mount res: 0
> 19:20:10.585, >>          -----  reset 2
> 19:20:10.585, >>          -----  wake up normal
> 19:20:10.585, sd§timeout updating: 70 -> 65535Server timeout updating: 40 -> 90timeout updating: 70 -> 65535Server timeout updating: 40 -> 900fs.start: size:188 Kb, offse
> 19:20:10.694, >>          -----  brak DS
> 19:20:11.567, {d<0>dśź|<0>Ślŕ|„lěc|‡ä›;’cŚb„űooź    -----  wake up norma
> 19:20:11.567, >>          -------  Tick AP 20 Station 20
> 19:20:11.677, f r0,
> 19:20:11.786, === WEB SERVER STARTED ===
> 19:20:11.786, ==============================
> 19:20:11.786, 0.0.0.0
> 19:20:11.786, bcn 0del if1add if1bcn 100AP configuration was updatedbcn 0del if1mode : sta(a0:2admina:f0:67)mode : sta(a0:20:a6:0a:f0:67) + softAP(a2:20:a6:0a:f0:67)add if1dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)bcn 100IP can be changed only in init() method+TCP connection'/' registered'/index.html' registered'/about.html' registered'/config.html' registered
> 19:20:11.786, scandonef r0,
> 19:20:11.895, FTP DoConnect >> shouldBeConnected
[... cut ...] htmlTCP received: 422 bytesonReadyToSendData: 1TcpConnection::staticOnReceive: pbuf is NULLTCP received: (null)time updated? 1>>Time      -----  09.03.2017 19:20:17
> 19:20:18.556, TCP connection closing-TCP connection>>tick          -----  Tick AP 99 Station 99
> 19:20:19.539, TCP connection closing-TCP connectionen: 78, Available: 78, isFinished: 1, PushCount: 1TcpClient request completed>>tick          -----  Tick AP 99 Station 99
> 19:20:20.522, >>tick          -----  Tick AP 99 Station 99
>

kwis2, 2017-03-09:
I have prepared sample program which hangs all ESP modules I have. [sming-hang-up](https://github.com/kwis2/sming-hang-up.git). 
Sorry for codding style

kwis2, 2017-03-10:
I am wodnering if it is problem of sming or my computer configuration or old Espressif version I use.
I would be grtefull if you could send information if you are able to reproduce problem with your modules.

slaff, 2017-03-10:
> 19:19:29.286, wdt reset
19:20:10.039, ���$�b�oâ|�Śěl���cŚňggď<0>d„Źl����g'd„d���gs‡““n�űl‡�dón�źŕc�'d„�Śűggď<0>�Ź�lpűoŕ���s„śśâŕ�b�oâ|�lŹb„űooď<0>lŚ�d����nodŚl���'sŽ››o�ěŽ�lűožŕb�o$ŚŚógnç<0>�Ź�lxňgŕ����{Śśśăŕě���cgă|�„äŹ�pŚňggç<0>d„�l����ogÇ�$��o{Ź›’g��d��r’“g����l`�don't use rtc mem data

what I can see is that actually WDT reset was triggered. And this is what this ticket is about. Therefore I will close it. 

If there other issues feel free to open new ticket. And, please, one problem per issue. This way the discussion is better organized and more helpful to the others.

closed 2017-03-10 by slaff