SmingHub/Sming #1018: Q: If I want to add a lib should I always use submodule?

ADiea, 2017-03-09:
Hi
I would like to add at least 2 libs in the future one is https://github.com/WereCatf/PCF8574_ESP 
Should I make a new submodule in thirdparty or just put it in the lib folder?

slaff, 2017-03-09:
Good question. If this is an arduino library, which the one that you have pointed out is, then we can build also mechanisms for Sming/Libraries, similar to third-party, that fetches remote libraries and patches them in the same manner as third-party libraries. 

It would have been great if we could use these libraries out of the box, but that, as far as I can tell is impossible for most of them. Mainly because they expect the "setup - loop" flow, which we do not have, and also depend on Arduino core stuff.

ADiea, 2017-03-09:
On the other hand submodule makes sense only for medium-big source code projects like axtls esplwip etc, and most importantly actively maintained. If it's not maintained we complicate the flow for no good reason

slaff, 2017-03-09:
> If it's not maintained we complicate the flow for no good reason

We can have best of both worlds - submodules for actively maintained libraries and local copies of Libraries that are small or not maintained.  

And that should be relatively easy to implement. As far as I know, and I am absolute beginner here so correct me if I am wrong, an Arduino module have [some structure](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification).  They must have `library.properties` file in its root folder. And the source code could be under src/ or, for older modules, in the root folder of the module.

That brings me to another idea for the feature: It would be great if we have command line package manager that searches Arduino libraries and allows our users to install them on demand.  Of course not all of them. Only those that have been proven to be compatible with Sming.

slaff, 2017-03-09:
But back to your question. 

> Should I make a new submodule in thirdparty 

No, the Arduino libraries belong to `Sming/Library`

> or just put it in the lib folder?

Put it in the lib folder for now. Once we have submodules support for `Sming/Libraries` similar to `Sming/third-party` we can move the code to a submodule.

ADiea, 2017-03-09:
ok closing this for now. Good idea about libs and versioning. then anybody cann have what version suits him.

closed 2017-03-09 by ADiea