SmingHub/Sming #1019: Sming on board with Ethernet

icodk, 2017-03-10:
I am looking for an hardware platform with Ethernet on which I can use my Sming based firmware.
Sometimes Ethernet is preferable (stability, security, wifi misconcetopn, etc.). It would be nice to be able to use the (almost) same firmware but to communicate Ethernet instead of wifi.
May be en ESP with Ethernet shield.
Any input is welcome

ADiea, 2017-03-10:
[ENC28J60 over SPI](http://www.ebay.com/itm/1PCS-MiNi-ENC28J60-Ethernet-LAN-Network-Module-For-Arduino-SPI-AVR-PIC-LPC-/381447240602?hash=item58d006fb9a:g:A5YAAOSwT5tWKuQ~)
there is some code up to osi layer 2 then when you have the frames you must adapt a layer over our lwip specifically over the interface part

ADiea, 2017-03-10:
For ex you could bring this lib into sming https://github.com/renatoaloi/EtherEncLib
Then write some glue to pass received frames to lwip. You can also make a router then if you define a eth interface to lwip + the wifi interface of the chip :)
For ex you receive your frame from enc60 in MACReadRXBuffer() (I think you must also use interrupt pin of encj to know ehen a new frame has arrived) 
Then you pass this frame to lwip. You mus define and add a new lwip interface with the api in esp-open-lwip/lwip/api/netifapi.c and set those functions in the netif struct to point to the functions of your ENCJ chip

icodk, 2017-03-10:
sounds like piece of cake :-)

ADiea, 2017-03-10:
It's a bit of work and testing but in the end you can use the rest of smings tcp/ip features transparently i think :)

icodk, 2017-03-10:
@ADiea Thanks for your input.

closed 2017-03-10 by icodk