SmingHub/Sming #1021: undefined reference to `init()'

kwis2, 2017-03-10:
With wrong `MODULES` setting in makefile there is misleading error generated. 
Another question how could I set MODULES if I would like to use current dir?

> C:\rob>make
> AR out/build/app_app.a
> LD out/build/app.out
> c:/tools/sming/Sming/compiler/lib/\libsming.a(user_main.o):(.text.user_init+0x4)
> : undefined reference to 'init()'
> c:/tools/sming/Sming/compiler/lib/\libsming.a(user_main.o): In function 'user_in
> it':
> C:\tools\sming\Sming/appinit/user_main.cpp:70: undefined reference to `init()'
> collect2.exe: error: ld returned 1 exit status
> make: *** [out/build/app.out] Error 1

anakod, 2017-03-10:
Lloks like your MODULES defination is wrong anв didn't actually include program source files (where init should be declared). Sming itself didn't have that function, but call it from user app.

kwis2, 2017-03-10:
@anakod - I moved source files to different dir so I knew what happened. But if I get such error without reason knowledge I wouldn't be able to guess what was wrong. In my oppinion message should be changed

anakod, 2017-03-10:
But this is standard error from linker? If you have ideas how to overwrite it - we will apply your PR of course.

closed 2017-03-10 by kwis2