SmingHub/Sming #1022: ICACHE_FLASH_ATTR not on library functions Why

icodk, 2017-03-10:
In my straggle to get as much free heap as possible I 'decorate ' all my function  with the ICACHE_FLASH_ATTR. However looking at Sming library functions I can't find this attribute.
Is this correct observation or  am I missing something?

anakod, 2017-03-10:
No, in Sming all function by default going to flash. You can force function to iram location by "IRAM_ATTR" attribute.

icodk, 2017-03-10:
By no you mean that all Sming functions are loads on demand and can't optimized farther 
What about Arduino libraries that are not part of Sming ?
Example I use this library: https://github.com/miguelbalboa/rfid
How can I make this library functions load on demand and occupying as little RAM as possible ?

anakod, 2017-03-15:
If that library  included in your app it will work same way and didn't require ICACHE_FLASH_ATTR. Sming linker script have default config:
```

  .irom0.text : ALIGN(4)
  {
    _irom0_text_start = ABSOLUTE(.);
    *(.irom0.literal .irom.literal .irom.text.literal .irom0.text .irom.text .irom.debug.*)
	out/build/app_app.a:*(.literal .text .literal.* .text.* .stub .gnu.warning .gnu.linkonce.literal.* .gnu.linkonce.t.*.literal .gnu.linkonce.t.* .irom.debug.*)
  *libsming.a:*(.literal .text .literal.* .text.* .stub .gnu.warning .gnu.linkonce.literal.* .gnu.linkonce.t.*.literal .gnu.linkonce.t.* .irom.debug.*)
  *libsmingssl.a:*(.literal .text .literal.* .text.* .stub .gnu.warning .gnu.linkonce.literal.* .gnu.linkonce.t.*.literal .gnu.linkonce.t.* .irom.debug.*)
    _irom0_text_end = ABSOLUTE(.);
	_flash_code_end = ABSOLUTE(.);
  } >irom0_0_seg :irom0_0_phdr
```

So it's going to flash instead of Espressif SDK implementation (and I think Sming concept is more right way for most cases).

anakod, 2017-03-15:
If you want some code to be placed in RAM permanently then please use IRAM_ATTR.

slaff, 2017-03-16:
> How can I make this library functions load on demand and occupying as little RAM as possible ?

First you need to make it compile and play nicely with Sming. This will require the following steps
```
cd $SMING_HOME/Libraries
git clone https://github.com/miguelbalboa/rfid.git
cd rfid
cp -r src/*.cpp .
cp -r src/*.h .
make -C $SMING_HOME
```

You cannot expect to use an Arduino library out-of-the box. Most of the times you would need to modify the library a bit. For example all `Serial.print(F(...))` can be replaced with `debugf(...)`, etc.

We could eventually add better Arduino Library compatibility so that such libraries can be used out of the box. (and IMHO it will bring great value to Sming) But I don't expect this to happen any time soon. 

> In my straggle to get as much free heap as possible I 'decorate ' all my function with the ICACHE_FLASH_ATTR. 

Back to your original question - as anakod has already mentioned this you don't need to decorate functions at all. The compilation mechanism in Sming is taking care for this. Only in the cases where you need a function to be in IRAM then you can add IRAM_ATTR attribute.

icodk, 2017-03-16:
@slaff Thanks
I am using this library already by adding it to my project as a regular cpp source code in my project app folder. Will I gain any RAM (or any other advantage)  by compiling it as a Sming Lib?
Otherwise, as a Sming lib it will be automatically available to any project, but my main concern is RAM

slaff, 2017-03-16:
> I am using this library already by adding it to my project as a regular cpp source code in my project app folder

It would be great if you share the modified library source code that is working with Sming. And we can add it later to the list of supported libraries.

> Will I gain any RAM (or any other advantage) by compiling it as a Sming Lib?

In terms of RAM - I guess you won't gain any extra free RAM. 

> but my main concern is RAM

You should worry about RAM (I presume you mean the free heap here) only if you are having less than 5-6K of available heap.

icodk, 2017-03-16:
I once sent it to alonewolfx2 and I think there was once Basic_NFC example. Can't find it now.
I can zip the cpp and h files and send it to you or what ?

slaff, 2017-03-16:
> I can zip the cpp and h files and send it to you or what ?

It would be better if you make PR. In the PR you can add the Basic_NFC example and the cpp and h files.

closed 2017-04-08 by slaff