SmingHub/Sming #1023: Framework seems to be broken

nevers, 2017-03-11:
I upgraded (my docker environment) to the latest 3.1.2 version of Sming and my project seems to be  broken. So then I tried running the Basic_Serial example: 
- compiling works fine
- flashing works fine too (although it seems to be quite a more recent version of the esptool).

But: when connecting the (serial) terminal, the device doesn't respond at all. Downgrading back to version 3.0.2 makes everything work fine again.

Any idea what happened? 
I see some changes/discussions on the esptool, is this related or am I missing something trivial here?

ADiea, 2017-03-11:
As usual please also post all relevant console output :)

nevers, 2017-03-11:
Here you go:

```
# docker exec -ti esplight_sming-ide_1 env TERM=xterm bash
# cd /workspace/Sming/samples/Basic_Serial
# make clean && make && make flash
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app.out

#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8390|     912
    rodata|           ReadOnly Data (RAM)|    3FFE8390|    3FFE93A8|    4120
       bss|      Uninitialized Data (RAM)|    3FFE93A8|    3FFEF7A8|   25600
      text|            Cached Code (IRAM)|    40100000|    40106A9C|   27292
irom0_text|           Uncached Code (SPI)|    4020A000|    4023AB60|  199520
Total Used RAM : 30632
Free RAM : 51288
Free IRam : 5494
------------------------------------------------------------------------------
# Generating image...
Generate firmware images successully in folder out/firmware.
Done
(!) Spiffs support disabled. Remove 'DISABLE_SPIFFS' make argument to enable spiffs.
(!) Spiffs support disabled. Remove 'DISABLE_SPIFFS' make argument to enable spiffs.
Killing Terminal to free /dev/ttyUSB0
/workspace/Sming/Sming/Makefile-project.mk:452: recipe for target 'flash' failed
make: [flash] Killed (ignored)
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm qio -fs 4m 0x00000 out/firmware/0x00000.bin 0x0a000 out/firmware/0x0a000.bin
WARNING: Flash size arguments in megabits like '4m' are deprecated.
Please use the equivalent size '512KB'.
Megabit arguments may be removed in a future release.
esptool.py v2.0-beta1
Connecting.......................
Detecting chip type... ESP8266
Uploading stub...
Running stub...
Stub running...
Attaching SPI flash...
Configuring flash size...
Flash params set to 0x0000
Compressed 32368 bytes to 22979...
Wrote 32368 bytes (22979 compressed) at 0x00000000 in 2.0 seconds (effective 127.0 kbit/s)...
Hash of data verified.
Compressed 199520 bytes to 146541...
Wrote 199520 bytes (146541 compressed) at 0x0000a000 in 13.0 seconds (effective 123.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 --dtr=0
--- forcing DTR inactive
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
```

Basically, I'm not receiving anything. In order to reproduce this, try building the latest docker image, and them try building and flashing the Basic_Serial sample within.

If you need anything please let me know.

slaff, 2017-03-12:
@nevers Can you type `make flash V=1` and paste the output. Also what is the ESP device that you use and what is the SPI_MODE that it works with?

nevers, 2017-03-13:
@slaff 
I'm running the Basic_Serial sample and I didn't alter any of the Makefiles. Therefore, the SPI_MODE variable was disabled by default, and judging from the previous CLI output I posted, it then uses the 'qio' argument when flashing. When enabling it with the default argument 'SPI_MODE = dio' (in Makefile-user.mk) and flashing again (both with and without the 'V=1' argument) I still get no response. Please refer to the output below. Yet, when I downgrade back to v3.0.1 it all works fine. 

I'm using the ESP-12 module: http://www.esp8266.com/wiki/lib/exe/detail.php?id=esp8266-module-family&media=12-01.jpg
```
# make flash V=1
pkill -9 -f "/dev/ttyUSB0 115200" || exit 0
/workspace/Sming/Sming/Makefile-project.mk:452: recipe for target 'flash' failed
make: [flash] Killed (ignored)
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm dio -fs 4m 0x00000 out/firmware/0x00000.bin 0x0a000 out/firmware/0x0a000.bin
WARNING: Flash size arguments in megabits like '4m' are deprecated.
Please use the equivalent size '512KB'.
Megabit arguments may be removed in a future release.
esptool.py v2.0-beta1
Connecting....
Detecting chip type... ESP8266
Uploading stub...
Running stub...
Stub running...
Attaching SPI flash...
Configuring flash size...
Flash params set to 0x0200
Compressed 32368 bytes to 22979...
Wrote 32368 bytes (22979 compressed) at 0x00000000 in 2.0 seconds (effective 126.5 kbit/s)...
Hash of data verified.
Compressed 199520 bytes to 146541...
Wrote 199520 bytes (146541 compressed) at 0x0000a000 in 13.0 seconds (effective 122.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 --dtr=0
--- forcing DTR inactive
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
```

slaff, 2017-03-15:
@nevers I wasn't able to reproduce your problem. I am getting the following results

```
bad5a3219e2f:/opt/esp-open-sdk# git clone https://github.com/espressif/esptool.git 
Cloning into 'esptool'...
remote: Counting objects: 1172, done.
remote: Total 1172 (delta 0), reused 0 (delta 0), pack-reused 1172
Receiving objects: 100% (1172/1172), 4.52 MiB | 825.00 KiB/s, done.
Resolving deltas: 100% (698/698), done.
Checking connectivity... done.
bad5a3219e2f:/opt/esp-open-sdk# cd $SMING_HOME/../samples/Basic_Serial/
bad5a3219e2f:/workspace/Sming/samples/Basic_Serial# make flash
(!) Spiffs support disabled. Remove 'DISABLE_SPIFFS' make argument to enable spiffs.
Killing Terminal to free /dev/ttyUSB0
/workspace/Sming/Sming/Makefile-project.mk:452: recipe for target 'flash' failed
make: [flash] Killed (ignored)
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm dio -fs 32m 0x00000 out/firmware/0x00000.bin 0x0a000 out/firmware/0x0a000.bin 
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.0-beta2
Connecting....
Detecting chip type... ESP8266
Uploading stub...
Running stub...
Stub running...
Attaching SPI flash...
Configuring flash size...
Flash params set to 0x0240
Compressed 32368 bytes to 22982...
Wrote 32368 bytes (22982 compressed) at 0x00000000 in 2.0 seconds (effective 127.7 kbit/s)...
Hash of data verified.
Compressed 199520 bytes to 146531...
Wrote 199520 bytes (146531 compressed) at 0x0000a000 in 12.9 seconds (effective 123.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
�␒␒nnl�l`␂␎␂nr���n␌␌rbl`␂`�n␌␌␌��b�nl�␌��nn�␀␌�␎lp�n�␐␂␌␌r�����␌␌␌␌b␌n�|␂ll�␌b��nn�␀l�␌l`␂�␒␒nn␌�␎l␎␂nr���n␌␌��`␂p�n�␐␂␌␌r������l␌␌b␌n�|␂␌�p�␌␌b��nn�␀␌�l`␂�␒␒nn␌�␎l␎␂nr���n␌␌r�l`␂␎r��n␌␌r�l`␂��␂�␒�n�r��n|�␌l␌lll`␂��r�l��(INFO) message printed on UART0
(WARNING) message printed on UART0
(ERROR) message printed on UART0
(INFO) message printed with debugf on UART0
(DEBUG) print me again on UART0

== PRINTF TEST START ==

Float numbers display test: 
Pi with 2 decimals: 3.14 and with 4 decimals: 3.1416 
Pi without specifying precision(default 9): 3.14159265
...
```

Comparing the outputs I see that you have the following command, which differs from mine:

python -m serial.tools.miniterm /dev/ttyUSB0 115200 **--dtr=0**

Where is that --dtr=0 parameter coming from? We do not have it in our code? Can you remove that parameter and try again?

nevers, 2017-03-19:
@slaff 
Thanks for looking into this, this is really annoying.

The dtr param is set from COM_OPTS in my makefile. It is necessary for me because it takes care of auto-resetting the ESP board so I don't have to manually press a button to get it into flash-mode. I have custom hardware that does this for me and the DTR pin of the serial device is driving it.
This is my setup:
https://github.com/nevers/esp-open-sdk-docker/blob/master/schematics/esp8266.png

Maybe my reset hardware is to blame for this? 
* Do you press a button for flashing or do you also use a circuit to automatically get it into flash-mode like me?
* Could you point me the schematics of how you connect your serial device to the chip?

slaff, 2017-04-24:
Closing due to inactivity. If you have more details, feel free to reopen the issue.

closed 2017-04-24 by slaff