SmingHub/Sming #1026: Internal pull-up does not seem to work

riban-bw, 2017-03-11:
Configuring a GPIO pin with mode INPUT_PULLUP does not seem to enable the internal pull-up. A simple push button application (with appropriate switch debounce) does not work without external pull-up resistors added.

neut2012, 2017-03-12:
which pins are you trying to set the INPUT_PULLUP on?

riban-bw, 2017-03-12:
I have tried on several pins including GPIO4, 5 & 15 in my current project. Each of these have just a switch attached, pulling the pin to 0V when pressed. If I don't add an external pull-up resistor to each, digitalRead remains LOW.

neut2012, 2017-03-12:
hold on let me try it and get back to you

neut2012, 2017-03-12:
Please flash this code and tell me if it works:

```
#include <user_config.h>
#include <SmingCore/SmingCore.h>

#define LED_PIN 2 // GPIO2

Timer procTimer;
bool state = true;

void blink() 
{
   digitalWrite(LED_PIN, state);
   state = !state;

   pinMode(12, state ? INPUT_PULLUP : OUTPUT);
}

void init() 
{
   procTimer.initializeMs(1000, blink).start();
}
```

Measure the voltage on pin 12, it should toggle between High and Low every second. It works on my ESP12.

riban-bw, 2017-03-13:
That code does toggle the pin voltage between 0V and 3.3V as expected. In fact, I have just done a very simple digitalRead test and it does work as expected so this issue is invalid. What is not working is pull-up in combination with attachInterrupt. I will close this issue and report as a new issue. Thanks for your time and sorry for not testing this more thoroughly before reporting.

closed 2017-03-13 by riban-bw