SmingHub/Sming #1027: Discussion: add setjmp / longjmp to sming libraries?

tius2000, 2017-03-12:
I recently implemented the missing SDK functions _setjmp_ and _longjmp_ for one of my projects. Perhaps it might be useful to add them to the sming libraries, what do you think? What would be the right place for that?

hreintke, 2017-03-12:
@tius2000 : 
Looked in your github repositories but did not find your implementation.
Do you have them somewhere for usage/download ?

tius2000, 2017-03-12:
```
/*
	setjmp.S
	
	replacement for the closed-source version (Call0 ABI) included with the 
	ESP8266 SDK, see also include/machine/setjmp.h 
	
	(tius2000)
*/

	.text
	
//=============================================================================
//	int _setjmp(jmp_buf context)
	
	.align	4
	.literal_position
	.global	setjmp
	.type	setjmp, @function

setjmp:	
	s32i.n  a0, a2, 0	
	s32i.n  a1, a2, 4
	s32i.n  a12, a2, 8
	s32i.n  a13, a2, 12
	s32i.n  a14, a2, 16
	s32i.n  a15, a2, 20
	
	movi.n  a2, 0
	
	ret.n
	
	.size	setjmp, . - setjmp
	

//=============================================================================
//	void _longjmp(jmp_buf context, int val)
	
	.align	4
	.literal_position
	.global	longjmp
	.type	longjmp, @function
	
longjmp:
	l32i.n  a0, a2, 0
	l32i.n  a12, a2, 8
	l32i.n  a13, a2, 12
	l32i.n  a14, a2, 16
	l32i.n  a15, a2, 20
	l32i.n  a1, a2, 4
	
	//	if (!val) val = 1
	movi.n  a2, 1
	movnez  a2, a3, a3
	
	ret.n
	
	.size	longjmp, .-longjmp
	
```

tius2000, 2017-03-12:
```
/*  setjmp/application.cpp
    
    $Author: Matthias.waldorf $
    $Date: 8.02.17 15:34 $
    $Revision: 23 $ 
  
    to do:
        
    remarks:
        
    history:
        0.00    initial version
*/       
 
#include "serial.h"
#include <setjmp.h>

#define LOG_LEVEL           9
#define LOG_PREFIX          "main: "

//-----------------------------------------------------------------------------
static void _test1();
static void _test2();

//=============================================================================
extern "C" uint32 __attribute__((weak)) user_rf_cal_sector_set(void) {
    return RF_CAL_ADDR >> 12;    
}    

//-----------------------------------------------------------------------------
void init() {
    serial_init();
    wifi_set_opmode(0);                     // no wifi required
    
    INFO("starting tests ...\n");
    _test1();
    _test2();
}

//=============================================================================
//  test 1 
//
//  should print:
//      * second
//      * main
//
//-----------------------------------------------------------------------------
static jmp_buf _context_test1;

static void _first();
static void _second();

//-----------------------------------------------------------------------------
static void _test1() {   
    if ( !setjmp( _context_test1 ) ) {
        //  when executed, setjmp returned 0
        _first(); 
        //  will never return
    }
    else {                        
        //  when longjmp jumps back, setjmp returns 1
        DBG("* main");              // prints
    }
    INFO("test 1 done.\n");
}

static void _first() {
    _second();                          // does not return
    DBG("* first");                 // does not print
}

static void _second() {
    DBG("* second");                // prints
    
    //  jumps back to where setjmp was called - making setjmp now returns 1
    longjmp( _context_test1, 1);             
}

//=============================================================================
//  test 2 
//
//  should print:
//      * child loop begin
//      * parent
//      * child loop end
//      * parent
//      * child loop begin
//      * parent
//      * child loop end
//
//-----------------------------------------------------------------------------
static jmp_buf _context_test2, _context_child;

static void _call_with_cushion();
static void _child();

//-----------------------------------------------------------------------------
static void _test2() {
    if ( !setjmp(_context_test2) ) {
        _call_with_cushion(); 
        //  child never returns 
    } 
    
    //  execution resumes here after first time that child yields 
    for (int i=0; i<3; i++) {
        DBG("* parent");
        if ( !setjmp(_context_test2) ) {
            //  yield - note that this is undefined under C99 
            longjmp(_context_child, 1); 
        }
    }
    INFO("test 2 done.\n");
}

static void _call_with_cushion() {
    //  reserve stack space for _test2() and prevent optimization
    char space[1000]; 
    space[999] = 1; 
    
    _child();
}

static void _child() {
    while (1) {
        DBG("* child loop begin");
        
        //  yield (invalidates _context_child in C99)
        if ( !setjmp(_context_child) ) longjmp(_context_test2, 1); 

        DBG("* child loop end");

        //  yield (invalidates _context_child in C99)
        if (!setjmp(_context_child)) longjmp(_context_test2, 1); 
    }
}

```

tius2000, 2017-03-12:
The sample application just contains two snippets from Wikipedia.

hreintke, 2017-03-12:
Thanks,
Included the include/machine/setjmp.h and added 
extern "C"{
void	_EXFUN(longjmp,(jmp_buf __jmpb, int __retval));
int	_EXFUN(setjmp,(jmp_buf __jmpb));
}

First tests show good results.

slaff, 2017-03-12:
>  I recently implemented the missing SDK functions setjmp and longjmp for one of my projects.

Why not use the ones that come with the compiler ?
For example in esp-open-sdk there is `${XTENSA_TOOLS_ROOT}/../xtensa-lx106-elf/sysroot/lib/libcirom.a` that contains  `lib_a-setjmp.o`.

tius2000, 2017-03-12:
I prefered to have an assembler file instead of an object file in my project. The code should be identical (see also [newlib-esp32 source code](https://github.com/espressif/newlib-esp32/blob/master/newlib/libc/machine/xtensa/setjmp.S)).

slaff, 2017-04-24:
@tius2000 setjmp and longjmp are more appropriate for the SDK itself. Can you create a PR in the [esp-open-sdk](https://github.com/pfalcon/esp-open-sdk) ?

tius2000, 2017-04-24:
I'm using the UDK, so someone using the esp-open-sdk needs to do this. But this would exclude windows users, wouldn't it?

slaff, 2017-04-24:
Ok, then add it as PR to Sming but make sure that these functions are included in the Sming library on-demand (via compiler directive).

slaff, 2020-01-02:
Should be fixed in commit  https://github.com/SmingHub/Sming/commit/b2e0911e9d7ae5dfe5ea8ca5361f3382603226a4

closed 2020-02-01 by slaff