SmingHub/Sming #1028: ARDUINOJSON_USE_LONG_LONG does not work in Sming

riban-bw, 2017-03-12:
Adding `#define ARDUINOJSON_USE_LONG_LONG` to the top of a Sming application source code instructs the Arduino JSON library to allow use of long long (64 bit) integers but this causes lots of compile errors including ambiguous overloads (probably because overload functions are not defined for this variant) and strtoll missing.

I want to store and retrieve the 64 bit ID of DS18S20 temperature sensors in a configuration file but this is proving challenging in Sming!

anakod, 2017-03-13:
Of course it's hack, but may be store it as String? :)

riban-bw, 2017-03-13:
@anakod I have converted to 16 character string representation of hex bytes to get my project working. It would be good to have a way to handle 64-bit numbers.

anakod, 2017-03-13:
1. strtoll should be fixed first
2. we need to check new version of ArduinoJson, may be it will resolve issue

slaff, 2019-07-04:
@riban-bw In latest develop we use ArduinoJson v6. If your issue is still present there feel free to reopen this ticket.

closed 2019-07-04 by slaff