SmingHub/Sming #1029: Sming Basic_SSL sample not working. FATAL ERROR

neut2012, 2017-03-12:
I'm getting this error when trying out the basic_ssl example in the samples folder.

I recently (yesterday) updated my Sming environment (did a clean install of everything) so everything should be up to date.

The test is done on an ESP8266 with 4M flash. All other samples work fine.

Things I tried (just in case):
Removed the sections of code that are switching CPU frequency.
Tried on other ESP modules.
Removed the SSL section code in the sample: it's runs with no fatal errors (but fails to do SSL handshaking obviously).

Please help. I'm behind schedule :)

Here's the startup log:

Ready for SSL tests
Station configuration is: Home
mode : sta(60:01:94:07:53:97) + softAP(62:01:94:07:53:97)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with Home, channel 5
dhcp client start...
ip:192.168.0.102,mask:255.255.255.0,gw:192.168.0.1
Connected. Got IP: 192.168.0.102
Download: https://www.google.com:443/
+TCP connection
connect to: www.google.com
Storing 38 bytes in stream
realloc 38 -> 169
Storing 2 bytes in stream
Storing 0 bytes in stream
DNS record found: www.google.com = 172.217.23.100
TcpConnection::connect result:, 0
OnConnected
staticOnConnected: useSSL: 1, Error: 0
SSL: Starting connection...
SSL: Switching to 160 MHz
SSL: handshake start (3083 ms)
please start sntp first !
SSL: handshake is in progress...
TCP sent: 101
please start sntp first !

***** Fatal exception 28
pc=0x40100e96 sp=0x3ffff640 excvaddr=0x000015b8
ps=0x00000033 sar=0x00000018 vpri=0x000000f0
r00: 0x40100d69=1074793833 r01: 0x3ffff640=1073739328 r02: 0x3fff3900=1073690880 
r03: 0x2b265ed4= 723934932 r04: 0x3fffbff0=1073725424 r05: 0x000015b8=      5560 
r06: 0x000015b8=      5560 r07: 0x14a8433b= 346571579 r08: 0x00000008=         8 
r09: 0x14a842f3= 346571507 r10: 0x3fff22d0=1073685200 r11: 0x00000003=         3 
r12: 0x00000048=        72 r13: 0x3fff38c0=1073690816 r14: 0x3ffe9da0=1073651104 
r15: 0x3fff1138=1073680696 

Stack dump:
To decode the stack dump call from command line:
   python $SMING_HOME/../tools/decode-stacktrace.py out/build/app.out
and copy & paste the text enclosed in '===='.
================================================================
3ffff640:  40100e96 00000033 00000018 000000f0  
3ffff650:  40100d69 3fff3900 2b265ed4 3fffbff0  
[... 323 lines cut ...]

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: GPIO skip mode (16)

Booting rom 0.
Œâì‚nì’r‚òn|ìllll`Œâr’lŒþReady for SSL tests
Station configuration is: Home
mode : sta(60:01:94:07:53:97) + softAP(62:01:94:07:53:97)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

And loops...

ADiea, 2017-03-12:
please decode stackdump

neut2012, 2017-03-12:
0x40100e96: pvPortRealloc at ??:                                                                                                                                                 ?
0x40100d69: pvPortMalloc at ??:?
0x402455f0: ax_port_malloc at C:\tools\Sming-3.1.2\Sming/third-party/axtls-8266/                                                                                                                                                 replacements/mem.c:51
0x4024561a: ax_port_calloc at C:\tools\Sming-3.1.2\Sming/third-party/axtls-8266/                                                                                                                                                 replacements/mem.c:63
0x40230acc: x509_new at c:\tools\Sming-3.1.2\Sming\third-party\axtls-8266/ssl/x5                                                                                                                                                 09.c:81
0x4023d3e4: m_vsnprintf at C:\tools\Sming-3.1.2\Sming/system/m_printf.cpp:69
0x4023d4a6: m_vprintf at C:\tools\Sming-3.1.2\Sming/system/m_printf.cpp:83
0x40000a39: ?? ??:0
0x4021418a: pp_attach at ??:?
0x40221d60: pbuf_alloc at ??:?
0x40221f0a: pbuf_copy at ??:?
0x4023d3e4: m_vsnprintf at C:\tools\Sming-3.1.2\Sming/system/m_printf.cpp:69
0x40225aaa: etharp_query at ??:?
0x40225a8b: etharp_query at ??:?
0x40225c19: etharp_output at ??:?
0x4023d4a6: m_vprintf at C:\tools\Sming-3.1.2\Sming/system/m_printf.cpp:83
0x40225520: ip_output_if_opt at ??:?
0x4023d4a6: m_vprintf at C:\tools\Sming-3.1.2\Sming/system/m_printf.cpp:83
0x40241f71: MemoryDataStream::write(unsigned char const*, unsigned int) at C:\to                                                                                                                                                 ols\Sming-3.1.2\Sming/SmingCore/../SmingCore/DataSourceStream.h:246
0x40001db4: ?? ??:0
0x40001f46: ?? ??:0
0x40101950: os_printf_plus at ??:?
0x40001da0: ?? ??:0
0x401041fe: pp_post at ??:?
0x40104ef5: trc_NeedRTS at ??:?
0x402455f0: ax_port_malloc at C:\tools\Sming-3.1.2\Sming/third-party/axtls-8266/                                                                                                                                                 replacements/mem.c:51
0x40245626: ax_port_calloc at C:\tools\Sming-3.1.2\Sming/third-party/axtls-8266/                                                                                                                                                 replacements/mem.c:65
0x4022ecee: process_certificate at c:\tools\Sming-3.1.2\Sming\third-party\axtls-                                                                                                                                                 8266/ssl/tls1.c:2135
0x40230299: do_clnt_handshake at c:\tools\Sming-3.1.2\Sming\third-party\axtls-82                                                                                                                                                 66/ssl/tls1_clnt.c:107
0x40241611: ax_port_read at C:\tools\Sming-3.1.2\Sming/third-party/axtls-8266/co                                                                                                                                                 mpat/lwipr_compat.c:76
0x4022fc66: do_handshake at c:\tools\Sming-3.1.2\Sming\third-party\axtls-8266/ss                                                                                                                                                 l/tls1.c:1576
 (inlined by) basic_read at c:\tools\Sming-3.1.2\Sming\third-party\axtls-8266/ss                                                                                                                                                 l/tls1.c:1452
0x4023d4a6: m_vprintf at C:\tools\Sming-3.1.2\Sming/system/m_printf.cpp:83
0x402231a2: tcp_write at ??:?
0x4021402f: pp_attach at ??:?
0x4021407e: pp_attach at ??:?
0x4022fde0: ssl_read at c:\tools\Sming-3.1.2\Sming\third-party\axtls-8266/ssl/tl                                                                                                                                                 s1.c:301
0x402194fe: ieee80211_output_pbuf at ??:?
0x401041fe: pp_post at ??:?
0x40241683: axl_ssl_read at C:\tools\Sming-3.1.2\Sming/third-party/axtls-8266/co                                                                                                                                                 mpat/lwipr_compat.c:76
0x40221c2c: pbuf_free at ??:?
0x4023d508: m_printf at C:\tools\Sming-3.1.2\Sming/system/m_printf.cpp:116
0x40241300: WDTClass::alive() at C:\tools\Sming-3.1.2\Sming/SmingCore/Platform/W                                                                                                                                                 DT.cpp:24
0x4023ffde: TcpConnection::staticOnReceive(void*, tcp_pcb*, pbuf*, signed char)                                                                                                                                                  at C:\tools\Sming-3.1.2\Sming/SmingCore/Network/TcpConnection.cpp:685
0x4021402f: pp_attach at ??:?
0x401041fe: pp_post at ??:?
0x40103ead: ppEnqueueRxq at ??:?
0x4021402f: pp_attach at ??:?
0x40103d96: ppEnqueueRxq at ??:?
0x40103b67: ppProcessTxQ at ??:?
0x401041fe: pp_post at ??:?
0x40103a6b: lmacRxDone at ??:?
0x401041fe: pp_post at ??:?
0x40104ef5: trc_NeedRTS at ??:?
0x40105080: trc_NeedRTS at ??:?
0x401054ae: wDev_ProcessFiq at ??:?
0x4000050c: ?? ??:0
0x40222502: tcp_seg_free at ??:?
0x402273f8: tcp_receive at tcp_in.c:?
0x40000f58: ?? ??:0
0x40227fa0: tcp_input at ??:?
0x40221d91: pbuf_alloc at ??:?
0x40225299: ip_input at ??:?
0x40225c95: ethernet_input at ??:?
0x402139b3: pp_tx_idle_timeout at ??:?
0x40213926: pp_tx_idle_timeout at ??:?
0x402072cb: ets_snprintf at ??:?

ADiea, 2017-03-12:
i would say you are out of RAM...

ADiea, 2017-03-12:
ENABLE_CUSTOM_HEAP=1

slaff, 2017-03-12:
```
0x40100e96: pvPortRealloc at ??:                                                                                                                                                 ?
0x40100d69: pvPortMalloc at ??:?
0x402455f0: ax_port_malloc at C:\tools\Sming-3.1.2\Sming/third-party/axtls-8266/                                                                                                                                                 replacements/mem.c:51
0x4024561a: ax_port_calloc at C:\tools\Sming-3.1.2\Sming/third-party/axtls-8266/                                                                                                                                                 replacements/mem.c:63
0x40230acc: x509_new at c:\tools\Sming-3.1.2\Sming\third-party\axtls-8266/ssl/x5
```

> i would say you are out of RAM... because realloc fails.
Exactly that is happening.

neut2012, 2017-03-12:
Thank you. But after compiling the framework and the sample with 

> make ENABLE_SSL=1 ENABLE_CUSTOM_HEAP=1

 I get this output @ 74880 baud:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 26756, room 16 
tail 4
chksum 0x30
load 0x3ffe8000, len 900, room 4 
tail 0
chksum 0xbe
load 0x3ffe8390, len 6916, room 8 
tail 12
chksum 0x9d
csum 0x9d
```

And with rBoot:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 1560, room 16 
tail 8
chksum 0x2d
load 0x3ffe8000, len 776, room 0 
tail 8
chksum 0x7f
csum 0x7f

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: GPIO skip mode (16)

Booting rom 0.
```
 And that's it.. nothing happens :(

I searched all files for a conflicting **-mforce-l32** flag with no luck. 

Can someone please try the Basic_ssl sample and tell me if they get similar results? I really need this to work. I've already wasted 2 full days on a sample which should've worked out of the box.

Thank you   @@:)

slaff, 2017-03-16:
> Can someone please try the Basic_ssl sample and tell me if they get similar results?

I tried it yesterday with my environment and inside a docker environment and things seem to work for me. The google servers usually use 4K SSL fragment sizes and therefore it is strange that you are getting out-of-memory segfaults. Can it be that you did some manual modifications to the library or samples source code? Can you follow these recommendations: https://github.com/SmingHub/Sming/wiki/Sample-Compilation-Issues and report your progress?

neut2012, 2017-03-16:
The page you referred is related to compilation errors. But my code is compiling fine. I haven't done any changes to any file. I did a fresh install of everything. The code runs but outputs the above error codes when processing SSL request.

However I get many warning log messages while compiling sming code. Is this normal?

For example:

```
make[1]: Entering directory `/C/tools/Sming/Sming'
make[2]: Entering directory `/C/tools/Sming/Sming/third-party/esp-open-lwip'
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -I/c/Espressif/ESP8266_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/core/def.o lwip/core/def.c
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -I/c/Espressif/ESP8266_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/core/dhcp.o lwip/core/dhcp.c
In file included from include/lwip/stats.h:37:0,
                 from lwip/core/dhcp.c:73:
lwip/core/dhcp.c: In function 'dhcp_cleanup':
include/lwip/mem.h:55:21: warning: implicit declaration of function 'vPortFree' [-Wimplicit-function-declaration]
 #define mem_free(s) vPortFree(s, "", 0)
                     ^
lwip/core/dhcp.c:629:5: note: in expansion of macro 'mem_free'
     mem_free(netif->dhcp);
     ^
lwip/core/dhcp.c: In function 'dhcp_start':
include/lwip/mem.h:58:23: warning: implicit declaration of function 'pvPortMalloc' [-Wimplicit-function-declaration]
 #define mem_malloc(s) pvPortMalloc(s, "", 0)
                       ^
lwip/core/dhcp.c:673:27: note: in expansion of macro 'mem_malloc'
     dhcp = (struct dhcp *)mem_malloc(sizeof(struct dhcp));
                           ^
lwip/core/dhcp.c: In function 'dhcp_bind':
lwip/core/dhcp.c:1059:3: warning: implicit declaration of function 'system_station_got_ip_set' [-Wimplicit-function-declaration]
   system_station_got_ip_set(&ip, &mask, &gw);
   ^
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -I/c/Espressif/ESP8266_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/core/dns.o lwip/core/dns.c
In file included from lwip/core/dns.c:80:0:
lwip/core/dns.c: In function 'dns_recv':
include/lwip/mem.h:67:28: warning: implicit declaration of function 'pvPortZalloc' [-Wimplicit-function-declaration]
 #define mem_zalloc(s)      pvPortZalloc(s, "", 0)
                            ^
include/lwip/mem.h:99:22: note: in expansion of macro 'mem_zalloc'
 #define os_zalloc(s) mem_zalloc((s))
                      ^
lwip/core/dns.c:746:38: note: in expansion of macro 'os_zalloc'
   u8_t* dns_payload_buffer = (u8_t* )os_zalloc(LWIP_MEM_ALIGN_BUFFER(DNS_MSG_SIZE));
                                      ^
include/lwip/mem.h:55:21: warning: implicit declaration of function 'vPortFree' [-Wimplicit-function-declaration]
 #define mem_free(s) vPortFree(s, "", 0)
                     ^
include/lwip/mem.h:102:20: note: in expansion of macro 'mem_free'
 #define os_free(p) mem_free((p))
                    ^
lwip/core/dns.c:856:3: note: in expansion of macro 'os_free'
   os_free(dns_payload_buffer);
   ^
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -I/c/Espressif/ESP8266_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/core/init.o lwip/core/init.c
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -I/c/Espressif/ESP8266_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/core/mem.o lwip/core/mem.c
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -I/c/Espressif/ESP8266_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/core/memp.o lwip/core/memp.c
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -I/c/Espressif/ESP8266_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/core/netif.o lwip/core/netif.c
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -I/c/Espressif/ESP8266_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/core/pbuf.o lwip/core/pbuf.c
In file included from include/lwip/stats.h:37:0,
                 from lwip/core/pbuf.c:66:
lwip/core/pbuf.c: In function 'pbuf_alloc':
include/lwip/mem.h:58:23: warning: implicit declaration of function 'pvPortMalloc' [-Wimplicit-function-declaration]
 #define mem_malloc(s) pvPortMalloc(s, "", 0)
                       ^
include/lwip/memp.h:87:31: note: in expansion of macro 'mem_malloc'
 #define memp_malloc(type)     mem_malloc(memp_sizes[type])
                               ^
lwip/core/pbuf.c:297:24: note: in expansion of macro 'memp_malloc'
     p = (struct pbuf *)memp_malloc(MEMP_PBUF_POOL);
                        ^
lwip/core/pbuf.c: In function 'pbuf_free':
include/lwip/mem.h:55:21: warning: implicit declaration of function 'vPortFree' [-Wimplicit-function-declaration]
 #define mem_free(s) vPortFree(s, "", 0)
                     ^
include/lwip/memp.h:88:31: note: in expansion of macro 'mem_free'
 #define memp_free(type, mem)  mem_free(mem)
                               ^
lwip/core/pbuf.c:742:9: note: in expansion of macro 'memp_free'
```

This is just part of the output. But the make process ends with no errors. And I can compile and run all samples.

slaff, 2017-03-16:
> And I can compile and run all samples.

I thought the title of this issue was "Sming Basic_SSL sample not working". Ok, can you close that issue or at least change the title to represent more closely the problem that you are having?

neut2012, 2017-03-16:
Man the problem is exactly what it says in the title. Basic_SSL is NOT WORKING. I didn't say it wasn't compiling.

Informatic, 2017-03-26:
`Basic_Ssl` seems to work perfectly fine for me on 7679480 running on Wemos D1 Mini (ESP12F). (after removing certificate hash pinning, and changing to current public key hash)

neut2012, 2017-03-26:
May you please tell me which SDK version are you using?

Informatic, 2017-03-26:
Sorry, here's my current SDK: `sdk -> ESP8266_NONOS_SDK_V1.5.4_16_05_20`

neut2012, 2017-03-26:
When you compile Sming (make ENABLE_SSL=1) do you see warning messages like I posted above?

Informatic, 2017-04-23:
Well, seems like SSL support is really buggy right now. Forgot to mention that, but i've been struggling with that, and it seems like some issue with memory allocation, but I couldn't figure it out. (well, it became quite a heisenbug, as enabling memory allocation debugging somehow made faults much more rare...)

closed 2018-11-19 by slaff