SmingHub/Sming #1032: INPUT_PULLUP does not work with attachInterrupt

riban-bw, 2017-03-13:
If you configure an GPIO with mode INPUT_PULLUP then attach an interrupt to the pin, the pull-up does not seem to work.

**Example**
Connect a push-to-make button between GPIO pin 4 and 0V and run this firmware:
```
#include <user_config.h>
#include <SmingCore/SmingCore.h>

Timer procTimer;
uint8_t nPin = 4;
uint32_t nCount = 0;

void showCount()
{
  Serial.println(nCount);
}

void onButton()
{
  ++nCount;
}

void init()
{
  Serial.begin(115200);
  pinMode(nPin, INPUT_PULLUP);
  procTimer.initializeMs(1000, showCount).start();
  attachInterrupt(nPin, onButton, GPIO_PIN_INTR_ANYEDGE);
}
```
Expected result: Each time the button is pressed or released the displayed count increases.
Actual result: Only the first press increases count.
Adding an external 4.7K resistor makes this work.

neut2012, 2017-03-13:
Hi again
Do this:
```
  attachInterrupt(nPin, onButton, GPIO_PIN_INTR_ANYEDGE);
  pinMode(nPin, INPUT_PULLUP);
```
Call pinMode **AFTER** attachInterrupt. I tried it. It works :)

riban-bw, 2017-03-13:
Yes, that work-around does work thanks. I will add to API docs but we should look to see if we can fix it also

anakod, 2017-03-13:
I think this is unobvious behavior and this issue can be fixed.

slaff, 2017-03-16:
@anakod Can you create a PR with the changes related to this issue?

softerra, 2017-06-16:
Hi all,
What do you think about fixing the issue in this way?
```
--- a/Sming/SmingCore/Interrupts.cpp
+++ b/Sming/SmingCore/Interrupts.cpp
@@ -51,7 +51,9 @@ void attachInterruptHandler(uint8_t pin, GPIO_INT_TYPE mode)
        _gpioInterruptsInitialied = true;
    }
.
-   pinMode(pin, INPUT);
+   if (!isInputPin(pin)) {
+       pinMode(pin, INPUT);
+   }
.
    gpio_pin_intr_state_set(GPIO_ID_PIN(pin), mode); // Enable GPIO pin interrupt
.
@@ -75,7 +77,9 @@ void interruptMode(uint8_t pin, GPIO_INT_TYPE type)
 {
    ETS_GPIO_INTR_DISABLE();
.
-   pinMode(pin, INPUT);
+   if (!isInputPin(pin)) {
+       pinMode(pin, INPUT);
+   }
.
    gpio_pin_intr_state_set(GPIO_ID_PIN(pin), type);
```
The pin will be configured as simple INPUT only if it has not been configured by user explicitly.
This neither require changing the interface nor break existing code.

frankdownunder, 2017-12-03:
Hi all,
It doesnt seem like this has been fixed yet?