SmingHub/Sming #1035: Better Arduino Library Compatability

slaff, 2017-03-16:
It would be great if we can use Arduino libraries that are already working with esp8266 directly in Sming. This means the following:

### Being able to compile the desired library without modifications to its source code. We will need "a compatibility layer" that requires:
- [ ] Defining functions that are present in Arduino using their Sming alternatives
- [ ] Adding constants that are global to Arduino to be present also here.

### Being able to install on-demand such libraries and their dependencies.
- [ ] We can use the `platformio` library package manager and adapt it to Sming so that libraries are saved under Sming/Libraries. Plus the package manager should show only esp8226/Arduino compatible packages or Sming compatible packages.

zgoda, 2017-04-10:
I would start with ensuring public API compatibility in base facilities like SPI, I2C, 1Wire. This way many libs that interface hw using standard protocols should be readily compatible.

avr39-ripe, 2017-04-10:
May be we should go further and port Sming on top of esp8266/Arduino core? I personally like Sming very much, especially for it event driven approach. As for me Sming uses esp8266 in much proper way than esp8266/Arduino. BUT.. Arduino have terribly bigger community and greater popularity so some parts of Arduino code is much more polished than what we have in Sming. Moreover, as I can see we port to Sming many parts of it already. So may be we shouldn't reinvent the wheel, but just use it?
Sming itself can be described as several components which together allow us to develop complete standalone app for esp8266:
- esp8266 HAL (digital io, spi, i2c, pwm, etc)
- spiffy fs
- application model (events/delegates/timers vs plain LOOP)
- some higher level WIFI classes (WifiStation/WifiAccessPoint)
- some app level protocols servers and clients (http/websocjets/mqtt etc)
- rBoot
- build system
- libraries ported to or written for Sming

May be we should use esp8266/Arduino as kind of BASE for esp8266 HAL and then port our good stuff on top of it? And instead of porting good features from greatly supported by their community Arduino we can directly  use it, push our fixes/additions/etc to upstream where applicable and SHARE effort between projects.
Do not blame me, I really like Sming, each time I look to esp8266/Arduino side I lack many of "teasty things" from Sming, but they also had strong base of low level code, that can make Sming better.

hreintke, 2017-04-10:
@avr39-ripe : 
I have been thinking along these lines too and think it is feasible. 
Already had a part of Sming (the network (tcp/http))  running on the "Arduino/ESP8266" environment.
I then considered Sming being a library itself on top of the Arduino,

avr39-ripe, 2017-04-10:
@hreintke GLAD to hear you again!!!
Can you share you work? You also said that you port some part of Sming on top of esp32/Arduino.. can you share this also?
I think we have many good things in Sming, better or just other than in Arduino, but foundation of Arduino, HAL to mention is what we should not reinvent. And as I said - code share between two projects is way to go..

slaff, 2017-07-21:
> Already had a part of Sming (the network (tcp/http)) running on the "Arduino/ESP8266" environment.

@hreintke  Can you share with us some code?

closed 2019-07-04 by slaff