SmingHub/Sming #1036: TCP client sends > 256 bytes

senilix, 2017-03-16:
Hi,

I have an application that sends HTTP requests larger than 256 bytes using the TCPClient class. The request is also received at the endpoint, but for some reason I do not receive the answer. If i send a message of only 254 bytes I get the response.

Any ideas as to what could cause this?

Thanks,
Mogens

senilix, 2017-03-18:
No one has any suggestions as what to check?

Basically what I do is simple. I have a instance of the TCPClient and if I do a client.send() with a message that is larger than 256 bytes my OnReceive() callback function is never called. The message I send is received by the endpoint, so for some reason the response is lost. And I can add that I use the same message to check this with the only difference than I add a bunch of "\r\n" to exceed the 256 byte boundary.

Also, I have tested the same message on another client not using Sming and then the response is received. So my best guess at the moment is that this issue is Sming related.

Thanks,
Mogens

slaff, 2017-05-04:
@senilix Can use the HttpClient class and test with it? Something like this should be good starting point for your code:

```C
int yourCallbackHandler(HttpConnection& connection, bool success)
{
    debugf("Got response code: %d", connection.getResponseCode());
    debugf("Success: %d", success);
    if(connection.getRequest()->method != HTTP_HEAD) {
        debugf("Got content starting with: %s", connection.getResponseString().substring(0, 50).c_str());
    }

    return 1;
}

HttpClient client;

void ready() 
{
      client.send(
           client.request("http://domain/path/")
           ->setMethod(HTTP_POST)
           ->setBody(<stream-string-or-raw-data>)
           ->onRequestComplete(yourCallbackHandler)
    );
}
```

For setBody you have the following possibilities // you should use only ONE at a time.

```C
HttpRequest* setBody(const String& body);
HttpRequest* setBody(IDataSourceStream *stream);
HttpRequest* setBody(uint8_t *rawData, size_t length);
```

You can also process the response body stream, if needed, using:
```
HttpRequest* setResponseStream(IOutputStream *stream);
```

slaff, 2017-07-30:
Closing due to inactivity. Feel free to reopen it if needed.

closed 2017-07-30 by slaff