SmingHub/Sming #1047: Disscussion/vote: remove waitConnection stuff from StationClass

avr39-ripe, 2017-03-22:
@all Please vote for removal of waitConnection stuff from StationClass (WifiStation) as it is replaced by WifiEvents. For me waitConnection never work reliable and predictable.

slaff, 2017-03-22:
@avr39-ripe Can you describe shortly the advantages and paste also how the Basic_Wifi example will look with WifiEvents so that also other developers can have a look at it.

avr39-ripe, 2017-03-23:
Look-n-feel of Basic_Wifi with Blackjack and Wh... with WifiEvents :)

Advantages - in short - it is PROPER way of attaching handlers to different WIfi-related events implemented inside esp-sdk it do not relay on additional timers etc. And in my env works fine for a long time.

``` C++
#include <user_config.h>
#include <SmingCore/SmingCore.h>

// If you want, you can define WiFi settings globally in Eclipse Environment Variables
#ifndef WIFI_SSID
	#define WIFI_SSID "PleaseEnterSSID" // Put you SSID and Password here
	#define WIFI_PWD "PleaseEnterPass"
#endif

#ifndef WIFIAP_SSID
	#define WIFIAP_SSID "WebApp" // Put you SSID and Password here
	#define WIFIAP_PWD "smingweb"
#endif

// Will be called when WiFi station network scan was completed
void listNetworks(bool succeeded, BssList list)
{
	if (!succeeded)
	{
		Serial.println("Failed to scan networks");
		return;
	}

	for (int i = 0; i < list.count(); i++)
	{
		Serial.print("\tWiFi: ");
		Serial.print(list[i].ssid);
		Serial.print(", ");
		Serial.print(list[i].getAuthorizationMethodName());
		if (list[i].hidden) Serial.print(" (hidden)");
		Serial.println();
	}
}

// Setup initial configuration for Station and SoftAP
void initialWifiConfig()
{
// Set DHCP hostname to WebAppXXXX where XXXX is last 4 digits of MAC address
	String macDigits =  WifiStation.getMAC().substring(8,12);
	macDigits.toUpperCase();
	WifiStation.setHostname("WebApp" + macDigits);

// One-time set own soft Access Point SSID and PASSWORD and save it into configuration area
// This part of code will run ONCE after application flash into the ESP
	if(WifiAccessPoint.getSSID() != WIFIAP_SSID)
	{
		WifiAccessPoint.config(WIFIAP_SSID, WIFIAP_PWD, AUTH_WPA2_PSK);
		WifiAccessPoint.enable(true, true);
	}
	else
		Serial.printf("AccessPoint already configured.\n");

// One-time set initial SSID and PASSWORD for Station mode and save it into configuration area
// This part of code will run ONCE after application flash into the ESP if there is no
// pre-configured SSID and PASSWORD found in configuration area. Later you can change
// Station SSID and PASSWORD from Web UI and they will NOT overwritten by this part of code
// WEB UI NOT COVERED BY THIS EXAMPLE, YOU MUST DEVELOP YOUR OWN

	if (WifiStation.getSSID().length() == 0)
	{
		WifiStation.config(WIFI_SSID, WIFI_PWD);
		WifiStation.enable(true, true);
		WifiAccessPoint.enable(false, true);
	}
	else
		Serial.printf("Station already configured.\n");
}

// Will be called when disconnected from AP or not connected to last known AP (ssid/password)
void STADisconnect(String ssid, uint8_t ssid_len, uint8_t bssid[6], uint8_t reason)
{
	debugf("DISCONNECT - SSID: %s, REASON: %d\n", ssid.c_str(), reason);

	if (!WifiAccessPoint.isEnabled())
	{
		debugf("Starting OWN AP");
		WifiStation.disconnect();
		WifiAccessPoint.enable(true);
		WifiStation.connect();
	}
}

// Will be called when WiFi station was connected to AP and got IP from it
void STAGotIP(IPAddress ip, IPAddress mask, IPAddress gateway)
{
	debugf("GOTIP - IP: %s, MASK: %s, GW: %s\n", ip.toString().c_str(),
																mask.toString().c_str(),
																gateway.toString().c_str());

	if (WifiAccessPoint.isEnabled())
	{
		debugf("Shutdown OWN AP");
		WifiAccessPoint.enable(false);
	}
	// Add commands to be executed after successfully connecting to AP and got IP from it
}

// Will be called when WiFi hardware and software initialization was finished
// And system initialization was completed
void ready()
{
	debugf("READY!");

	// If AP is enabled:
	debugf("AP. ip: %s mac: %s", WifiAccessPoint.getIP().toString().c_str(), WifiAccessPoint.getMAC().c_str());
}

void init()
{
	Serial.begin(SERIAL_BAUD_RATE);
	Serial.systemDebugOutput(true); // Allow debug print to serial
	Serial.println("Sming. Let's do smart things!");

	// Initially setup wifi
	initialWifiConfig();

	// Attach Wifi events handlers
	WifiEvents.onStationDisconnect(STADisconnect);
	WifiEvents.onStationGotIP(STAGotIP);

	// Set system ready callback method
	System.onReady(ready);

	// Print available access points
	WifiStation.startScan(listNetworks); // In Sming we can start network scan from init method without additional code
}
```

slaff, 2017-03-24:
@avr39-ripe The code above looks good. One more question though - if we are interested in failed connection attempts , alternative to WifiStation on connectFail, then what should be the replacement?

avr39-ripe, 2017-03-24:
@slaff whether station is disconnected or NOT CONNECTED to last known AP this
`` C++
WifiEvents.onStationDisconnect(STADisconnect);
``
is fired. So inside you can check reason and make further decision. In my code I fire own AP which gives user ability to reconfigure ssid/password to AP or temporary gain access to device, but device continuously tries to reconnect to AP.

slaff, 2017-03-24:
@avr39-ripe Sounds good. So what we could do are the following steps:
- [ ] Change the Basic_Wifi example to use events only
- [ ] Mark `WifiStation::waitConnection` as deprecated. 
- [ ] Replace the usage of  `WifiStation::waitConnection`  from all samples with `WifiEvents::onStationGotIP` and `WifiEvents::onStationDisconnect`
- [ ] Take advantage of new possibilities like - detecting the reason for disconnect - AUTH failure or AP not present, etc.

Shall we add something more?

avr39-ripe, 2017-03-24:
@slaff roadmap looks good :) WifiEvents are here for a long time, so we should push our users to actually use them and get additional advantages over waitConnection.

avr39-ripe, 2017-03-24:
@slaff @all one more question:
I think we should agree new basic wifi boilerplate, not only WifiEvents related. If you carefully look into current Basic_Wifi it encourage users to enable Station and/or AccessPoint every time application starts, as far as configure STA/AP ssid/password. It leads to unnecessary flash wear, and isn't proper way at all. In code above I provide such kind of boilerplate code. Comments and improvements are welcome! :)

slaff, 2017-03-28:
>  It leads to unnecessary flash wear,

@avr39-ripe  Not really. It the new SSID and password are the same as the ones stored in flash it won't erase the flash sectors and save the same data. That was how it worked the last time I was looking at the code. I hope it didn't change in the meantime.

zgoda, 2017-04-07:
One point is missing from @slaff list - documentation. At least for disconnect reasons, but documenting meaning of particular arguments in delegates would be welcome as well. Now it's just argument type that is available in WifiEvents.h.

slaff, 2017-04-10:
In PR #1071 waitConnect is deprecated and will be removed in the coming versions.

slaff, 2017-04-10:
The follow-up work from this PR #1072 will replace all occurrences of waitConnect with their WifiEvent alternatives.

closed 2017-04-10 by slaff