SmingHub/Sming #1048: Basic_ProgMem failure

zgoda, 2017-03-22:
This is on current HEAD:

```
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
tail 12
chksum 0x19
ho 0 tail 12 room 4
load 0x3ffe8000, len 900, room 12 
tail 8
chksum 0xf4
load 0x3ffe8390, len 2772, room 0 
tail 4
chksum 0xa1
csum 0xa1
��{�d�l␛�␄␄�d`␂��r�l�$␓�␄␌␄l ␃��{�l␄��␀�␌␄ld`␃{$�> Read bytes from PROGMEM:
1 2 3 4 5 6 7 8 
> Copy a string from PROGMEM:
assertEqualsString: Demo != Demo�� @p��?
Demo�� @p��?
assertEqualsString: DemoDemo != Demo�� @p��?Demo
assertEqualsString: Demo 1 != (...)
> Copy bytes from PROGMEM:
1 2 3 4 5 
> ESP8266EX Memory Layout:
> 0x3FFE8000 ~ 0x3FFFBFFF - User data RAM, 80kb. Available to applications.
> 0x40200000 ~ ...        - SPI Flash.
> demoRam array address: 0x3FFE89E9 is in the RAM
> demoPgm array address: 0x40234AEC is in the Flash
```

The code looks like a test case so to me if is clearly failing.

slaff, 2017-03-24:
@ADiea when you are back can you look at that issue?

ADiea, 2017-03-25:
Next week i plan to start sming-ing again. Will have a look

zgoda, 2017-04-06:
The change from #1062 has no impact on this issue.

ADiea, 2017-04-06:
@zgoda I think PR #1065 will fix this can you check please?

zgoda, 2017-04-06:
Okay, will try this and report back.

ADiea, 2017-04-06:
because stuff like 
assertEqualsString: Demo != Demo�� @p��?
could mean that zero character at the end was not present perhaps

zgoda, 2017-04-07:
With this fix:

```
Read bytes from PROGMEM:
1 2 3 4 5 6 7 8 
> Copy a string from PROGMEM:
Demo
assertEqualsString: Demo 1 != (...)
> Copy bytes from PROGMEM:
1 2 3 4 5 
> ESP8266EX Memory Layout:
> 0x3FFE8000 ~ 0x3FFFBFFF - User data RAM, 80kb. Available to applications.
> 0x40200000 ~ ...        - SPI Flash.
> demoRam array address: 0x3FFE89E9 is in the RAM
> demoPgm array address: 0x40234AEC is in the Flash
```

So `strcpy_P` test passes but `sprintf_P` is still broken.

tius2000, 2017-04-18:
Can you try #1077? It might fix the "(...)" issue.

slaff, 2019-07-04:
Feel free to re-open this ticket if your issue is still present in the latest `develop` code.

closed 2019-07-04 by slaff