SmingHub/Sming #1050: Memory corruption?

zgoda, 2017-03-24:
This is quite strange to me and I can't find any reasonable reason for what I see.

I have a code that allocates 17 byte `char` buffer (https://github.com/zgoda/sming_playground/blob/master/app/ui.cpp#L220) and any operation on this buffer causes firmware crash, eg. an attempt to write something to it (https://github.com/zgoda/sming_playground/blob/master/app/ui.cpp#L221) but even simple `memset(fmtBuf, 0, sizeof(fmtBuf))` causes crash. The same operation in other places works fine, eg https://github.com/zgoda/sming_playground/blob/master/app/ui.cpp#L241 and later. There is nothing special there. The code works fine if I do not allocate the buffer and just use string literal.

Decoded stack trace:

```
0x40237aa4: gdb_exception_handler at /opt/Sming/Sming/gdb/gdbstub.c:89
0x400005e1: ?? ??:0
0x4000df1b: ?? ??:0
0x4023871d: memcpy_P at /opt/Sming/Sming/Wiring/FakePgmSpace.cpp:10
0x40234aa8: __divsf3 at ??:?
0x40234aa8: __divsf3 at ??:?
0x40239218: delayMicroseconds(unsigned int) at /opt/Sming/Sming/SmingCore/Clock.cpp:29
0x40239218: delayMicroseconds(unsigned int) at /opt/Sming/Sming/SmingCore/Clock.cpp:29
0x4023be70: SoftI2cMaster::write(unsigned char) at /opt/Sming/Sming/Wiring/I2cMaster.cpp:141
0x40239218: delayMicroseconds(unsigned int) at /opt/Sming/Sming/SmingCore/Clock.cpp:29
0x4023be70: SoftI2cMaster::write(unsigned char) at /opt/Sming/Sming/Wiring/I2cMaster.cpp:141
0x4021ecfa: ieee80211_ht_updateparams at ??:?
0x40225330: cnx_update_bss_more at ??:?
0x40239218: delayMicroseconds(unsigned int) at /opt/Sming/Sming/SmingCore/Clock.cpp:29
0x40239218: delayMicroseconds(unsigned int) at /opt/Sming/Sming/SmingCore/Clock.cpp:29
0x4023be70: SoftI2cMaster::write(unsigned char) at /opt/Sming/Sming/Wiring/I2cMaster.cpp:141
0x40239218: delayMicroseconds(unsigned int) at /opt/Sming/Sming/SmingCore/Clock.cpp:29
0x4023be70: SoftI2cMaster::write(unsigned char) at /opt/Sming/Sming/Wiring/I2cMaster.cpp:141
0x40239218: delayMicroseconds(unsigned int) at /opt/Sming/Sming/SmingCore/Clock.cpp:29
0x4023bd1c: SoftI2cMaster::stop() at /opt/Sming/Sming/Wiring/I2cMaster.cpp:114
0x40100ad4: ets_timer_arm_new at ??:?
0x40238760: strcpy_P at /opt/Sming/Sming/Wiring/FakePgmSpace.cpp:37
0x40234aa8: __divsf3 at ??:?
0x4023a1e2: I2CIO::write(unsigned char) at /opt/Sming/Sming/Libraries/LiquidCrystal/I2CIO.cpp:81
0x4000050c: ?? ??:0
0x40236c84: UIManager::_showWifiScreen() at /home/jazg/work/sming_playground/app/ui.cpp:65
0x4023a089: LiquidCrystal_I2C::write4bits(unsigned char, unsigned char) at /opt/Sming/Sming/Libraries/LiquidCrystal/LiquidCrystal_I2C.cpp:283
0x40000f58: ?? ??:0
0x4023a0c4: LiquidCrystal_I2C::send(unsigned char, unsigned char) at /opt/Sming/Sming/Libraries/LiquidCrystal/LiquidCrystal_I2C.cpp:255
0x40239218: delayMicroseconds(unsigned int) at /opt/Sming/Sming/SmingCore/Clock.cpp:29
0x4023c999: LCD::command(unsigned char) at /opt/Sming/Sming/Libraries/LiquidCrystal/LCD.cpp:334
0x40105f06: Timer::stop() at /opt/Sming/Sming/SmingCore/Timer.cpp:99
0x40237166: UIManager::showScreen(screen_t) at /home/jazg/work/sming_playground/app/ui.cpp:313
0x40100ad4: ets_timer_arm_new at ??:?
0x40237224: UIManager::_mainScreenKeyPress(char) at /home/jazg/work/sming_playground/app/ui.cpp:113
0x402372a8: UIManager::keyPress(char) at /home/jazg/work/sming_playground/app/ui.cpp:332
0x4010083e: wdt_feed at ??:?
0x40236943: AppController::_processInput() at /home/jazg/work/sming_playground/app/appcontroller.cpp:33 (discriminator 1)
0x401008fa: ets_timer_setfn at ??:?
0x402374c5: MethodCaller<void (AppController::*)()>::invoke() at /opt/Sming/Sming/Wiring/../SmingCore/Delegate.h:80 (discriminator 3)
0x4020ab34: ets_timer_handler_isr at ??:?
0x40105ffc: Timer::processing(void*) at /opt/Sming/Sming/SmingCore/Timer.cpp:99
0x4020ab41: ets_timer_handler_isr at ??:?
0x4020ab86: ets_timer_handler_isr at ??:?
0x4020e86b: ets_snprintf at ??:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
```

Here's complete code if you want to test - it uses standard 1602 LCD with I2C adapter and 1x4 matrix keypad -> https://github.com/zgoda/sming_playground

slaff, 2017-03-24:
Well, if you want to store data into FLASH you should align it properly
```
char fmtBuf[LCD_COLS + 1];
strcpy_P(fmtBuf, SCR_WIFI_MI1);
```

should be

```
char PROGMEM fmtBuf[LCD_COLS + 1];
strcpy_P(fmtBuf, SCR_WIFI_MI1);
```

Different is the question if you really need to go over the effort of storing it into Flash. With that sample the phrase "premature optimization is evil" comes to my mind.

zgoda, 2017-03-24:
`SCR_WIFI_MI1` is stored in flash and has to be copied to `fmtBuf`.

This issue is completely not related to `PROGMEM` storage, because **at this particular place** simple
```
char buf[17];
memset(buf, 0, sizeof(buf));
```
crashes firmware.

zgoda, 2017-03-24:
But wait, there's more! Even trying to get address of the variable causes crash:
```
Serial.printf("ADDR: %p\r\n", (void * )&buf);
```

zgoda, 2017-03-24:
Turned on system debug on serial, exc dump:
```
***** Fatal exception 3
pc=0x4000df1b sp=0x3ffffd90 excvaddr=0x40234ab4
ps=0x00000030 sar=0x00000000 vpri=0x00000008
r00: 0x40238775=1076070261 r01: 0x3ffffd90=1073741200 r02: 0x3ffffed1=1073741521 
r03: 0x40234ab4=1076054708 r04: 0x0000000c=        12 r05: 0x3ffffed1=1073741521 
r06: 0x00000004=         4 r07: 0x00000000=         0 r08: 0x60000838=1610614840 
r09: 0x60000838=1610614840 r10: 0x00000008=         8 r11: 0x3ffef460=1073673312 
r12: 0x3ffffed1=1073741521 r13: 0x00000000=         0 r14: 0x40234ab4=1076054708 
r15: 0x60000600=1610614272 
```

AFAIK exc 3 is LoadStoreError and this should not happen when barely requesting memory address of a variable.

zgoda, 2017-03-24:
The code does not crash if I declare buffer as `char buf[20]`, both memset and getting memory address work. As soon as I change it to `char buf[17]` the firmware crashes. In the code I have many 17 byte buffers (this is 1602 LCD line length + 1) and it crashes only in this one place.

zgoda, 2017-03-25:
Changed code to use buffers as private instance variables and now firmware crashes at another point with the same cause -> https://github.com/zgoda/sming_playground/blob/master/app/ui.cpp#L134

And LoadStoreError suggests this is a problem with variable misalignment.

jcmvbkbc, 2017-04-03:
> LoadStoreError suggests this is a problem with variable misalignment.

It does not, actually. Misalignment causes exception 9, LoadStoreAlignment.
Your exception dump says excvaddr=0x40234ab4, that address is in the FLASH area. And it is properly aligned. Looks like you're trying to access FLASH when it's not mapped.

zgoda, 2017-04-04:
I'm trying to reproduce this behaviour in more isolated case.

zgoda, 2017-04-04:
Okay, more on this. The code below crashes firmware:
```
#include <SmingCore/SmingCore.h>

#include "user_config.h"

static const char LONG_STR[] PROGMEM = "0123456789012345";
static const char FORMAT[] PROGMEM = "format in flash: %s";

void init() {
	Serial.begin(SERIAL_BAUD_RATE);
	Serial.systemDebugOutput(true);
	Serial.println("start");
	char buf[17];
	strcpy_P(buf, LONG_STR);
	buf[16] = 0;
	/*
	Serial.printf("string from memory: %s\n", buf);
	Serial.printf("len: %d, in flash: %d\n", strlen(buf), strlen_P(LONG_STR));
	*/
	char lgBuf[81];
	strcpy_P(lgBuf, FORMAT);
	//Serial.printf(lgBuf, buf);
	//Serial.println();
}
```

Error dump:
```
***** Fatal exception 3
pc=0x4000df1b sp=0x3ffff930 excvaddr=0x40234a68
ps=0x00000030 sar=0x00000000 vpri=0x40006545
r00: 0x40236d21=1076063521 r01: 0x3ffff930=1073740080 r02: 0x3ffffab1=1073740465 
r03: 0x40234a68=1076054632 r04: 0x00000010=        16 r05: 0x3ffffab1=1073740465 
r06: 0x6000001c=1610612764 r07: 0xff000000= -16777216 r08: 0x2a007472= 704672882 
r09: 0x00000000=         0 r10: 0x00000000=         0 r11: 0x3fffc174=1073725812 
r12: 0x3ffffab1=1073740465 r13: 0x3ffeda34=1073666612 r14: 0x3ffef0f0=1073672432 
r15: 0x60000600=1610614272 
```

Decoded stack trace:
```
0x40236d21: memcpy_P at /opt/Sming/Sming/Wiring/FakePgmSpace.cpp:10
0x40234a68: __divsf3 at ??:?
0x40001da0: ?? ??:0
0x40001db4: ?? ??:0
0x40001f46: ?? ??:0
0x40211498: tx_atten_set_interp at ??:?
0x40240000: ?? ??:0
0x4020ec34: malloc at C:\DEV\ESP8266\esp_microc/malloc/malloc.c:10
0x40237af4: uart_start_isr(uart_*) at /opt/Sming/Sming/system/uart.cpp:449
0x40237d18: uart_init at /opt/Sming/Sming/system/uart.cpp:449
0x40238434: Print::write(unsigned char const*, unsigned int) at /opt/Sming/Sming/Wiring/Print.cpp:32
0x4023845c: Print::print(char) at /opt/Sming/Sming/Wiring/Print.cpp:42
0x40236d64: strcpy_P at /opt/Sming/Sming/Wiring/FakePgmSpace.cpp:37
0x40234a68: __divsf3 at ??:?
0x40236aa4: Print::println(char const*) at /opt/Sming/Sming/Wiring/Print.cpp:190
0x402142ea: phy_after_init_enrx at ??:?
0x40235d3b: init() at /home/jazg/work/sming_memerror/app/application.cpp:14
0x4020fe08: chip_v6_set_chan at ??:?
0x40236bfe: String::copy(char const*, unsigned int) at /opt/Sming/Sming/Wiring/WString.cpp:729
0x40105b38: String::String(char const*) at /opt/Sming/Sming/Wiring/WString.cpp:729
0x40105b38: String::String(char const*) at /opt/Sming/Sming/Wiring/WString.cpp:729
0x40236454: cpp_core_initialize at /opt/Sming/Sming/system/esp_cplusplus.cpp:11 (discriminator 1)
0x4020a278: wdt_init at ??:?
0x40234080: __divsf3 at ??:?
0x40235f57: user_init at /opt/Sming/Sming/appinit/user_main.cpp:70
0x4020a8b6: user_rf_pre_init at ??:?
0x4020a8b1: user_rf_pre_init at ??:?
0x40234080: __divsf3 at ??:?
0x4020a8b9: user_rf_pre_init at ??:?
0x40100734: wdt_feed at ??:?
0x400003e9: ?? ??:0
```

ADiea, 2017-04-04:
Hi I don't understand...
* using the exact code? because you have print lines commented out but they show up in stack
* are you sure you are extracting the stack correctly?  I see uart init after prints.
Are you usre the code you posted matches what is on target?

zgoda, 2017-04-04:
Yes, this is exact code I use to generate crash (whole `application.cpp` file).
[dump.txt](https://github.com/SmingHub/Sming/files/892531/dump.txt) contains complete terminal session.

ADiea, 2017-04-04:
Hi again
Based on what @jcmvbkbc said in previous post did some googling: (link1)[http://esp8266-re.foogod.com/wiki/Cache_Read_Enable] (link2)[https://richard.burtons.org/2015/06/12/esp8266-cache_read_enable/] 

It appears that fucntion Cache_Read_Enable is responsible for mapping a part of the SPI flash to adddress 0x40200000. 
In sming I don't see it called anywhere explicitly maybe it is called by sdk earlier...
Can  you try calling it in init() ?
I believe Cache_Read_Enable(0, 0, 1); is a valid call. Pleas elet us know if any effect.
No sure who is responsible for mapping the area in the first place.

zgoda, 2017-04-04:
I found this function only in linker script in SDK https://github.com/espressif/ESP8266_NONOS_SDK/blob/06b117d691cb2cd1a20d6d0f000b76cf65b2ad67/ld/eagle.rom.addr.v6.ld

How am I supposed to call it?

zgoda, 2017-04-04:
Okay, found it.

```
#include <SmingCore/SmingCore.h>

#include "user_config.h"

static const char LONG_STR[] PROGMEM = "0123456789012345";
static const char FORMAT[] PROGMEM = "format in flash: %s";

extern "C" void Cache_Read_Enable(uint32_t a, uint32_t b, uint32_t c);

void init() {
	Cache_Read_Enable(0, 0, 1);
	Serial.begin(SERIAL_BAUD_RATE);
	Serial.systemDebugOutput(true);
	Serial.println("start");
	char buf[17];
	strcpy_P(buf, LONG_STR);
	buf[16] = 0;
	/*
	Serial.printf("string from memory: %s\n", buf);
	Serial.printf("len: %d, in flash: %d\n", strlen(buf), strlen_P(LONG_STR));
	*/
	char lgBuf[81];
	strcpy_P(lgBuf, FORMAT);
	//Serial.printf(lgBuf, buf);
	//Serial.println();
}
```

But this does not change anything, firmware still crashes.

jcmvbkbc, 2017-04-04:
@zgoda could you please also share the elf file of your crashing program?

zgoda, 2017-04-05:
Here it is

[elf.zip](https://github.com/SmingHub/Sming/files/895368/elf.zip)

jcmvbkbc, 2017-04-05:
Ok. It's a bug in memcpy_P (which is called from strcpy_P):
https://github.com/SmingHub/Sming/blob/develop/Sming/Wiring/FakePgmSpace.cpp#L9 makes wrong assumption that aligned source address and length guarantee that memcpy uses 32-bit-wide accesses to the source buffer. Looking at the following disassembly of the memcpy from ROM I see that that's not the case, type of access also depends on the alignment of destination buffer:
```
4000df18:       e974b6          bltui   a4, 7, 4000df05 <memcmp+0x5d>
4000df1b:       000362          l8ui    a6, a3, 0
...

4000df48 <memcpy>:
4000df48:       205220          or      a5, a2, a2
4000df4b:       c9e207          bbsi    a2, 0, 4000df18 <memcmp+0x70>
```
See how it goes straight to l8ui a6, a3, 0 when a2 == 0x3ffffab1 and a4 == 0x00000010 (taken from your crash dump).
And FLASH area can only be read with aligned 32-bit-wide reads, resulting in the exception you see.

So, I believe it's memcpy_P bug. The condition at https://github.com/SmingHub/Sming/blob/develop/Sming/Wiring/FakePgmSpace.cpp#L9 should look like
```if (!((unsigned long)src_P & 3) && !((unsigned long)dest & 3)&& !(length & 3))```.

zgoda, 2017-04-06:
Thank you @jcmvbkbc - I changed this line to read exactly what you suggest but the firmware still crashes in the same way.

jcmvbkbc, 2017-04-06:
@zgoda got new crash dump and elf image?

zgoda, 2017-04-06:
I'm sorry, please disregard my previous comment. I forgot I had `SMING_HOME` set in Makefile and it took precedence over newly built test version. Now it looks OK, will do couple other tests and prepare a PR with this change.

slaff, 2017-04-06:
@jcmvbkbc Max, thanks a lot for your help. From time to time we have some strange issues and your support is facilitating us to understand better what is actually going on. 

> (Quoted from https://github.com/SmingHub/Sming/issues/977: .. are you sure you understand mforce-l32i completely? It forces all reads and writes to be aligned. This can be a big performance penality + code size.

@jcmvbkbc I wanted to ask you what are the disadvantages of using GCC with mforce-l32 flag compared to using  PGM imitation functions like memcpy_P or other *_P  ?  Can you elaborate a bit?

jcmvbkbc, 2017-04-06:
> what are the disadvantages of using GCC with mforce-l32 flag compared to using PGM imitation functions like memcpy_P or other *_P

Speed and code size: every byte-sized access is expanded into 6-command-long 18-bytes-long sequence. Sequences accessing adjacent bytes don't get merged. Perhaps this can be improved, I tried once but failed.
[Here](https://github.com/SuperHouse/esp-open-rtos/issues/11#issuecomment-111627480) a 3% program image size growth was reported with micropython built with this option.

The idea behind this option was source-level portability: if the only code that might need to read bytes from FLASH is the user code then with this option it doesn't need to be changed to work with FLASH. This is hard to guarantee though in the presence of ROM and binary blobs.

Also memcpy may be rewritten, so that it always does aligned 32-bit reads, making separate memcpy_P unnecessary.

ADiea, 2017-04-06:
Original topic: So this means that for optimizations we could decorate stack allocated destination buffers with __attribute__((aligned(4))) ?

mforce-l32 topic: So on the plus side we get rid of all the _P functions and also for all the debug strings will not have to generate so much code, this could balance the size of code but not sure about the speed which matters especially on aes and ssl stuff... We will have to test and see i guess

jcmvbkbc, 2017-04-06:
> So this means that for optimizations we could decorate stack allocated destination buffers with attribute((aligned(4))) ?

Yes, that would help with the current memcpy_P implementation.

> on the plus side we get rid of all the _P functions

Only if their replacements are compiled with the -mforce-l32. You can get rid of pgm_read_byte/pgm_read_word.

commit 3d48e7eb (2017-04-10): [Fixes #1050] check alignment of dest address when copying from flash to (#1062)

RAM. Fix thanks to @jcmvbkbc debugging.
closed 2017-04-10 by slaff