SmingHub/Sming #1051: Release 3.1.2 is broken?

unit-by, 2017-03-26:
Any sample from release 3.1.2 cause watchdog reset. 

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 26064, room 16
tail 0
chksum 0xd5
load 0x3ffe8000, len 896, room 8
tail 8
chksum 0xfc
load 0x3ffe8380, len 1484, room 0
tail 12
chksum 0x20
csum 0x20
rf_cal[0] !=0x05,is 0xFF
rf cal sector: 251
rf[112] mode : softAP(5e:cf:7f:83:6a:cb)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)

***** Fatal exception 28
pc=0x40101c26 sp=0x3ffff990 excvaddr=0x00000004
ps=0x00000030 sar=0x00000005 vpri=0x00000000
r00: 0x00000000=         0 r01: 0x3ffff990=1073740176 r02: 0x00000000=         0
r03: 0x00000001=         1 r04: 0x00000000=         0 r05: 0xffffffff=        -1
r06: 0x3ffe8bb4=1073646516 r07: 0x00000070=       112 r08: 0x00000000=         0
r09: 0x00000000=         0 r10: 0x3ffed5e8=1073665512 r11: 0x00000001=         1
r12: 0x3ffed5e8=1073665512 r13: 0x00000002=         2 r14: 0x3ffe8b10=1073646352
r15: 0x00000001=         1

piperpilot, 2017-03-26:
I saw the same thing yesterday, but it only triggered a WDT reset when I hit the webserver.  I haven't had a chance to look into it and diagnose any more.

unit-by, 2017-03-26:
I found. It was strange problem with ESP sdk after update from 3.1.1 to 3.1.2. Fresh install resolv problems

closed 2017-03-26 by unit-by