SmingHub/Sming #1052: Problem with make SmartConfig project

philipstefan, 2017-03-28:
When i'm trying to make with this line:

> WifiStation.smartConfigStart(SCT_EspTouch, smartConfigCallback);

It's not compiling, the following errors occurs:

/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_sniffer.o): In function `sc_system_init_done':
(.irom0.text+0x670): undefined reference to `espconn_delete'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_sniffer.o): In function `sc_system_init_done':
(.irom0.text+0x6a6): undefined reference to `espconn_delete'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_sniffer.o): In function `smartconfig_start':
(.irom0.text+0x748): undefined reference to `espconn_delete'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_airkiss.o): In function `KISS_Find_channel':
(.irom0.text+0x18fc): undefined reference to `espconn_sent'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_airkiss.o): In function `KISS_Find_channel':
(.irom0.text+0x196d): undefined reference to `espconn_sent'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_airkiss.o): In function `KISS_Find_channel':
(.irom0.text+0x19de): undefined reference to `espconn_delete'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_airkiss.o): In function `KISS_Find_channel':
(.irom0.text+0x1a1c): undefined reference to `espconn_port'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_airkiss.o): In function `KISS_Find_channel':
(.irom0.text+0x1a20): undefined reference to `espconn_regist_sentcb'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_airkiss.o): In function `KISS_Find_channel':
(.irom0.text+0x1a24): undefined reference to `espconn_create'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_airkiss.o): In function `KISS_Find_channel':
(.irom0.text+0x1a85): undefined reference to `espconn_port'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_airkiss.o): In function `KISS_Find_channel':
(.irom0.text+0x1ab0): undefined reference to `espconn_regist_sentcb'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_airkiss.o): In function `KISS_Find_channel':
(.irom0.text+0x1ab8): undefined reference to `espconn_create'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_esptouch.o): In function `TOUCH_Find_sync_ht40':
(.irom0.text+0x22f5): undefined reference to `espconn_sent'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_esptouch.o): In function `TOUCH_Find_sync_ht40':
(.irom0.text+0x2396): undefined reference to `espconn_delete'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_esptouch.o): In function `TOUCH_Find_sync_ht40':
(.irom0.text+0x23dc): undefined reference to `espconn_regist_recvcb'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_esptouch.o): In function `TOUCH_Find_sync_ht40':
(.irom0.text+0x241a): undefined reference to `espconn_port'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_esptouch.o): In function `TOUCH_Find_sync_ht40':
(.irom0.text+0x2445): undefined reference to `espconn_regist_sentcb'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_esptouch.o): In function `TOUCH_Find_sync_ht40':
(.irom0.text+0x2451): undefined reference to `espconn_regist_recvcb'
/opt/esp-open-sdk/sdk/lib/libsmartconfig.a(sc_esptouch.o): In function `TOUCH_Find_sync_ht40':
(.irom0.text+0x245a): undefined reference to `espconn_create'
collect2: error: ld returned 1 exit status
make: *** [out/build/app.out] Error 1

I'm on MacOS and installed this open-sdk: https://github.com/pfalcon/esp-open-sdk#macos, following the instructions on this issue: https://github.com/SmingHub/Sming/issues/775

philipstefan, 2017-03-28:
After reinstalling esp-open-sdk and Sming it just worked. Please close this issue.

slaff, 2017-03-28:
We are using the open LWIP by default and in it the espconn_* functions are not compiled by default. If you want to use smartconfig you should use the lwip library with espconn in it. The easiest way to do this is to add ENABLE_ESPCONN=1 in your application Makefile-user.mk (as shown here: https://github.com/SmingHub/Sming/blob/develop/samples/Basic_SmartConfig/Makefile-user.mk#L41 ) and recompile your app (make clean; make ).

philipstefan, 2017-03-28:
Thank you! Please close this issue.

closed 2017-03-29 by slaff