SmingHub/Sming #1057: Confusion over three PWM alternatives

pief, 2017-03-30:
I tried to assess the situation with regards to PWM alternatives in Sming. If I grasp things correctly, it goes like this:

1. There is ``Sming/SmingCore/DriverPWM.cpp`` which is used by ``samples/Basic_PWM``. This seems to be a Sming-specific software timer-only implementation. For this one I can't find minimum and maximum allowed values for ``analogWrite()`` documented and, what is worse, a simple ``AnalogWrite(pin, 1)`` to get the lowest possible mean voltage seems to output a higher mean voltage than the alternatives (tested with a simple LED connected).
1. There is ``Sming/SmingCore/DriverPWM.cpp`` which is used by ``samples/Basic_HwPWM``. This one by default uses the ``pwm.c`` module from the Espressif SDK. For this analogWrite minimum and maximum values are defined and a simple ``AnalogWrite(pin, 1)`` works well for me. But the Espressif SDK has [some issues of its own](https://lurchi.wordpress.com/2016/06/29/esp8266-pwm-revisited-and-reimplemented/) which is why...
1. There is also ``Sming/SmingCore/DriverPWM.cpp`` using [Stefan Bruens' alternative PWM](https://github.com/StefanBruens/ESP8266_new_pwm) which gets activated when the Sming app is compiled with ``ENABLE_CUSTOM_PWM=1`` (merged in original PR #808, app compile time switch added in PR #842).

There is however little help as to when to the "why" there are now three different PWM implementations to choose from and, if there are good reasons for multiple to exist, when to pick which. I'd thus like to collect some info so I can contribute to updating the docs accordingly.

1. While in my personal quick testing ``DriverPWM`` did not deliver good results, I assume there's a good reason it is "still there"? Or should it be considered deprecated?
2. If not deprecated, what are reasons for using ``DriverPWM`` and what are reasons for using ``HardwarePWM``? I assume because the latter uses a hardware timer (at least in the Espressif case), one argument could be "If you use hardware timer functions yourself, you need to use ``DriverPWM``"?
3. Are there scenarios where one does *not* want to use Stefan Bruens Custom PWM but use the Espressif PWM instead?

Thanks in advance!

slaff, 2017-03-30:
@anakod Is there a need for DriverPWM ( the original SoftwarePWM) or we can remove it completely and recommend our users to use HardwarePWM with the custom PWM library from Stefan Bruens? What is your opinion?

anakod, 2017-04-03:
Personally I'm prefer to remove all old stuff what have actual and better replacements.

> If you use hardware timer functions yourself, you need to use DriverPWM
That's may be right point, original code use standard software timer. Is this option needed? If yes, anyway default sample _should_ use better alternative (and may be explain other alternatives also, if some other options can be useful potentially).

And I'm sure, we don't need two separated examples + one independent compilation flag, because that's too complicated :)

pief, 2017-04-04:
So if I understand you correctly you suggest to
1. just provide ONE sample
1. that uses HardwarePWM (with or without Stephans custom PWM?)
1. and in its comments explain that there is alternatively DriverPWM and when one might want to use that?

slaff, 2017-04-24:
PR #1101 is the initial step in proving ONE recommended PWM driver. Step 2 will be in the next release where DriverRWM will be removed and Stefan Bruens PWM will be used by default.

closed 2017-04-24 by slaff