SmingHub/Sming #1058: I2C do not work after 3.1.0 RELEASE

avr39-ripe, 2017-03-31:


slaff, 2017-03-31:
Can you share some more information about the problem? How to prove that it works and how to check that the latest version is not working?

avr39-ripe, 2017-03-31:
Anybody can just pick whatever i2c related example app and try it yourself.
I try it with DS3232RTC_NTP_Setter as example. When properly connected DS3231 module, this app try to sync time via ntp and set both local time and DS3231 time and in loop it prints rtc time in local and utc time zone AND it prints time obtained from DS3231 module. So when module connected and i2c subsystem works fine there is some meaningful readings from DS3231, but when either module is disconnected or i2c do not work there are only  default values printed (01.01.1970 00:00:00). So I compile this app with recent develop and with patched develop, no library changes, no app changes, no wiring changes - develop - I get 01.01.1970 00:00:00... and patched develop I get proper readings, and same thing with Sming prior to commit that I mention in first post.
Also can easily be tested wuth LiquidCrystal_44780 example app. It wil either show info on LCD or not. With working i2c it shows SMING: Lest do smart things, and without - only uninitialized lcd screen.
Also test with MCP23017 lib.
So i2c subsystem definitively broken.

zgoda, 2017-04-02:
I had no problems with HD44780 display on PCF8574 expander running `develop` couple of days ago, definitely post-3.1.0 release.

avr39-ripe, 2017-04-04:
My fault (core dumped :)
It turns out, that _almost_ nothing was broken in i2c in sming after 3.1.0. It was my hardware which cause trouble. I use MCP23017 i2c portexpender connected to esp8266 without SDA/SCL pullup resistors, but on Sming prior 3.1.0 it works just fine.But 3.1.0 introduces new pinMode variant incompatible with old one in aspect of enabling pin input pullup.  All properly made i2c modules have pullup resistors, so they continue to work as a charm, but my crappy hardware haven't such resistors and starts to work odd. There are two software only fixes for this: #1060 or newer Wire + i2c HAC ported from esp8266/Arduino.

closed 2017-04-04 by avr39-ripe