SmingHub/Sming #1059: Repeat calls to webserver ends in CONNECTION DROPPED (low heap)

piperpilot, 2017-04-01:
This issue has a long history and was all just previously close.  Check out the following issues:

https://github.com/SmingHub/Sming/issues/742
https://github.com/SmingHub/Sming/issues/656
https://github.com/SmingHub/Sming/issues/220

This is caused by the following code found in TcpServer.cpp:

`	
// Anti DDoS :-)
	if (system_get_free_heap_size() < 6500)
	{
		debugf("\r\n\r\nCONNECTION DROPPED\r\n\t(%d)\r\n\r\n", system_get_free_heap_size());
		return ERR_MEM;
	}
`

I see the intent...its to keep the heap from going too low.  So I guess the question is...how do we free up heap quicker when  connection closes.  You can see the debug here:

`
path=/ajax/smobot
Name: host
host === 192.168.1.18
parsed
Request: GET, nodata
TCP received: 357 bytes
onReadyToSendData: 1
response sendHeader
response sendBody
Written: 170, Available: 170, isFinished: 1, PushCount: 1
Stream completed
TCP connection closing
~TCP connection
-TCP connection
onAccept state: 0 K=0
Free heap size=6944, K=0
+TCP connection
timeout updating: 70 -> 90
TcpServer onClient: 192.168.1.13

path=/ajax/smobot
Name: host
host === 192.168.1.18
parsed
Request: GET, nodata
TCP received: 357 bytes
onReadyToSendData: 1
response sendHeader
response sendBody
Written: 170, Available: 170, isFinished: 1, PushCount: 1
Stream completed
TCP connection closing
~TCP connection
-TCP connection
onAccept state: 0 K=0
Free heap size=6760, K=0
+TCP connection
timeout updating: 70 -> 90
TcpServer onClient: 192.168.1.13

path=/ajax/smobot
Name: host
host === 192.168.1.18
parsed
Request: GET, nodata
TCP received: 357 bytes
onReadyToSendData: 1
response sendHeader
response sendBody
Written: 170, Available: 170, isFinished: 1, PushCount: 1
Stream completed
TCP connection closing
~TCP connection
-TCP connection
onAccept state: 0 K=0
Free heap size=6576, K=0
+TCP connection
timeout updating: 70 -> 90
TcpServer onClient: 192.168.1.13

path=/ajax/smobot
Name: host
host === 192.168.1.18
parsed
Request: GET, nodata
TCP received: 357 bytes
onReadyToSendData: 1
response sendHeader
response sendBody
Written: 170, Available: 170, isFinished: 1, PushCount: 1
Stream completed
TCP connection closing
~TCP connection
-TCP connection

CONNECTION DROPPED
.(6392)
`
In this case my javascript is calling for an AJAX data return every 1 second.  I will try to increase the time between calls, but I would expect once a second to return a simple string would be fine.  I also believe in older versions this worked just fine without this issue.  It seems to take a REALLY long time for memory to be freed back up.  Does anyone have any ideas how we can fix this?

icodk, 2017-04-01:
I think the heap vanishing is because the stack tries to conform to the TCP standard which keep closed connections in the WAIT_STATE for up to few minutes. This makes a lot of sense in a multi user OS but has no relevance for a platform like the esp. Especially when you call the exact the same server again.
I think it is a build in misconception in the SDK from Espressif.
Can it be done better with lwIP ?

slaff, 2017-04-01:
At the moment I am working on complete rewrite for the HttpClient. As [part of the plan](https://github.com/slaff/Sming/issues/5) I thought about refactoring also the HttpServer. If you are interested to help, please let me know.

slaff, 2017-04-01:
> In this case my javascript is calling for an AJAX data return every 1 second.

If your script is calling every second then that should result in one Tcp connection to the server as long as the server does not close the connection immediately but has a keep-alive of 5 seconds, for example.  I have to check if that is the case with the current HttpServer and eventually refactor it allow  setting a keep-alive timeout.

piperpilot, 2017-04-02:
The only thing I can find with respect to keep-alive or timeout is in the TcpServer.  We do set a timeout...

`TcpServer::TcpServer() : TcpConnection(false)
{
	timeOut = 40;
	TcpConnection::setTimeOut(USHRT_MAX);
}`

But I don't see that we are using the timeOut variable.  There is a USHRT_MAX value which I can't find what that is...it must be something in the SDK which I haven't looked at yet.  I suspect that is a large number, but haven't confirmed yet. 

I'm still learning how the server code all works, but would be glad to collaborate on the re-write.  I use both HTTPServer and HTTPClient as the core of my application, so I want to make it as stable as possible.

piperpilot, 2017-04-02:
Something weird is definitely going on.  I turned on some additional debugging and issued a single REST call for some simple information.  I got the following...it looks like it is opening 4 connections for each request.

`onAccept state: 0 K=0
Free heap size=23576, K=0
+TCP connection
timeout updating: 70 -> 90
TcpServer onClient: 192.168.1.13

onAccept state: 0 K=1
Free heap size=23320, K=1
+TCP connection
timeout updating: 70 -> 90
TcpServer onClient: 192.168.1.13

onAccept state: 0 K=2
Free heap size=23160, K=2
+TCP connection
timeout updating: 70 -> 90
TcpServer onClient: 192.168.1.13

onAccept state: 0 K=3
Free heap size=23016, K=3
+TCP connection
timeout updating: 70 -> 90
TcpServer onClient: 192.168.1.13

onAccept state: 0 K=4
Free heap size=22864, K=4
+TCP connection
timeout updating: 70 -> 90
TcpServer onClient: 192.168.1.13

path=/ajax/smobot
Name: host
host === 192.168.1.18
parsed
Request: GET, nodata
TCP received: 419 bytes
onReadyToSendData: 1
response sendHeader
response sendBody
Written: 164, Available: 164, isFinished: 1, PushCount: 1
Stream completed
TCP connection closing
~TCP connection
-TCP connection
TcpConnection::staticOnReceive: pbuf is NULL
TCP received: (null)
TCP connection closing
~TCP connection
-TCP connection
TcpConnection::staticOnReceive: pbuf is NULL
TCP received: (null)
TCP connection closing
~TCP connection
-TCP connection
TcpConnection::staticOnReceive: pbuf is NULL
TCP received: (null)
TCP connection closing
~TCP connection
-TCP connection
TcpConnection::staticOnReceive: pbuf is NULL
TCP received: (null)
TCP connection closing
~TCP connection
-TCP connection`

anakod, 2017-04-02:
That's right, issue related to TCP protocol internals - WAIT_STATE, we can't skip closed connection some time after completion (60 seconds, I think). Most effective way for high frequency communication - use WebSickets. It will not close previous opened connection, so issue will be resolved by design.

slaff, 2017-05-04:
>  I got the following...it looks like it is opening 4 connections for each request.

How do you test? Did you use low-level tools like `curl` or higher-level Chrome plugins? Chrome, and other high-level tools, usually make one HEAD request before the actual request to check things like CORS headers, supported methods, etc. 

Try with the latest `develop` version and see what is reported there. I took extra care to refactor also the HttpServer class so it should be more predictable now.

slaff, 2017-11-13:
Feel free to reopen the issue if the problem is still present and you have more details concerning it.

closed 2017-11-13 by slaff