SmingHub/Sming #1066: m_vsnprintf does not return size of required buffer

tius2000, 2017-04-07:
It seems to me, that _m_vsnprintf_ does not return the size of the required buffer if _maxLen_ is too small. _Print::printf(...)_ does handle this by increasing the size by one until the string fits, but this **very** inefficient.

ADiea, 2017-04-07:
it's difficult to know the exact size needed. If you run out of buffer you cannot print more. There is actually an overflow guard and if current size < maxSize-overflow_guard then maxlength is returned.
I agree that increasing by 1 byte is inefficient I would increase by 16 for ex. This way we also make fewer mallocs => better control of fragmentation

tius2000, 2017-04-11:
Hi @ADiea, I will have a look on it try to find a solution.

tius2000, 2017-04-11:
PR #1077 should fix this, perhaps you can take a look @ADiea? Unfortunately I had to change the existing logic, hope I got it right.

closed 2018-10-08 by slaff