SmingHub/Sming #1067: possible extension: PROGMEM methods for Print.cpp

tius2000, 2017-04-07:
I recently added some missing methods to Print.cpp:

```
//	allow the user to implement methods with variable number of arguments
size_t	vprintf(const char *, va_list va);

//	allow the user to save some ram by moving strings to PROGMEM
size_t	write_P(const char* data, size_t size);
size_t	write_P(const char* data);
size_t	printf_P(const char* data, ...);
size_t	vprintf_P(const char *, va_list va);
```

I can contribute the implementation if there is any interest. However, there is a conflict with the macro _printf_P_ in FakePgmSpace.h :-(

I do not fully understand the implications, but IMHO it would be better to use inline functions instead of macros in FakePgmSpace.h - what do you think?

ADiea, 2017-04-07:
Probably your implementation of printf_P is the same so should be no issue replacing printf_P macro with inline function. For debug strings printf_P_stack is used which does no malloc and is safer.

tius2000, 2017-04-18:
I created a PR (#1094), perhaps you can have a look @ADiea? Stack space is allocated on demand, so malloc is not needed. However, I do not know how to fix the conflict with the macro _printf_P_ in FakePgmSpace.h :-(

ADiea, 2017-04-19:
Reviewed the PR. I think only way to avoid name collision is to make printf_P a function in FakePgmSpace...  
Or to rename the printf_P function in Print class. Or to rename printf_ in Fakepgmspace.h to something like PRINTF_P (macro case) but please make sure it is not used in other parts of the code. Also this last option will break user code if someone is relying on printf_P as a global symbol.

slaff, 2019-07-04:
That should be solved in the latest `develop` code.

closed 2019-07-04 by slaff