SmingHub/Sming #1069: Is a SSL websocket server supported?

Programming4life, 2017-04-07:
Can I create a SSL/TLS websocket server? I know the websocket client supports ssl, but what about the server?

slaff, 2017-04-07:
@Programming4life Not right now. But I am working heavily on this and it will be available for testing soon. I will let you know when it is ready.

slaff, 2017-05-02:
With the upcoming PR #1112 it would be possible to create a simple WebSocket Server.  The following sample code can be used as a start for such an implementation

```C
HttpServer* server;

void onWsMessage(WebSocketConnection& connection, const String& text) {
	String answer = "ping:" + text;
	connection.sendString(answer);
}

void onWsConnected(WebSocketConnection& connection) {
	String answer = "SYSTEM: User joined";
	connection.sendString("Welcome!");
	connection.broadcast(answer.c_str(), answer.length());
}

void onWsDisconnected(WebSocketConnection& connection) {
	connection.sendString("Bye!");
	String answer = "SYSTEM: User left";
	connection.broadcast(answer.c_str(), answer.length());
}

void ready() 
{
         server = new HttpServer();

#include "ssl/server_private_key.h"
#include "ssl/server_cert.h"

	SSLKeyCertPair clientCertKey;
	clientCertKey.certificate = new uint8_t[default_certificate_len];
	memcpy(clientCertKey.certificate, default_certificate, default_certificate_len);
	clientCertKey.certificateLength = default_certificate_len;
	clientCertKey.key = new uint8_t[default_private_key_len];
	memcpy(clientCertKey.key, default_private_key, default_private_key_len);
	clientCertKey.keyLength = default_private_key_len;
	clientCertKey.keyPassword = NULL;

	server->setServerKeyCert(clientCertKey);

	bool useSsl = true;

	server->listen(1111, useSsl);

	WebsocketResource* wsResource = new WebsocketResource();
	wsResource->setMessageHandler(onWsMessage);
	wsResource->setConnectionHandler(onWsConnected);
	wsResource->setDisconnectionHandler(onWsDisconnected);

	server->addPath("/ws", wsResource);
}
```

Please, have in mind that a single SSL connection requires more than ~17K memory for the initial handshake. This means that it will be unrealistic to expect two or more simultaneous SSL connections to work as expected.

closed 2017-07-18 by slaff