SmingHub/Sming #1076: Liberal use of String class

zgoda, 2017-04-11:
I'm just a noob when it comes to embedded programming so bear with me if I am asking funny questions.

I've been told many times that dynamic memory allocation in embedded programming is root of all evil and everytime you use `String` instance a sweet little chimpanzee is killed. Now I am looking at Sming code and I see `String` everywhere, passed by value to functions, concatenated many times, some modules are just interfaces to SDK functions only to return `String` like in javascript.

What's the key here? Does Sming have that magic memory allocator that squashes heap every so often? Or what?

ADiea, 2017-04-12:
You are right... if you have time please find as many places where String is passed as value and could be passed as ref/ptr and make a PR it will improve fragmentation.
Also all core functions in sming should NOT use String at all on the long term.

When you create a string object internally malloc is used to reserve string_length bytes of RAM on heap. When a string is destructed free is used to unreserve the RAM, but strings are tipically small and create holes in RAM known as fragmentation. 
For instance you start with 32 bytes of free heap. If you need 16 bytes it will give you first 16 bytes and mark next 16 bytes as free. then a string comes in of 4 bytes and next 4 bytes are given to the string remaining 10 bytes free. then a 10Byte string comes in place and allocates the last 10 bytes. Then the first 16 bytes and the last 10 bytes are freed as not needed but the 4 byte string is there ight in the middle
<16B free> <4B string> <10Bytes free>
Now if you need 20 bytes even if you have 16+10=26 free bytes reported by system_free_heap() you will have nowhere 20 continuous bytes because the 4 bytes are right in the middle.

Now, if you were to use stack memory to work with strings like this
```
void fct(char* buffer)
{
///
int len = strlen(buffer);
if(len < 128)
{
 char buf[len]; //stack ponter is increased by len
//work on buf
}
else
{
debugf("string too long, could cause stack overflow");
}
} //stack pointer is decreased by len
```
Now the memory is used by function and on fct return all memory is reclaimed and can be used by caller function etc
With stack you have to be carefull not to allocate too much as this coupled with too deep funciton call tree can cause stack overflow (no more memory available) or in our system it can cause to write over the heap space, because we increase heap from bottom to top of free ram and the stack from top to bottom so there is posibillity of meetoing in the center and we currently have no protection for that.

I hope I have provided enough details if not let me know and I can explain more. 
Bottom line is we should use malloc as little as possible and if you already constructed a string try to pass it by refference always when possible.

zgoda, 2017-04-12:
If by "core" you mean all what is included in `SmingCore.h` then this is going to be massive. And will break a hell of backward compatibility.

ADiea, 2017-04-12:
I was thinking of the Tcp stuff. TCP client, server and derived classes. I guess here we use a lot of Strings to operate on buffers.

zgoda, 2017-04-12:
Is there anything (in SDK?) that could help compare heap fragmentation? Measure it in quantitative way so it can be compared between invocations. Because `system_get_free_heap_size()` will return only total free memory size, including "holes".

ADiea, 2017-04-13:
Does not exist yet. I plan to do it based on umm_malloc but did not yet had time to start this task.

zgoda, 2017-04-13:
I had something like this on AVR, bunch of functions that returned eg largest free block, smallest block, number of free blocks and so on.

zgoda, 2017-05-29:
When it comes to URL manipulation I would suggest looking at https://github.com/jacketizer/libyuarel, works as advertized.

closed 2019-12-31 by slaff