SmingHub/Sming #1095: Do not start network operations in system ready handler

zgoda, 2017-04-19:
In discussion in PR #1092 I've been told that system ready handler is wrong place to start any network operations because it is related only to TCP stack and not to actual network connectivity or readiness of particular components like STA or AP. The samples that use system ready handler to start networked operations are HttpServer_ConfigNetwork, DNSCaptivePortal and to some extent Basic_WiFi.

slaff, 2017-04-20:
# Short Story
If ESP is running as software Access Point only then it is ok to use System.onReady to register a callback that will do network operations. In all other cases that is not ok. 

# Long Story
There are two major prerequisites in order to have useful network operations.
1.  The TCP stack has to be fully initialized
2. There should be connection to a remote Access Point (AP) or the ESP itself has to be running as (software) AP.

Let me elaborate a bit on both pre-requisites. 
-  The TCP stack has to be fully initialized
That means that at least lwip_init() should have been called. If we run a piece of code that requires the TCP stack, in our case LWIP, and it is called before lwip_init() this will most certainly mean that our code will not work as expected. 
In Sming that requirement can be fulfilled by registering the function/code that needs  TCP/IP by passing it as parameter to System.onReady(callback);
System.onReady will trigger the callbacks once the system is ready, which includes also TCP stack being ready. It does not guarantee that ESP will be connected to AP or there is a software AP running on it. 
- There should be connection to a remote Access Point (AP) or the ESP itself has to be running as (software) AP.
Let's say we have a function called `startServers` that is registered with System.onReady and will be called when the system is ready. If  ESP is not connected to an AP and it is not itself an AP then we cannot access those servers. They are just useless. While not technically wrong, logically it makes more sense to start your servers once you are sure they can be accessed. 
When ESP acts as a station that connects to an AP then WifiEvents provides us with a way to register a callback after an IP is given back from the AP's DHCP. In that case it is highly recommended to replace System.onReady callbacks with WifiEvents.onStationGotIP callback. 
When ESP acts as software AP itself then there is a hidden promise that the AP will start. In addition there is no callback available for us in WifiEvents for AP started event. That is why we are left with only one callback to choose from - namely System.onReady. And that is why HttpServer_ConfigNetwork is using System.onReady.

zgoda, 2017-04-20:
Okay, now it's clear. Thanks.

closed 2017-04-20 by zgoda