SmingHub/Sming #1099: Example MRFC522 Not Working

fernandopasse, 2017-04-20:
What should be the pin settings on the Wemos D1 board?
MFRC522 mfrc522 (SS_PIN, SS_PIN); it seems strange

icodk, 2017-04-20:
@fernandopasse 
D1 Mini connections (same for any nodemcu derivative):
`MISO <-------> D6`
`MOSI <-------> D7`
`CLK       <-------> D5`
`SDA       <--------> D8`
Notes:

- SDA can be connected to any output that you control. I used D8 because it must be low on start up    so I can't use it as an input.

- RST  is deliberetly not connected, same as IRQ

- You are right regarding initialization with the same pin for both cs and reset. However it works for me. My SS_PIN is 15 for D8

- The example is for two NFC antennas, this requires 2 different SS_PINs .... And that the two antennas understands the sda line and do not mass with the MISO line when not active. 
The sda line is important and you can't just set it to GND, ln a single antenna configuration. I tried in order to save a pin/cable but didn'work for me.

fernandopasse, 2017-04-20:
I tried this...but it didn't

#include <user_config.h>
#include <SmingCore/SmingCore.h>
#include <Libraries/MFRC522/MFRC522.h>

Timer procTimer;
static Timer nfcScanTimer;
int helloCounter = 0;

void ICACHE_FLASH_ATTR scanNfc(byte scanner);

#define SS_PIN	15  	// D8

MFRC522 mfrc522(SS_PIN, SS_PIN);	// Create MFRC522 instance
byte ss_pin[] =
{ 15, 15 };

void sayHello()
{
	for (int pinNdx = 0; pinNdx < 2; pinNdx++)
	{
		byte pin = ss_pin[pinNdx];
		mfrc522.setControlPins(pin, pin);
		mfrc522.PCD_Init();    // Init MFRC522
		scanNfc(pin);
	}

}
//---------------------------------
static void ICACHE_FLASH_ATTR dump_byte_array(byte *buffer, byte bufferSize)
{
	String hexOut;
	for (byte i = 0; i < bufferSize; i++)
	{
		hexOut += String(buffer[i], HEX);
	}
	debugf("%s", hexOut.c_str());

}
//---------------------------------
void ICACHE_FLASH_ATTR
scanNfc(byte scanner)
{
	static int cnt;
	if (!mfrc522.PICC_IsNewCardPresent())
	{
		debugf("Scanning nfc Scanner:%d \r\n", scanner);
		return;
	}
	if (!mfrc522.PICC_ReadCardSerial())
	{	// Select one of the cards
		debugf("Selecting card failed...");

	}
	else
	{
		// Show some details of the PICC (that is: the tag/card)
		debugf("Card UID on scanner:%d:", scanner);
		dump_byte_array(mfrc522.uid.uidByte, mfrc522.uid.size);
		debugf();
	}
	mfrc522.PICC_HaltA();
	// Stop encryption on PCD
	mfrc522.PCD_StopCrypto1();
	mfrc522.PCD_Init();    // Init MFRC522

	//nfcScanTimer.restart();
}

void init()
{
	Serial.begin(SERIAL_BAUD_RATE); // 115200 by default

	procTimer.initializeMs(2000, sayHello).start();

	//----- NFC
	MFRC522 mfrc522(SS_PIN, SS_PIN);
	SPI.begin();
	mfrc522.PCD_Init();    // Init MFRC522

	//nfcScanTimer.initializeMs(50, scanNfc).startOnce();

}

I used one NFC antenna...

icodk, 2017-04-20:
@fernandopasse 
Do you get an error or it just can't read the  tag ?
Can you add a link to the shop/product of your antenna ?

fernandopasse, 2017-04-20:
I have this one: http://s.aliexpress.com/bInE77JV
On the Arduino IDE the NFC antenna worked but on Sming it didn't.

icodk, 2017-04-20:
@fernandopasse 
Attache is a working example that works for me
see if you can get it to work
[Basic_NFC.zip](https://github.com/SmingHub/Sming/files/945527/Basic_NFC.zip)

fernandopasse, 2017-05-05:
I tested your solution but it did not work either :(

icodk, 2017-05-05:
Sorry to here that. 
I will test it again and let you know

icodk, 2017-05-06:
@fernandopasse 
Attached are
[Basic_NFC_170506.zip](https://github.com/SmingHub/Sming/files/981005/Basic_NFC_170506.zip)
 a video and a program that I just run on my nodemcu that reads an NFC tag
You can also see the wiring so check it out and let me know.
[IMG_0078.zip](https://github.com/SmingHub/Sming/files/981004/IMG_0078.zip)
Note:
Out of 12 NFC antennas I bought, 3 of them could not read tags.

closed 2019-08-07 by slaff