SmingHub/Sming #1104: FileSystem.fileGetContent() and null bytes issue

MayaPosch, 2017-04-25:
While trying to read binary files from a SpifFS segment in the Flash memory, I found that using the fileGetContent() convenience function I would consistently get only short strings back, of just a few bytes. While digging into the function's code, I found that the issue is in how it assigns the character array to a new String's constructor:

`String res = buffer;`

This invokes a constructor which will read in a null-terminated string, which explains the issue I was having. In order to read in the full buffer, one has to explicitly define the buffer size for the constructor:

`String res(buffer, size);`

I successfully tested this code in a current project.

I think it would make sense to incorporate this change so that it will always return the entire file's contents, as that's what the function's name suggests it will do. There should be no limitation on whether one can use binary files, either.

slaff, 2017-04-25:
@MayaPosch Can you submit a Pull Request (PR) with the suggested change? Thanks in advance.

MayaPosch, 2017-04-25:
I have submitted a PR with the change (#1106).

slaff, 2017-04-25:
PR is merged.

closed 2017-04-25 by slaff