SmingHub/Sming #1109: ** Preparation for Release 3.3.0 **

slaff, 2017-04-28:
This issue is created to collect all the tasks we need to do before the 3.2.0

### Source Code
- [ ] Update Sming/SmingCore/SmingCore.h (#1107)
- [ ] Update Readme.md (#1107)
- [ ] Merge develop into master
- [ ] Tag the source code in the master branch

### Refactoring 
- [ ] Http Client and Http Server

kwis2, 2017-06-16:
I found `chocolately` problem was lost between 3.2.0 and 3.3.0 versions

slaff, 2017-06-16:
> I found chocolately problem was lost between 3.2.0 and 3.3.0 versions

We don't have the people that can solve it for now. Anyone who is working on Windows is welcome to help us.

Vity01, 2017-06-17:
Maybe I should define new issue for this, but current latest develope branch is not compilable on Windows. 
I tried to do that without Chocolately. Eg. the makefile in SmingFramework contains '$(shell find SmingCore -type d)' which calls unix command. I replaced that with concrete folders, after more tuning folder settings I was able to compile without error.
Next I tried to build sample HttpClient_Teamspeak with an error 'collect2.exe: error: ld returned 1 exit status'. Wiki describes this error as a problem with a framework compilation...

icodk, 2017-06-17:
@Vity01 
Thanks for mention it. There are few  previous issues (1140 and 1149)  that unfortunately  never addressed by any of the experts on  this repository. Looks like Sming's  Windows  comparability is  silently deprecated.
It is not a complain it is an observation. The probable cause is shortage of experts time.

closed 2017-07-26 by slaff

Vity01, 2017-07-26:
Is 3.3 already compatible with Windows?

slaff, 2017-07-26:
@Vity01 It should be. If you are using choco something like this should download for you the latest version
```
choco upgrade -y -f sming.core --version 0.0.4
```