SmingHub/Sming #1113: HttpServer opens multiple connections for one request

piperpilot, 2017-05-04:
Not exactly sure what I am looking at here...but with the new httpServer, it looks like multiple connections are being opened and they aren't closing the request properly after sending the data.  This is using the code from httpserver_configNetwork example, the get-networks method.  Here is what I am seeing in the log when I hit it with a browser:

`
onAccept state: 0 K=0
Free heap size=27664, K=0
+TCP connection
Opening connection. Total connections: 1
timeout updating: 70 -> 10
TcpServer onClient: 192.168.4.2

onAccept state: 0 K=1
Free heap size=27152, K=1
+TCP connection
Opening connection. Total connections: 2
timeout updating: 70 -> 10
TcpServer onClient: 192.168.4.2

onAccept state: 0 K=2
Free heap size=26760, K=2
+TCP connection
Opening connection. Total connections: 3
timeout updating: 70 -> 10
TcpServer onClient: 192.168.4.2

The headers are complete
Network Count: 6)
TCP received: 441 bytes
realloc 1 -> 131
realloc 130 -> 260
realloc 259 -> 325
realloc 324 -> 390
realloc 389 -> 455
realloc 454 -> 520
Written: 511, Available: 511, isFinished: 1, PushCount: 1
Body stream completed
onReadyToSendData: 1
TCP sent: 644
onReadyToSendData: 2
TCP connection closed by timeout: 10 (from 10)
Closing connection. Total connections: 2
TCP connection closing
~TCP connection
TCP connection closed by timeout: 10 (from 10)
Closing connection. Total connections: 1
TCP connection closing
~TCP connection
TCP connection closed by timeout: 10 (from 10)
Closing connection. Total connections: 0
TCP connection closing
~TCP connection
-TCP connection
-TCP connection
-TCP connection
`

I have attached a couple of screenshots of the chrome console as well.  You can see it takes like 12 seconds to download the content which is only 644 bytes.  Something is unstable here.

[image]
[image]

slaff, 2017-05-04:
That issue should be fixed with that PR #1114.

piperpilot, 2017-05-04:
OK, cool...so there are 2 issues here...the first one is the JSON timeout...your fix in #1114 seems to fix that.  What about the multiple connections.  That still seems strange to me.  Here is a new log:

`The headers are complete
Network Count: 4)
TCP received: 398 bytes
realloc 1 -> 131
realloc 130 -> 260
realloc 259 -> 325
realloc 324 -> 390
Written: 363, Available: 363, isFinished: 1, PushCount: 1
Body stream completed
onReadyToSendData: 1
onAccept state: 0 K=1
Free heap size=23328, K=1
+TCP connection
Opening connection. Total connections: 2
timeout updating: 70 -> 10
TcpServer onClient: 192.168.4.2

onAccept state: 0 K=2
Free heap size=22928, K=2
+TCP connection
Opening connection. Total connections: 3
timeout updating: 70 -> 10
TcpServer onClient: 192.168.4.2

onAccept state: 0 K=3
Free heap size=22512, K=3
+TCP connection
Opening connection. Total connections: 4
timeout updating: 70 -> 10
TcpServer onClient: 192.168.4.2

TCP sent: 517
onReadyToSendData: 2
TCP connection closed by timeout: 10 (from 10)
Closing connection. Total connections: 3
TCP connection closing
~TCP connection
TCP connection closed by timeout: 10 (from 10)
Closing connection. Total connections: 2
TCP connection closing
~TCP connection
TCP connection closed by timeout: 10 (from 10)
Closing connection. Total connections: 1
TCP connection closing
~TCP connection
-TCP connection
-TCP connection
-TCP connection
TCP connection closed by timeout: 10 (from 10)
Closing connection. Total connections: 0
TCP connection closing
~TCP connection
-TCP connection`

slaff, 2017-05-04:
> HttpServer opens multiple connections for one request

The title of the issue is not exactly correct. The web browser makes a request to get_networks. Because there is no active tcp connection it opens ONE new tcp connection. Then the server responds.

```
Network Count: 6)
TCP received: 441 bytes
Body stream completed
onReadyToSendData: 1
TCP sent: 644
onReadyToSendData: 2
```
The data is sent from the server to the browser but  due to the bug fixed in #1114  the browser was waiting until the server timeouts the request.

```
TCP connection closed by timeout: 10 (from 10)
```

Timeout causes the tcp connection to be closed from the server. Since the browser did not get correct response it retries again. The previous Tcp connection is closed due to timeout. That is why it opens a new connection. Waits again for the timeout and repeats again...

piperpilot, 2017-05-04:
Sure looks like its opening multiple TCP connections (4 in this case).  With the fix in place, you can see 4X

onAccept state: 0 K=0
Free heap size=27464, K=0
+TCP connection
Opening connection. Total connections: 1
timeout updating: 70 -> 10
TcpServer onClient: 192.168.4.2

K=0 through K=3

And then 4 subsequent closes....
TCP connection closed by timeout: 10 (from 10)
Closing connection. Total connections: 4
TCP connection closing
~TCP connection
TCP connection closed by timeout: 10 (from 10)
Closing connection. Total connections: 3
TCP connection closing
~TCP connection
TCP connection closed by timeout: 10 (from 10)
Closing connection. Total connections: 2
TCP connection closing
~TCP connection
TCP connection closed by timeout: 10 (from 10)
Closing connection. Total connections: 1
TCP connection closing

I must be missing something.  When I do a simple text request, I only get one...it seems when it is JSON stream it is multiples.

slaff, 2017-05-04:
> I must be missing something. When I do a simple text request, I only get one...it seems when it is JSON stream it is multiples.

I am not sure about that. You can test this using curl. Run the command below and see how many requests are logged in ESP.

curl -vvv -X "POST" 'http://192.168.4.1/ajax/get-networks' -H 'Pragma: no-cache' -H 'Origin: http://192.168.4.1' -H 'Accept-Encoding: gzip, deflate' -H 'Accept-Language: en-GB,' -H 'User-Agent: Mozilla/5.0' -H 'Content-Type: application/x-www-form-urlencoded; charset=UTF-8' -H 'Accept: */*' -H 'Cache-Control: no-cache' -H 'X-Requested-With: XMLHttpRequest' -H 'Connection: keep-alive' -H 'Referer: http://192.168.4.1/' -H 'DNT: 1' --data 'reconnect=false' --compressed

Make sure to adjust the IP, if needed.

piperpilot, 2017-05-04:
Interesting, with curl it is only 1 TCP connection, but with Chrome its 4.

slaff, 2017-05-04:
Chrome wants to be sure that the CORS requirements are met. This has nothing to do with the HttpServer.

piperpilot, 2017-05-04:
OK cool.  I'll close this issue and open others as needed.

closed 2017-05-04 by piperpilot