SmingHub/Sming #1116: Content Length Mismatch

piperpilot, 2017-05-04:
Getting a consistant issue with GET http://192.168.4.1/ net::ERR_CONTENT_LENGTH_MISMATCH

Here is the log and a screenshot of the chrome info.

Its possibly an issue with the template and the variable smobotversion.  I'm going to remove that and see if there are still issues.

The headers are complete
attached file: index.html (3670 bytes)
TCP received: 424 bytes
READ Template (0)
plain template text pos: 0, len: 1024
Written: 1024, Available: 1024, isFinished: 0, PushCount: 1
READ Template (0)
found var: smobotversion, at 608 (1631) - 621, send size 607
Written: 607, Available: 607, isFinished: 0, PushCount: 2
READ Template (2)
StartVar smobotversion
Written: 5, Available: 5, isFinished: 0, PushCount: 3
READ Template (3)
continue to plaint text
plain template text pos: 1646, len: 1024
Written: 1024, Available: 1024, isFinished: 0, PushCount: 4
READ Template (0)
plain template text pos: 2670, len: 4
Written: 4, Available: 4, isFinished: 0, PushCount: 5
onReadyToSendData: 1
TCP sent: 2780
READ Template (0)
plain template text pos: 2674, len: 996
Written: 996, Available: 996, isFinished: 1, PushCount: 1
Body stream completed
onReadyToSendData: 2
The headers are complete

[image]

slaff, 2017-05-05:
> Getting a consistant issue with GET http://192.168.4.1/

Which sample are we talking about? What is the response information and what are the headers? 

Please run the request with curl and paste the output.

```
curl -vv http://192.168.4.1/
```

slaff, 2017-05-05:
In my two browsers (Chrome and Firefox) and using curl I was not able to reproduce the issue. Observing the headers gave me some more ideas about better client caching though.

Can you please try with another browser? Are you modifying the sample or you use the stock HttpServer_ConfigNetwork example?

slaff, 2017-05-30:
@piperpilot I will close the issue for now. Feel free to reopen it once you can answer the questions above.

closed 2017-05-30 by slaff