SmingHub/Sming #1120: Idea: global handler for failing malloc

zgoda, 2017-05-05:
Suppose I have firmware that does much web work both as server and as client. These functions often allocate memory on heap and working async means memory is allocated and freed in semi-random order, effectively causing heap fragmentation. In such case the total free amount may be sufficient eg to serve another request, but there will be no free continuous area to perform allocation and `malloc()` will fail. I'd like to know when this happens, eg. to restart firmware. Is there any way to know in my code that `malloc()` somewhere failed?

ADiea, 2017-05-06:
I have submitted a pr to ummalloc upstream but no response yet. Define a fct pointer and modify ummalloc function to call your function if pointer not null. Then set fptr from your code

zgoda, 2017-05-08:
That sounds fine but will it work for Sming library functions, when using eg. HttpServer or TcpClient?

ADiea, 2017-05-08:
Yes, everyone calls the same malloc. But you have to compile with ENABLE_CUSTOM_HEAP=1

closed 2020-01-29 by slaff