SmingHub/Sming #1121: error: 'for' loop initial declarations are only allowed in C99 mode

fvpalha, 2017-05-05:
Hi.

I am using esp-open-sdk version 2.0 in Ubutu Linux 15.10.

`export ESP_HOME=/opt/esp-open-sdk`
`export SMING_HOME=/home/palha/git/Sming/Sming`

The Basic_Debug sample doesn't build:

`make[3]: Leaving directory '/home/palha/git/Sming/Sming/third-party/esp-open-lwip'`
`AS system/irq_check.s`
`AS gdb/gdbstub-entry.s`
`CC system/flashmem.c`
...
`CC third-party/http-parser/bench.c`
`CC third-party/ws_parser/ws_parser.c`
`third-party/ws_parser/ws_parser.c: In function 'ws_parser_execute':`
`third-party/ws_parser/ws_parser.c:247:21: error: 'for' loop initial declarations are only allowed in C99 mode`
`                     for(size_t i = 0; i < chunk_length; i++) {`
`                     ^`
`third-party/ws_parser/ws_parser.c:247:21: note: use option -std=c99 or -std=gnu99 to compile your code`
`Makefile:477: recipe for target 'out/build/third-party/ws_parser//ws_parser.o' failed`
`make[2]: *** [out/build/third-party/ws_parser//ws_parser.o] Error 1`
`make[2]: Leaving directory '/home/palha/git/Sming/Sming'`
`Makefile:451: recipe for target 'reload' failed`
`make[1]: *** [reload] Error 2`
`make[1]: Leaving directory '/home/palha/git/Sming/Sming'`
`/home/palha/git/Sming/Sming/Makefile-project.mk:404: recipe for target ``'/home/palha/git/Sming/Sming/compiler/lib//libsming.a' failed`
`make: *** [/home/palha/git/Sming/Sming/compiler/lib//libsming.a] Error 2`

What is wrong?

slaff, 2017-05-05:
> What is wrong?

Allow our Make system to patch the file. Try doing the following:

```
cd $SMING_HOME
make dist-clean
make
```

And then compile the sample

```
cd ../samples
cd Basic_Debug
make
```

fvpalha, 2017-05-05:
Thank you @slaff

closed 2017-05-05 by slaff