SmingHub/Sming #1126: After `make flash` problem

yeralin, 2017-05-10:
Hi,

I have NodeMCU ESP8266. I flashed it with Basic_WiFi sample. It flashed it successfully, and everything was working. I connected to my Wi-Fi and got IP address. I pressed CTRL+] and quit from terminal. ESP8266 was still working. But when I disconnected it from my computer and connected it back again. ESP8266 did not start working with flashed program as expected. It did nothing :( I tried to restart it, no luck.

It will start working again, only if I flash it again, up until I disconnect it. Seems like it resets or something. 
Maybe I'm doing something wrong?

Thank you!

ADiea, 2017-05-11:
Check if you have GP0 always connected to ground. It should stay at 3.3V when normal booting and 0V when programming

yeralin, 2017-05-11:
Yep, I figured it out last night :) GP0 was connected to the ground

closed 2017-05-11 by yeralin