SmingHub/Sming #1130: HttpServer/HttpClient suddenly stopped working

ZyperPL, 2017-05-19:
Hi
I've HttpServer and HttpClient copied from examples running on my ESP8266-01. HttpClient makes HTTPS request every  30 minutes. Also ESP is getting temperature from two DS18S20 sensors and saving it to spiffs.

I've got ESP running 3 days. After that time HttpClient and also HttpServer stopped working.
But when I was pinging it by ping command on my local network I've got response. Only thing that wasn't working were those Http things. 
After restarting ESP everything works. I also get temperature readings from the time when HttpServer and HttpClient were not responding. So ESP was running.

ESP was connected via 850mA USB charger with 3.3V. 

Does anyone know what might cause those things to stopped responding? 
What can I do to prevent that?

ADiea, 2017-05-19:
As a side note: On such a small embedded system I would reset it automatically every night at 2 am :)

slaff, 2017-05-19:
> Only thing that wasn't working were those Http things.

@ZyperPL Your description is way too broad to be able to figure out what could potentially be the problem. Please, paste here the logs that are in the serial console.  And don't forget to mention version of Sming, version of SDK, etc.

ZyperPL, 2017-05-19:
@ADiea Thanks for advice. :) 

@slaff I wish I had any logs, but it was connected to USB charger so I don't. 
That's why I'm writing this post. I thought that maybe someone would have any idea what might have caused this issue.

I'm using SDK 2.0.0(5a875ba) and Sming (develop bc64de4). Can those latest commits HttpServer make that instability?

It would be nice if you could aswer some of my questions:
How can I debug this issue? Do I need to set systemDebugOuput to true to get error messages or will they be printed by default? Maybe I would make it save some logs to a file, but I don't know what info should I save. 

Is it possible that problem might be related to ESP losing WiFi connection? Is server bound to only specific IP? Or can hardware be somehow broken?

slaff, 2017-05-21:
> How can I debug this issue? Do I need to set systemDebugOuput to true to get error messages or will they be printed by default?

You need to set the debug output. 
```C
Serial.begin(SERIAL_BAUD_RATE);
Serial.systemDebugOutput(true); // Allow debug print to serial
```
See [Basic_WebClient](https://github.com/SmingHub/Sming/blob/develop/samples/Basic_WebClient/app/application.cpp#L173)

After that recompile, flash and watch the logs (attach your USB-to-Serial to your computer and wait ). To make things faster decrease the 30 minutes interval to 1 minute (or less). Hopefully the issue will happen 30 times faster .

Once we have the logs we can analyze them. Meanwhile how can I reproduce your application? As far as I understood the HttpClient on the ESP sends requests to a SSL enabled remote SSL server. What is the HttpServer for? 

>  After that time HttpClient and also HttpServer stopped working.
Does that mean the the HttpClient stopped sending requests and trying to access the HttpServer was not possible from the same network? 

> Is it possible that problem might be related to ESP losing WiFi connection?
Can you send also the logs from your WIFI router so that we can see if there was disconnect or reconnect?

ZyperPL, 2017-06-03:
Ok, I was able to reproduce this crash two more times. It was very hard, because sometimes everything works for very long time (nearly 2 weeks) without any problems. But sometimes HTTP Server stops responding and Serial output gives me "CONNECTION DROPPED". 
I saved serial's output to file (it also crashed the whole ESP this time with "E:M 528" messages but I don't know if it is relevant). 
[httpcrash_connectiondropped.log.txt](https://github.com/SmingHub/Sming/files/1049646/httpcrash_connectiondropped.log.txt)
It starts at line 570. I've got those messages when I've tried to connect to my webpage (I've got: Connection was reset error).

I also checked my router logs and there was nothing wrong with internet connection.
This issue happens with or without SSL enabled.

ADiea, 2017-06-03:
Because of heap memory fragmentation over time and/or mem leaks you end up with only 1.8kB free heap RAM. Tcp will drop conn if mem is under a threshold i think 6k

ZyperPL, 2017-06-03:
@ADiea That might be it. 
Maybe it is related to https://github.com/SmingHub/Sming/issues/1138

slaff, 2017-06-04:
> Maybe it is related to #1138

Yes, that is very likely.

slaff, 2017-07-18:
@ZyperPL Can you check the latest `develop` version? We made a lot of changes there to optimize the code.

slaff, 2017-07-21:
Feel free to reopen the issue if the problem occurs again.

closed 2017-07-21 by slaff