SmingHub/Sming #1131: Install problems

hanpal, 2017-05-21:
You should check the quick install guides. Tried both on Mac and Ubuntu, did not succeed on any of them. Maybe this is a great product but it is way too hard to install so I guess I'll stick to the Arduino IDE.

hanpal, 2017-05-21:
This solved my problem for linux but it was rather complicated to find the usb device and sudo was required for make flash.

> Allow our Make system to patch the file. Try doing the following:
> 
> cd $SMING_HOME
> make dist-clean
> make

The guides should  be tried out to ensure that everything works without troubleshooting or guesswork.

kwis2, 2017-06-02:
In my opinion paradox is that Sming is very good for tool for beginner user, while to install it you have to be very advanced. It is a very weak point of Sming.... And for long time there is no solution.

slaff, 2018-02-09:
@hanpal @kwis2 We have updated the Mac OS X getting started guide and the precompiled Mac OS X toolchain.  Can you try to follow the steps in the [document](https://github.com/SmingHub/Sming/wiki/MacOS-Quickstart) and report your progress?

kwis2, 2018-02-10:
I have tested and since about 3 months nothing changes. It is impossible to install Sming according to #1297

slaff, 2018-02-10:
> It is impossible to install Sming according to 

@kwis2 Did you test it on Mac OS X? We updated the Mac OS X installation procedure only.

closed 2024-07-04 by mikee47