SmingHub/Sming #1133: Discussion: UDK 2.2.2 / SDK 2.1

tius2000, 2017-05-24:
Just update my windows machine to UDK 2.2.2 with SDK 2.1, seems to work fine (at least when using sming.mk, see #985). The new version of esptool.exe is included within the UDK. 

Are there any known issues with the new SDK?

icodk, 2017-05-24:
@tius2000 
Could you please share the steps you took for the upgrade
Thanks

tius2000, 2017-05-24:
@icodk 
I made a backup of c:\espressif, downloaded version 2.2.2 of the UDK (http://programs74.ru/udkew-en.html) and executed the setup file.

icodk, 2017-05-24:
@tius2000 Thanks
What about Sming?
Backup c:\tools and pulled/download developer/v3.2.0 ?

tius2000, 2017-05-24:
I did not change anything on my Sming setup (it should be almost up-to-date with the latest develop branch). However, I'm using the refactored makefile from #985.

icodk, 2017-05-24:
@tius2000 
Is the refactored makefile  in the development branch ?

tius2000, 2017-05-24:
I did not test with the normal makefile as my projects all use _sming.mk_ from #985. But I do not see any reasons why it should not work. 

The PR is not not merged yet. However, it does not replace any existing file so you may just give it a try ;-)

icodk, 2017-05-24:
@tius2000 
Will try
Thanks again

icodk, 2017-05-27:
I renamed C:\Espressif to  C:\Espressif_ but when I try to install the latest Unofficial Devkit 
However when I try to run the installation of the newest devkit (Espressif-ESP8266-DevKit-v2.2.1-x86) I get the following message:
An old version of Unofficial Development Kit for Espressif ESP8266 was detected. Do you want to uninstall it ?
If I click No the new devkit is not installed
What did you do to hide the older version ?

tius2000, 2017-05-28:
Dear @icodk, I made a backup of _c:\espressif_ first and let the installer then remove it and install the new version.

icodk, 2017-05-28:
@tius2000 
Can you now switch between the two versions by just restoring the old version ?
I am concerned by the possibility that none of them will work after the switch

tius2000, 2017-05-28:
Yes, I switched several times with no problems. If you want to be absolutely safe, make a backup of your system partition first.

icodk, 2017-05-28:
@tius2000 
Thanks, I will try it in a VM

icodk, 2017-05-31:
@tius2000 I tried with the  develop branch but got the errors I reported in #1140 
For some reason I get make[2]: xtensa-lx106-elf-gcc: Command not found  when compiling third-party libs
when I set ENABLE_SSL=1. may be your refactored makefile  can fix it. Where can I download it or better if I could clone the repository that works

slaff, 2018-10-09:
We will not support Esspressif's NON-OS SDK versions >= 2.1 and < 3.0. Once this #1470 is merged we will add support for SDK 3.0.

closed 2018-10-09 by slaff