SmingHub/Sming #1134: Discussion: Adding C++14/17 support with -std=c++1z

tius2000, 2017-05-24:
Would it be a good idea to allow some of the newer language features or are there reasons to stick with C++11? The only issue I found so far is that you need to add a missing function to _esp_plusplus.cpp_:

```
//	required when compiling with -std=c++14 or -std=c++1z
void operator delete(void* ptr, size_t n)
{
	if (ptr != NULL)
		free(ptr);
}
```

slaff, 2017-07-13:
> allow some of the newer language features or are there reasons to stick with C++11?

@tius2000 It would be great to use C++14/17 features as long as the GCC xtensa compiler supports them. Please, create a new PR and so that we can review it and evaluate it(compiled code size, performance and resource usage(esp. memory), etc.

jairov4, 2017-07-31:
I would love it

mikee47, 2019-06-30:
Current situation seems to be that C++14 and C++1z (17) work on all arches except Linux Esp8266 compiler, so we're stuck with C++11 for now. Haven't tried the Arduino compilers, they're probably more up to date.

mikee47, 2019-08-29:
Sming install provides Windows MinGW with GCC 6.3, Linux gets GCC 4.8.
It's the < GCC 5 which is preventing moving forwards, 6.3 is good enough for C++17
which I would like to see become the minimum standard.

Some impressive work has been done with the `esp quick toolchain`.
Current 2.5.0-4 release installs GCC 4.8.2 for all platforms, however source is
available and according to docs. will build for 4.8, 7.2, and 9.1.

I've tried the pre-release for Linux which includes GCC 9.1.0:

https://github.com/earlephilhower/esp-quick-toolchain/releases/tag/3.0.0-gnu1

Sming requires patching to compile, see #1825

mikee47, 2019-11-30:
The other thing the new toolchains offer is support for C++ exception handling. The Arduino framework offers this as an option, so we can do the same for those as wish to use it.

slaff, 2020-01-02:
See #1825.

closed 2020-01-02 by slaff