SmingHub/Sming #1136: Windows install problem

kwis2, 2017-05-25:
After some time I am going to return to sming. There is a lot of changes, but chocolatey problem still exists.
I have sming 3.1.1 instaled and I have tried to install or upgrade to newest version.

install log:
> 2017-05-25 11:30:00,507 [DEBUG] - _ Chocolatey:ChocolateyInstallCommand - Normal Run Mode _
> 2017-05-25 11:30:00,507 [INFO ] - Installing the following packages:
> 2017-05-25 11:30:00,507 [INFO ] - sming
> 2017-05-25 11:30:00,507 [INFO ] - By installing you accept licenses for the packages.
> 2017-05-25 11:30:00,897 [WARN ] - sming v2.1.0 already installed.
>  Use --force to reinstall, specify a version to install, or try upgrade.
> 2017-05-25 11:30:00,912 [WARN ] - 
> Chocolatey installed 0/1 packages. 0 packages failed.

and upgrade log

> 2017-05-25 11:40:18,282 [DEBUG] - _ Chocolatey:ChocolateyUpgradeCommand - Normal Run Mode _
> 2017-05-25 11:40:18,282 [INFO ] - Upgrading the following packages:
> 2017-05-25 11:40:18,297 [INFO ] - sming
> 2017-05-25 11:40:18,297 [INFO ] - By upgrading you accept licenses for the packages.
> 2017-05-25 11:40:20,123 [INFO ] - sming v2.1.0 is the latest version available based on your source(s).
> 2017-05-25 11:40:20,154 [WARN ] - 
> Chocolatey upgraded 0/1 packages. 0 packages failed.

And the most impotrnant question - how could I check what sming version I have?

riegaz, 2017-06-11:
Same problem for me. I cloned the develop branch but now I face a lot of undefined reference errors. I opened a issue, lets see how long it takes for an answer. I hope the community is more active than the one of Micropython

igormorse, 2017-08-05:
Chocolatey was too tricky and intrusive in my opinion. Several packages, hard to update.

The Docker solution worked fine for me, its clean, quick and easy enough for everyone.

Vity01, 2017-08-25:
@igormorse Not for everyone. My older (but still fast enough) CPU does not support special instructions set for virtualization....
Also using Docker adds additionally complexity for users. Standard Windows distribution should be preserved.

igormorse, 2017-08-31:
@Vity01 I not saying to take down the Windows Version and I don't agree with the additional complexity. If you install the Docker Toolkit you are ready to go with docker, yes if your machine doesn't support virtualization ( almost any newer CPU support those instructions) then it's not an option. 

The main downside on the window version is the amount of stuff that it does to your personal computer ( cresting environment variables, setting up a lot of stuff) and if something goes wrong you'll have a tough time fixing it. As docker is an sandbox environment you can do whatever you want and if something gone wrong you just erase the container and install a fresh one again, pretty clean to me. 

Docker seems tricky at first but when you get into it it's a great tool.