SmingHub/Sming #1137: 1MB FLASH rboot problem

kwis2, 2017-05-25:
I am trying to modyfy Basic_rBoot sample to work with ESP8266-01 with 8Mb FLASH (2 x 512kB) - RBOOT_BIG_FLASH ?= 0 and RBOOT_TWO_ROMS  ?= 1
I think with sming version changes the rom0.ld was moved to sming/compiler/ld, renamed and reduced in size. So simple  `Copy rom0.ld to rom1.ld`  from readme.txt doesn't work. 
In my oppinion there should be `Makefile-rboot.mk` modified `RBOOT_LD_1 ?= rboot.rom1.ld`. 
I am not sure if copy `rboot.rom0.ld` to `rboot.rom1.ld` which is much smaler than old `rom0.ld` file. 
Finally question from beginer - how should I modyfy `rboot.rom0.ld` nad `rboot.rom1.ld` to work.
Are you able to prepare working example of `Basic_rBoot` for ESP8266-01

kwis2

mr-sven, 2017-06-24:
I faced out the same Issue. I've a ESP8285 with integrated 1MB Flash. At first i struggled over it by using OTA.

If I flash the same ROM to slot 0 and 1 I can boot from both. If I flash a different ROM to slot 1 I cannot boot from slot 1, it will fail. Sometimes it hangs or it will show exceptions.

I think there is a mapping problem. If I boot from slot 1 and try to update slot 0 with OTA the system will crash after erase of the sector 2.

mr-sven, 2017-06-24:
Ok, I've found the issue. I copied `rboot.rom0.ld` to `rboot.rom1.ld` and modified the `Makefile-rboot.mk`. I modified `irom0_0_seg` of the rom1 file to `org = 0x40282010, len = (1M - 0x2010)`.

I'm not firm with ld files, maybe there could be an option to modify the linker file automatically or dynamic create the file.

mr-sven, 2017-06-24:
ok, Update on this. It is enough to copy the `rboot.rom0.ld` as `rom1.ld` to your project root and modify the `irom0_0_seg` depending on your flash size: (https://github.com/raburton/rboot#linking-user-code)