SmingHub/Sming #1138: HttpClient.downloadString() is leaking

midiway, 2017-05-28:
```
#include <user_config.h>
#include <SmingCore/SmingCore.h>

#define WIFI_SSID "NETN16"
#define WIFI_PWD "netdomidi"

Timer pushTimer;
HttpClient httpPusher;

int onDataSent(HttpConnection& client, bool successful)
{
	if(successful)
	{
		Serial.println("###### Success sent");
		Serial.println("###### Success sent");
		Serial.println("###### Success sent");
	} else {
		Serial.println("###### Failed");
		Serial.println("###### Failed");
		Serial.println("###### Failed");
	}
	return 0;
}

void pushItTime()
{
	Serial.println("###### downloadString");
	Serial.println("###### downloadString");
	Serial.println("###### downloadString");
	httpPusher.downloadString("http://pudim.com.br/", onDataSent);
	
	Serial.print("Free memory: ");
	Serial.println(system_get_free_heap_size());// SEE, FREE MEMORY ALWAYS DECREASING!!
}

void STAGotIP(IPAddress ip, IPAddress mask, IPAddress gateway)
{
	Serial.printf("GOTIP - IP: %s, MASK: %s, GW: %s\n", ip.toString().c_str(),
																mask.toString().c_str(),
																gateway.toString().c_str());
    
	pushTimer.initializeMs(1000, pushItTime).start();
}

void init()
{
	Serial.begin(SERIAL_BAUD_RATE); // 115200 by default
	Serial.systemDebugOutput(true);
	
	WifiStation.config(WIFI_SSID, WIFI_PWD);
	WifiStation.enable(true);
	WifiAccessPoint.enable(false);
	WifiEvents.onStationGotIP(STAGotIP);
}
```
I am making a HttpClient request every second, and free memory is constantly decreasing.
Or I am doing something stupid..

slaff, 2017-05-29:
@midiway I will be able to test this on a real device at the end of the week. Until then can you send me the log from the serial console. Also did you compile Sming with SSL support of without it? What is the SDK that you have used?

midiway, 2017-05-29:
I don't know how discover the SDK version.
I am using the HEAD commit of this repository...
```
connected with NETN16, channel 6
dhcp client start...
connect to ssid NETN16, channel 6

ip:192.168.2.46,mask:255.255.255.0,gw:192.168.2.1
ip:192.168.2.46,mask:255.255.255.0,gw:192.168.2.1

GOTIP - IP: 192.168.2.46, MASK: 255.255.255.0, GW: 192.168.2.1
###### downloadString
###### downloadString
###### downloadString
Creating new httpConnection
HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
HttpConnection::connecting ...
+TCP connection
connect to: pudim.com.br
Free memory: 44776
DNS record found: pudim.com.br = 54.207.20.104
TcpConnection::connect result:, 0
OnConnected
HttpConnection::onConnected: waitingQueue.count: 1
Storing 36 bytes in stream
realloc 36 -> 184
Storing 19 bytes in stream
Storing 2 bytes in stream
TCP connected
onReadyToSendData: 0
Written: 57, Available: 57, isFinished: 1, PushCount: 1
TcpClient request completed
TCP sent: 57
onReadyToSendData: 2
The headers are complete
staticOnMessageComplete: Execution queue: 0, http://pudim.com.br:80/
###### Success sent
###### Success sent
###### Success sent
HttpConnection::onConnected: waitingQueue.count: 0
TCP connected
onReadyToSendData: 0
TCP received: 1106 bytes
onReadyToSendData: 1
###### downloadString
###### downloadString
###### downloadString
HttpConnection::connect: TCP state: 4, isStarted: 1, isActive: 1
Free memory: 43464
###### downloadString
###### downloadString
###### downloadString
HttpConnection::connect: TCP state: 4, isStarted: 1, isActive: 1
Free memory: 43160
###### downloadString
###### downloadString
###### downloadString
HttpConnection::connect: TCP state: 4, isStarted: 1, isActive: 1
Free memory: 42872
###### downloadString
###### downloadString
###### downloadString
HttpConnection::connect: TCP state: 4, isStarted: 1, isActive: 1
Free memory: 42608
###### downloadString
###### downloadString
###### downloadString
HttpConnection::connect: TCP state: 4, isStarted: 1, isActive: 1
Free memory: 42344
TcpConnection::staticOnReceive: pbuf is NULL
TCP received: (null)
TCP connection closing
-TCP connection
###### downloadString
###### downloadString
###### downloadString
Removing stale connection: State: 3, Active: 0
Creating new httpConnection
HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
HttpConnection::connecting ...
+TCP connection
connect to: pudim.com.br
Free memory: 41352
DNS record found: pudim.com.br = 54.207.20.104
TcpConnection::connect result:, 0
OnConnected
HttpConnection::onConnected: waitingQueue.count: 6
Storing 36 bytes in stream
realloc 36 -> 184
Storing 19 bytes in stream
Storing 2 bytes in stream
Storing 36 bytes in stream
Storing 19 bytes in stream
Storing 2 bytes in stream
Storing 36 bytes in stream
Storing 19 bytes in stream
Storing 2 bytes in stream
realloc 171 -> 336
Storing 36 bytes in stream
Storing 19 bytes in stream
Storing 2 bytes in stream
Storing 36 bytes in stream
Storing 19 bytes in stream
Storing 2 bytes in stream
Storing 36 bytes in stream
realloc 321 -> 405
Storing 19 bytes in stream
Storing 2 bytes in stream
TCP connected
onReadyToSendData: 0
Written: 342, Available: 342, isFinished: 1, PushCount: 1
TcpClient request completed
TCP sent: 342
onReadyToSendData: 2
The headers are complete
staticOnMessageComplete: Execution queue: 5, http://pudim.com.br:80/
###### Success sent
###### Success sent
###### Success sent
TCP received: 1106 bytes
onReadyToSendData: 1
The headers are complete
staticOnMessageComplete: Execution queue: 4, http://pudim.com.br:80/
###### Success sent
###### Success sent
###### Success sent
TCP received: 1106 bytes
onReadyToSendData: 1
The headers are complete
staticOnMessageComplete: Execution queue: 3, http://pudim.com.br:80/
###### Success sent
###### Success sent
###### Success sent
TCP received: 1106 bytes
onReadyToSendData: 1
The headers are complete
staticOnMessageComplete: Execution queue: 2, http://pudim.com.br:80/
###### Success sent
###### Success sent
###### Success sent
TCP received: 1106 bytes
onReadyToSendData: 1
The headers are complete
staticOnMessageComplete: Execution queue: 1, http://pudim.com.br:80/
###### Success sent
###### Success sent
###### Success sent
TCP received: 1106 bytes
onReadyToSendData: 1
The headers are complete
staticOnMessageComplete: Execution queue: 0, http://pudim.com.br:80/
###### Success sent
###### Success sent
###### Success sent
HttpConnection::onConnected: waitingQueue.count: 0
TCP connected
onReadyToSendData: 0
TCP received: 1106 bytes
onReadyToSendData: 1
###### downloadString
###### downloadString
###### downloadString
HttpConnection::connect: TCP state: 4, isStarted: 1, isActive: 1
Free memory: 41488
###### downloadString
###### downloadString
###### downloadString
HttpConnection::connect: TCP state: 4, isStarted: 1, isActive: 1
Free memory: 41216
pm open,type:2 0
###### downloadString
###### downloadString
###### downloadString
HttpConnection::connect: TCP state: 4, isStarted: 1, isActive: 1
Free memory: 40160
###### downloadString
###### downloadString
###### downloadString
HttpConnection::connect: TCP state: 4, isStarted: 1, isActive: 1
Free memory: 39896
```

slaff, 2017-05-30:
Thanks for the information. Can you tell me also if you compiled Sming with SSL support of without it?

midiway, 2017-05-30:
without

slaff, 2017-06-19:
@midiway Can you update Sming to its latest version from develop and try again?

midiway, 2017-06-19:
sure

midiway, 2017-06-21:
Just tested it, it is still leaking, though much more slowly, 32 bytes to be precise, I think

However, the most strange thing is that the onDataSent callback is not called immediatelly, it takes a while...
```
mode : sta(5c:cf:7f:86:3c:fe)
add if0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 6
cnt 

connected with NETN16, channel 6
dhcp client start...
connect to ssid NETN16, channel 6

ip:192.168.2.17,mask:255.255.255.0,gw:192.168.2.1
ip:192.168.2.17,mask:255.255.255.0,gw:192.168.2.1

GOTIP - IP: 192.168.2.17, MASK: 255.255.255.0, GW: 192.168.2.1
###### downloadString
###### downloadString
###### downloadString
Creating new httpConnection
HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
HttpConnection::connecting ...
+TCP connection
connect to: pudim.com.br
Free memory: 41064
DNS record found: pudim.com.br = 54.207.20.104
TcpConnection::connect result:, 0
OnConnected
staticOnConnected: useSSL: 0, Error: 0
HttpConnection::onConnected: waitingQueue.count: 1
Storing 36 bytes in stream
realloc 36 -> 184
Storing 19 bytes in stream
Storing 2 bytes in stream
Nothing in the waiting queue
TCP connected
onReadyToSendData: 0
Written: 57, Available: 57, isFinished: 1, PushCount: 1
TcpClient request completed
TCP sent: 57
onReadyToSendData: 2
The headers are complete
staticOnMessageComplete: Execution queue: 0, http://pudim.com.br:80/
###### Success sent
###### Success sent
###### Success sent
HttpConnection::onConnected: waitingQueue.count: 0
Nothing in the waiting queue
TCP connected
onReadyToSendData: 0
TCP received: 1106 bytes
onReadyToSendData: 1
###### downloadString
###### downloadString
###### downloadString
HttpConnection::connect: TCP state: 4, isStarted: 1, isActive: 1
Free memory: 39848
TcpConnection::staticOnReceive: pbuf is NULL
TCP received: (null)
TCP connection closing
-TCP connection
pm open,type:2 0
###### downloadString
###### downloadString
###### downloadString
Removing stale connection: State: 3, Active: 0
~TCP connection
Creating new httpConnection
HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
HttpConnection::connecting ...
+TCP connection
connect to: pudim.com.br
Free memory: 39896
DNS record found: pudim.com.br = 54.207.20.104
TcpConnection::connect result:, 0
OnConnected
staticOnConnected: useSSL: 0, Error: 0
HttpConnection::onConnected: waitingQueue.count: 2
Storing 36 bytes in stream
realloc 36 -> 184
Storing 19 bytes in stream
Storing 2 bytes in stream
Storing 36 bytes in stream
Storing 19 bytes in stream
Storing 2 bytes in stream
Nothing in the waiting queue
TCP connected
onReadyToSendData: 0
Written: 114, Available: 114, isFinished: 1, PushCount: 1
TcpClient request completed
TCP sent: 114
onReadyToSendData: 2
The headers are complete
staticOnMessageComplete: Execution queue: 1, http://pudim.com.br:80/
###### Success sent
###### Success sent
###### Success sent
TCP received: 1106 bytes
onReadyToSendData: 1
The headers are complete
staticOnMessageComplete: Execution queue: 0, http://pudim.com.br:80/
###### Success sent
###### Success sent
###### Success sent
HttpConnection::onConnected: waitingQueue.count: 0
Nothing in the waiting queue
TCP connected
onReadyToSendData: 0
TCP received: 1106 bytes
onReadyToSendData: 1
###### downloadString
###### downloadString
###### downloadString
HttpConnection::connect: TCP state: 4, isStarted: 1, isActive: 1
Free memory: 38976
TcpConnection::staticOnReceive: pbuf is NULL
TCP received: (null)
TCP connection closing
-TCP connection
###### downloadString
###### downloadString
###### downloadString
Removing stale connection: State: 3, Active: 0
~TCP connection
Creating new httpConnection
HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
HttpConnection::connecting ...
+TCP connection
connect to: pudim.com.br
Free memory: 39840
DNS record found: pudim.com.br = 54.207.20.104
TcpConnection::connect result:, 0
OnConnected
staticOnConnected: useSSL: 0, Error: 0
HttpConnection::onConnected: waitingQueue.count: 2
Storing 36 bytes in stream
realloc 36 -> 184
Storing 19 bytes in stream
Storing 2 bytes in stream
Storing 36 bytes in stream
Storing 19 bytes in stream
Storing 2 bytes in stream
Nothing in the waiting queue
TCP connected
onReadyToSendData: 0
Written: 114, Available: 114, isFinished: 1, PushCount: 1
TcpClient request completed
TCP sent: 114
onReadyToSendData: 2
The headers are complete
staticOnMessageComplete: Execution queue: 1, http://pudim.com.br:80/
###### Success sent
###### Success sent
###### Success sent
TCP received: 1106 bytes
onReadyToSendData: 1
The headers are complete
staticOnMessageComplete: Execution queue: 0, http://pudim.com.br:80/
###### Success sent
###### Success sent
###### Success sent
HttpConnection::onConnected: waitingQueue.count: 0
Nothing in the waiting queue
TCP connected
onReadyToSendData: 0
TCP received: 1106 bytes
onReadyToSendData: 1
```

PORT CLOSED

slaff, 2017-06-21:
> However, the most strange thing is that the onDataSent callback is not called immediatelly, it takes a while...

All requests for the same domain and port are queued until the connection is established (DNS resolution + TCP connection). Once that happens all requests are grouped (pipelined if they are only GET requests) and then the respective callbacks called.

> Just tested it, it is still leaking, though much more slowly, 32 bytes to be precise, I think

That is most probably the memory needed to keep the queue(s) and tracking.  I bet that if you run this app for some hours the memory consumption will stay within a reasonable range.

midiway, 2017-06-21:
so there is DNS no cache in Sming?

slaff, 2017-06-22:
> so there is no DNS cache in Sming?

There is DNS cache in LWIP. And yes, Sming has DNS cache too because it uses LWIP.

Let me try to explain what is happening using the log data that you have pasted. You will see first an excerpt from you log containing...

```
logged commands
```

Followed directly by explanation.

# Start

```
GOTIP - IP: 192.168.2.17, MASK: 255.255.255.0, GW: 192.168.2.1
```

Your ESP is connected to the router and gets and IP from the DHCP. Your application is ready to start sending requests.

```
###### downloadString
###### downloadString
###### downloadString
Creating new httpConnection
HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
HttpConnection::connecting ...
+TCP connection
```

The first http request is scheduled. Since this is the first request a new TCP connection needs to be established.

```
connect to: pudim.com.br
Free memory: 41064
DNS record found: pudim.com.br = 54.207.20.104
TcpConnection::connect result:, 0
OnConnected
staticOnConnected: useSSL: 0, Error: 0
HttpConnection::onConnected: waitingQueue.count: 1
```

For the TCP connection we need the IP address of the domain. This is done ONCE and when it is ready we are connected and can send the queued requests. Up to now we have only 1 waiting request.

```
Storing 36 bytes in stream
realloc 36 -> 184
Storing 19 bytes in stream
Storing 2 bytes in stream
Nothing in the waiting queue
TCP connected
onReadyToSendData: 0
Written: 57, Available: 57, isFinished: 1, PushCount: 1
TcpClient request completed
TCP sent: 57
onReadyToSendData: 2
```

The request is sent to the remote server and we are ready to receive the response.

```
The headers are complete
staticOnMessageComplete: Execution queue: 0, http://pudim.com.br:80/
###### Success sent
###### Success sent
###### Success sent
HttpConnection::onConnected: waitingQueue.count: 0
Nothing in the waiting queue
TCP connected
onReadyToSendData: 0
TCP received: 1106 bytes
onReadyToSendData: 1
```
The headers arrive followed by the body. That request is finished and there are no pending requests in the waiting queue.

```
###### downloadString
###### downloadString
###### downloadString
HttpConnection::connect: TCP state: 4, isStarted: 1, isActive: 1
Free memory: 39848
TcpConnection::staticOnReceive: pbuf is NULL
TCP received: (null)
TCP connection closing
-TCP connection
pm open,type:2 0
```

Every second a new request is triggered.  The http client tries to reuse the existing TCP connection to send the request but the remote side has closed the connection (probably due to short keep-alive). This request unfortunately is still pending because it needs a new TCP connection.

```
###### downloadString
###### downloadString
###### downloadString
Removing stale connection: State: 3, Active: 0
~TCP connection
```
Another second. One old and one new request are now in the queue. The old HttpConnection object is discarded because the TCP connection is no longer active.

```
Creating new httpConnection
HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
HttpConnection::connecting ...
+TCP connection
connect to: pudim.com.br
Free memory: 39896
DNS record found: pudim.com.br = 54.207.20.104
TcpConnection::connect result:, 0
OnConnected
staticOnConnected: useSSL: 0, Error: 0
HttpConnection::onConnected: waitingQueue.count: 2
```

A new TCP connection is created. DNS resolution uses cache and soon we are connected. At that moment we have two pending requests in the queue. 

```
Storing 36 bytes in stream
realloc 36 -> 184
Storing 19 bytes in stream
Storing 2 bytes in stream
Storing 36 bytes in stream
Storing 19 bytes in stream
Storing 2 bytes in stream
Nothing in the waiting queue
TCP connected
onReadyToSendData: 0
Written: 114, Available: 114, isFinished: 1, PushCount: 1
TcpClient request completed
TCP sent: 114
onReadyToSendData: 2
The headers are complete
staticOnMessageComplete: Execution queue: 1, http://pudim.com.br:80/
###### Success sent
###### Success sent
###### Success sent
TCP received: 1106 bytes
onReadyToSendData: 1
The headers are complete
staticOnMessageComplete: Execution queue: 0, http://pudim.com.br:80/
###### Success sent
###### Success sent
###### Success sent
```

Since both of the requests are using the HTTP GET method they are pipelined and sent to the remote server. The remote server supports pipelining and returns the response to both requests directly using this one TCP connection.

```
HttpConnection::onConnected: waitingQueue.count: 0
Nothing in the waiting queue
TCP connected
onReadyToSendData: 0
TCP received: 1106 bytes
onReadyToSendData: 1
```

The waiting queue is empty and there are no pending requests that need to be sent or processed.

```
###### downloadString
###### downloadString
###### downloadString
HttpConnection::connect: TCP state: 4, isStarted: 1, isActive: 1
Free memory: 38976
TcpConnection::staticOnReceive: pbuf is NULL
TCP received: (null)
TCP connection closing
-TCP connection
```

The story repeats...

@midiway Is it now more clear for you what is happening behind the hood?

midiway, 2017-06-22:
well, I won't bother with the details, gonna make a stress test to check if it is ok!

slaff, 2017-07-10:
> gonna make a stress test to check if it is ok!

@midiway Feel free to reopen the ticket if the issue persists.

closed 2017-07-10 by slaff