SmingHub/Sming #1151: Sming build fails with Espressif SDK 2.1.0

zgoda, 2017-06-13:
```
[...]
CC system/flashmem.c
In file included from include/user_config.h:29:0,
                 from system/flashmem.h:11,
                 from system/flashmem.c:1:
system/include/esp_systemapi.h:47:13: error: conflicting types for 'ets_timer_arm_new'
 extern void ets_timer_arm_new(ETSTimer *ptimer, uint32_t milliseconds, bool repeat_flag, int isMstimer);
             ^
In file included from system/include/esp_systemapi.h:7:0,
                 from include/user_config.h:29,
                 from system/flashmem.h:11,
                 from system/flashmem.c:1:
/home/jazg/esp8266/esp-open-sdk/sdk/include/osapi.h:65:6: note: previous declaration of 'ets_timer_arm_new' was here
 void ets_timer_arm_new(os_timer_t *ptimer, uint32_t time, bool repeat_flag, bool ms_flag);
      ^
```
Same for `ets_delay_us` and couple other `ets_*` functions.

slaff, 2017-06-13:
@zgoda Can you create a PR with the changes that should focus on SDK 2.1.0 support?

avr39-ripe, 2017-06-13:
Yesterday I managed to patch whole Sming to let it compile with sdk 2.1.0 try to make pr for this

avr39-ripe, 2017-06-13:
PR created #1153 ;)

slaff, 2017-11-28:
With that commit https://github.com/SmingHub/Sming/commit/c2cc852e40b1971a1987aceeedab5b28b8c7297c it is now possible to use SDK 2.1.0.

closed 2017-11-28 by slaff