SmingHub/Sming #1155: Reboots on tcpClient

piperpilot, 2017-06-13:
Switching to latest SMING branch, I am getting application reboots consistently.  I believe it is related to TCPClient.  Here's the code I am executing:

`void sendDataPoint(String json) {

	HttpHeaders requestHeaders;
	requestHeaders["ContentType"] = "application/json";
	requestHeaders["Accept"] = "application/json";

	String url = smobotStats;
	url += "/api/datapoint/";
	url += system_get_chip_id();
	datapush.send(
			datapush.request(url)
			->setMethod(HTTP_POST)
			->setHeaders(requestHeaders)
			->onRequestComplete(processed)
			->setBody(json)

	);
}`

Here is the serial debug log:
`Removing stale connection: State: 3, Active: 0
Creating new httpConnection
HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
HttpConnection::connecting ...
+TCP connection
connect to: mysmobot.smobot.com
DNS record found: mysmobot.smobot.com = 54.175.88.36
TcpConnection::connect result:, 0
OnConnected
HttpConnection::onConnected: waitingQueue.count: 2
Storing 66 bytes in stream
realloc 66 -> 226
Storing 31 bytes in stream
Storing 26 bytes in stream
Storing 21 bytes in stream
Storing 2 bytes in stream
realloc 146 -> 370
Storing 159 bytes in stream
Stream completed
TCP connected
onReadyToSendData: 0
Written: 305, Available: 305, isFinished: 1, PushCount: 1
TcpClient request completed
TCP sent: 305
onReadyToSendData: 2
The headers are complete
On chunk header
On chunk complete
On chunk header
On chunk complete
staticOnMessageComplete: Execution queue: 0, http://mysmobot.smobot.com:80/api/datapoint/16699536
HttpConnection::onConnected: waitingQueue.count: 1
Storing 66 bytes in stream
realloc 66 -> 226
Storing 31 bytes in stream
Storing 26 bytes in stream
Storing 21 bytes in stream
Storing 2 bytes in stream
realloc 146 -> 370
Storing 159 bytes in stream
Stream completed
TCP connected
onReadyToSendData: 0
Written: 305, Available: 305, isFinished: 1, PushCount: 1
TcpClient request completed
HTTP parser error: HPE_CB_message_complete
TCP received: (null)
TCP sent: 305
onReadyToSendData: 2 `

Attached is the decoded stacktrace.

[decodedStacktrace.txt](https://github.com/SmingHub/Sming/files/1071701/decodedStacktrace.txt)

Any idea what might be going on?

slaff, 2017-06-13:
```
0x40235c86: m_vprintf at /Users/cpope/Development/Sming/Sming//system/m_printf.cpp:77
0x402401c5: HttpConnection::onReceive(pbuf*) at /Users/cpope/Development/Sming/Sming//SmingCore/Network/Http/HttpConnection.cpp:514
```
Paste also the code of the `processed` function. Did you make sure to return 0 on success?

piperpilot, 2017-06-13:
processed doens't really do anything...where do I need to return 0?

`int processed(HttpConnection& connection, bool success)
{
//	Serial.println(client.getResponseString());
}`

just return 0 inside this function on success?

slaff, 2017-06-13:
> just return 0 inside this function on success?

Yes. the example above should be changed to

```
int processed(HttpConnection& connection, bool success)
{
//	Serial.println(client.getResponseString()); // Note: `client` is no longer present. Use `connection` instead.

        return 0; // on success.
}

```

slaff, 2017-06-13:
BUT my wild guess is that the issue is related to a print statement in the `processed` callback (0x40235c86: m_vprintf at /Users/cpope/Development/Sming/Sming//system/m_printf.cpp:77). If you can present  sample code for the `processed` then I can help more.  If that is not possible just comment out all lines inside of the callback except `return 0` and test. Compile and test. If the modified application is working then try to find the offending line inside of the `processed` callback.

piperpilot, 2017-06-13:
processed is pasted above.  It is basically an empty function, the print statement is commented out.  That got me past that issue, now running into problems with the httpServer.  Its running out of memory loading a pretty simple page that worked in previous versions.  Have you tested the httpConfigNetwork example with the latest changes?

slaff, 2017-06-13:
> That got me past that issue, 

Glad to hear that. Make sure to close the issue.

> ... now running into problems with the httpServer.

Open another issue and try to explain as best as possible how to reproduce the issue.

closed 2017-06-13 by piperpilot