SmingHub/Sming #1157: httpServer reboots - looks like low memory

piperpilot, 2017-06-13:
My application uses a modified version of httpdConfigNetwork sample.  When loading pages with lots of assets, it doesn't complete and sometimes reboots the ESP.  It looks like it is running out of memory.  I can see this happen when hitting the settings page on my app as well as my main page.  I haven't tried the base example yet.  Here's the log:

TCP sent: 2780
Written: 1024, Available: 1024, isFinished: 0, PushCount: 1
Written: 1024, Available: 1024, isFinished: 0, PushCount: 2
Written: 732, Available: 732, isFinished: 0, PushCount: 3
onReadyToSendData: 2
TCP sent: 2780
Written: 1024, Available: 1024, isFinished: 0, PushCount: 1
Written: 1024, Available: 1024, isFinished: 0, PushCount: 2
Written: 732, Available: 732, isFinished: 0, PushCount: 3
onReadyToSendData: 2
-TCP connection
TCP sent: 2780
Written: 1024, Available: 1024, isFinished: 0, PushCount: 1
Written: 1024, Available: 1024, isFinished: 0, PushCount: 2
Written: 732, Available: 732, isFinished: 0, PushCount: 3
onReadyToSendData: 2
TCP sent: 2780
Written: 1024, Available: 1024, isFinished: 0, PushCount: 1
Written: 970, Available: 970, isFinished: 1, PushCount: 2
Body stream completed
onReadyToSendData: 2
TCP sent: 2780
Written: 1024, Available: 1024, isFinished: 0, PushCount: 1
Written: 1024, Available: 1024, isFinished: 0, PushCount: 2
Written: 732, Available: 732, isFinished: 0, PushCount: 3
onReadyToSendData: 2
TCP sent: 1994
onReadyToSendData: 2
The headers are complete
found jquery.js.gz
attached file: jquery.js.gz (30153 bytes)
TCP received: 364 bytes
Written: 1024, Available: 1024, isFinished: 0, PushCount: 1
Written: 1024, Available: 1024, isFinished: 0, PushCount: 2
Written: 530, Available: 530, isFinished: 0, PushCount: 3
onReadyToSendData: 1

CONNECTION DROPPED
.(3712)

TCP sent: 2780
Written: 1024, Available: 1024, isFinished: 0, PushCount: 1
Written: 1024, Available: 1024, isFinished: 0, PushCount: 2
Written: 732, Available: 732, isFinished: 0, PushCount: 3
onReadyToSendData: 2
TCP sent: 993
onReadyToSendData: 2
TCP sent: 2780
Written: 1024, Available: 1024, isFinished: 0, PushCount: 1
Written: 1024, Available: 1024, isFinished: 0, PushCount: 2
Written: 732, Available: 732, isFinished: 0, PushCount: 3
onReadyToSendData: 2
TCP sent: 2780
Written: 1024, Available: 1024, isFinished: 0, PushCount: 1
Written: 1024, Available: 1024, isFinished: 0, PushCount: 2
Written: 732, Available: 732, isFinished: 0, PushCount: 3
onReadyToSendData: 2
TCP sent: 2780
Written: 1024, Available: 1024, isFinished: 0, PushCount: 1
Written: 1024, Available: 1024, isFinished: 0, PushCount: 2
Written: 732, Available: 732, isFinished: 0, PushCount: 3
onReadyToSendData: 2

CONNECTION DROPPED
.(5640)

TCP sent: 2780
Written: 1024, Available: 1024, isFinished: 0, PushCount: 1
Written: 1024, Available: 1024, isFinished: 0, PushCount: 2
Written: 732, Available: 732, isFinished: 0, PushCount: 3
onReadyToSendData: 2
TCP sent: 2780
Written: 1024, Available: 1024, isFinished: 0, PushCount: 1
Written: 1024, Available: 1024, isFinished: 0, PushCount: 2
Written: 732, Available: 732, isFinished: 0, PushCount: 3
onReadyToSendData: 2
TCP sent: 2780
Written: 1024, Available: 1024, isFinished: 0, PushCount: 1
Written: 1024, Available: 1024, isFinished: 0, PushCount: 2
Written: 732, Available: 732, isFinished: 0, PushCount: 3
onReadyToSendData: 2
TCP sent: 2780
Written: 1024, Available: 1024, isFinished: 0, PushCount: 1
Written: 1024, Available: 1024, isFinished: 0, PushCount: 2
Written: 732, Available: 732, isFinished: 0, PushCount: 3
onReadyToSendData: 2

CONNECTION DROPPED
.(5640)

TCP sent: 2780
Written: 1024, Available: 1024, isFinished: 0, PushCount: 1
Written: 1024, Available: 1024, isFinished: 0, PushCount: 2
Written: 732, Available: 732, isFinished: 0, PushCount: 3
onReadyToSendData: 2

Its every inconsistent, but definitely doesn't load as well as the previous httpServer.

I will try the httpConfigNetwork example and see what it does a little later, but since my code is almost identical to that I suspect it will do the same thing.

slaff, 2017-08-01:
@piperpilot Did you had the chance to try your app with the latest Sming version?

slaff, 2017-11-13:
Feel free to reopen it if you have tested the latest Sming version and your problems still persist.

closed 2017-11-13 by slaff