SmingHub/Sming #1158: httpServer_ConfigNetwork example Detected overlap at address: 0xa000

piperpilot, 2017-06-13:
When building httpServer_ConfigNetwork, you get the following error:

/opt/esp-alt-sdk//utils/esptool.py -p /dev/tty.usbserial-A50285BI -b 115200 write_flash -ff 40m -fm qio -fs 4m 0x00000 out/firmware/0x00000.bin 0x0a000 out/firmware/0x0a000.bin  0x4C000 out/firmware/spiff_rom.bin
usage: esptool write_flash [-h] [--flash_freq {40m,26m,20m,80m}]
WARNING: Flash size arguments in megabits like '4m' are deprecated.
Please use the equivalent size '512KB'.
Megabit arguments may be removed in a future release.
                           [--flash_mode {qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE] [--ucIsHspi]
                           [--ucIsLegacy] [--no-progress] [--verify]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0xa000 for file: out/firmware/0x0a000.bin
make: *** [flash] Error 2

slaff, 2017-06-14:
> Please use the equivalent size '512KB'.

@piperpilot What is the size of your flash? If it is 512KB then you have little chance to put rom and spiffs. If the size is wrong then make sure to set it before you flash the application. Example:

```
export SPI_SIZE=4M # for 4 megabytes
make flashinit # (Optional) Might be needed if you have the wifi settings in the wrong location
make flash
```

piperpilot, 2017-06-14:
I'm using a 4MB flash.  I have the size set in my makefile like:
`
SPIFF_FILES = web/build
ESPTOOL		 ?= $(ESP_HOME)/utils/esptool.py
COM_PORT = /dev/tty.usbserial-A50285BI
SPI_MODE ?= qio
SPI_SIZE        = 4M
`
And still get an overlap message:
`esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0xa000 for file: out/firmware/0x0a000.bin
make: *** [flash] Error 2`

slaff, 2017-06-14:
I cannot reproduce the issue with the overlapping.

Here is the output from my flashing:

```
/home/slavey/dev/esp8266.dev.box/dev/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm dio -fs 32m 0x00000 out/firmware/0x00000.bin 0x0a000 out/firmware/0x0a000.bin  0x4E000 out/firmware/spiff_rom.bin
```

Notice that I have `-fs 32m` (your one in the output pasted in your comment is `-fs 4m `. This setting instructs the flashing tool that I have 4 MegaBytes ( and you have 512 KiloBytes) . Also the spiff offset, based on the SPI_SIZE is calculated to be ` 0x4E000`. Your one is ` 0x4C000`

piperpilot, 2017-06-14:
I will look into it more.  It builds fine on the master branch, but not on current develop, so something has changed.

piperpilot, 2017-06-16:
Slaff,  I have no idea whats going on here.  Here is the compile:

15:41:41 **** Build of configuration Sming for project HttpServer_ConfigNetwork ****
make all 
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app.out

#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8384|     900
    rodata|           ReadOnly Data (RAM)|    3FFE8390|    3FFEB1CC|   11836
       bss|      Uninitialized Data (RAM)|    3FFEB1D0|    3FFF1F58|   28040
      text|            Cached Code (IRAM)|    40100000|    4010702B|   28715
irom0_text|           Uncached Code (SPI)|    4020A000|    4024B692|  267922
Total Used RAM : 40776
Free RAM : 41144
Free IRam : 4071
------------------------------------------------------------------------------
# Generating image...
Generate firmware images successully in folder out/firmware.
Done
# Generating spiffs_bin
Checking for spiffs files
web/build directory exists. Creating out/firmware/spiff_rom.bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x80000 (524288) bytes.
Adding files in directory 'web/build'.
Added 'bootstrap.css.gz' to spiffs (15615 bytes).
Added 'index.html' to spiffs (12168 bytes).
Added 'jquery.js.gz' to spiffs (30153 bytes).
Added 'settings.html' to spiffs (4096 bytes).
Added 'style.css' to spiffs (7329 bytes).
Added 'wifi-sprites.png' to spiffs (1769 bytes).
out/firmware/spiff_rom.bin---------->0x4C000

15:41:43 Build Finished (took 1s.353ms)

And here is the flash command:

15:42:56 **** Build of configuration Sming for project HttpServer_ConfigNetwork ****
make flash 
Killing Terminal to free /dev/tty.usbserial-A50285BI
/opt/esp-alt-sdk//utils/esptool.py -p /dev/tty.usbserial-A50285BI -b 115200 write_flash -ff 40m -fm qio -fs 32m 0x00000 out/firmware/0x00000.bin 0x0a000 out/firmware/0x0a000.bin  0x4C000 out/firmware/spiff_rom.bin
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.
usage: esptool write_flash [-h] [--flash_freq {40m,26m,20m,80m}]
                           [--flash_mode {qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE] [--ucIsHspi]
                           [--ucIsLegacy] [--no-progress] [--verify]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0xa000 for file: out/firmware/0x0a000.bin
make: *** [flash] Error 2

15:42:57 Build Finished (took 233ms)

I am using  stock SMING...no changes with SDK 2.0.

The only changes to the makefile i have made are:

ESPTOOL		 ?= $(ESP_HOME)/utils/esptool.py
SPI_SIZE        = 4M
SMING_HOME = /Users/cpope/Development/Sming/Sming
COM_PORT = /dev/tty.usbserial-A50285BI
SPI_MODE ?= qio

Interestingly enough, BasicWebskeleton works.

I'll try to create a stripped down version of my app that shows the problem with large number of files and share it with you since I can't get httpConfigNetwork working

jssilva25, 2017-07-04:
I also feel:
Detected overlap at address: 0xa000 for file: out/firmware/0x0a000.bin
But not on all builds, this was on MeteoControl where 0x00000.bin is 44048 bytes long (0xac010, obviously it must overlap).
But building the Basic_Blinky doesn't overlap because 0x00000.bin is 28592 bytes long (0x6fb0); Basic_Debug and other small samples also don't overlap because 0x00000.bin is smaller than 0x0a000.

What I did to overcome the problem was, on the Makefile-project.mk:
- IMAGE_SDK	?= 0x0a000.bin
+IMAGE_SDK	?= 0x0b000.bin
and on compiler/ld/standalone.rom.ld
-  irom0_0_seg :                         org = 0x4020a000, len = (1M - 0x0a000)
+  irom0_0_seg :                         org = 0x4020b000, len = (1M - 0x0b000)

Is this correct? What are the consequences?

PR #1208 (2017-07-29): Added dynamic recalculation of image offsets to prevent rom overlapping.
Fixes: #1186, 
Fixes: #1158.
closed 2017-07-29 by slaff